jp112sdl/SonoffHMLOX #20: Schaltzustand wird nicht an Zentrale übertragen, wenn übers Web geschaltet wird

jp112sdl, 2017-10-10:
Beim Schalten AN/AUS über die Button im Web wird der Schaltzustandswechsel nicht an die Zentrale übertragen.

closed 2017-10-10 by jp112sdl

walleye13, 2017-10-17:
Hi, habe die 11 Version installiert. Mit dem Webinterface kann ich den Sonoff Aktor Schalten. Auch wenn ich den Taster am Aktor selbst betätige wird der richtige Status im Webinterface angezeigt. Mit der Homematic kann ich auch den Aktor steuern. Nur wenn ich den Taster am Aktor oder mit dem Webinterface steuere, wird der geänderte Zustand nicht auf der Homematic angezeigt. War die Homematic in diesem Issue nicht als Zenrale gemeint oder hilft das Update auf Version 11 alleine nicht?

jp112sdl, 2017-10-17:
Hallo walleye13,
ja, an sich ist dein beschriebenes Verhalten in der 1.0.11 behoben. Jedoch nur was das Web betrifft. Der Zustandswechsel beim Drücken des Tasters wurde bisher in allen Releases unterstützt.

Ich habe soeben noch mal die 11er .bin hier aus den Releases runtergeladen und geflasht.
Wenn ich am Taster oder im Browser schalte, wird der Zustand korrekt nach HomeMatic übertragen. 
Der CUxD Schalter springt sofort um. 

Ich kann mir höchstens noch vorstellen, dass dein CUxD Kanalname nicht mit dem Gerätenamen, den du bei der Sonoff-Wlan-Konfiguration angegeben hast, übereinstimmt. 

CUxD Kanalname:
[image]

Sonoff Config:
[image]

walleye13, 2017-10-17:
Hi jp112sdl, DANKE für die super schnelle Hilfe! Jetzt funzt es :-)

commit 481a3cbf (2017-10-10): fixed #20