jp112sdl/SonoffHMLOX #21: Feature Request: Howto for adding additional Sonoff devices

lmaertin, 2017-10-18:
Thanks for your brilliant work! I want to extend your code due to the 4ch pro from Sonoff. Can you please provide a short developer howto in the wiki. Thank you :)

jp112sdl, 2017-10-18:
Vielen Dank!

Hmm... wo fängt man an, wo hört man da auf?
Generell find ich gut, was du vor hast.  Aber Code ist wohl nicht so durchstrukturiert (Hobbyprogrammierer :) ), als dass man da mal fix noch neue Modelle hinzufügen könnte.
Deshalb ist auch ein Developer-HowTo nicht so einfach machbar.
Ggf. kannst du ja konkrete Fragen stellen, die ich dir beantworten kann.

lmaertin, 2017-10-18:
Hey, danke für die schnelle Antwort. Ich versuche mich mal an dem Basic-Sonoff zu orientieren und dann einfach nur 3 Kanäle hinzuzufügen.

Vorab aber eine Frage zum Verständnis. 
Sieht die Integration eigentlich überhaupt einen Rückkanal zur Bestätigung des Schaltbefehls vor oder sind die Sonoffs eigentlich immer nur Empfänger von Schaltbefehlen? Touch und Pow sind als "Sensoren" vermutlich eher die Ausnahme.

jp112sdl, 2017-10-18:
Deinen Eifer in Ehren, aber so einfach 3 Kanäle hinzufügen ist nur die halbe Miete. 

Was meinst du mit Rückkanal?
Grundsätzlich arbeitet dort ein ESP8266 oder equivalenter Microcontroller mit digitalen IO Pins. Die kannst du lesen (zB Taster) oder setzen (zB Relais). 
Das impliziert noch keine Antwort... darum muss man sich im Code selbst kümmern. 

Ich habe es so gemacht, dass jeder Befehl bestätigt wird. Also du sendest ein AN hin und bekommst eine Quittung zurück. 

Sonoff Switch,S20 und Touch sind intern identisch. Nur beim Touch ist der Taster eine berührungsempfindliche Fläche. 

Der Taster hat auch erstmal nichts direkt mit dem Relais zu tun.

Und der POW hat auch erstmal keinen „Rückkanal“... da drin ist ein Chip, der Pulse abgibt - je mehr Last dran hängt, desto höher die Pulsfrequenz. Vereinfacht erklärt, zählt man die und macht dann was draus. Zum
Beispiel an ein Ziel senden via HTTP. 
Das liegt alles in der Hand des Programmierers. 

Man bekommt an sich erstmal nur ne verdrahtete Platine mit angeschlossenen LEDs, Tastern und Relais. Wie dann was miteinander zusammenarbeitet oder worauf und wohin auch immer antwortet, ist alles Sache der Programmierung.

lmaertin, 2017-10-18:
Danke für die Infos, das mit dem Rückkanal war eher Homematic-Spreche ;) Dass eine Firmware für einen mC auch beim ESP low-level bleibt, ist mir bewusst. Hab auch nen paar NodeMcu s hier liegen. Wobei das heute alles einfacher ist als damals mit dem ATmega32 ohne tolle Libs, github und miesen IDEs...

Ich werde bei Zeiten Mal forken und mein Glück versuchen. Aktuell ist die Hardware noch nicht hier... Konkrete Fragen gibt's dann sicherlich. Danke bis hierhin schonmal.

jp112sdl, 2017-10-19:
Ja, was das betrifft, ist man heutzutage echt gut dran.
Ich nutze sonst vorzugsweise den Wemos D1 mini. 
Aber ist schon Wahnsinn, was man für 3,50 EUR bekommt. Einen SoC mit 80MHz, 4Mbit Flash und WiFi Controller.

Den 4Ch Pro habe ich bisher nicht berücksichtig, da ich im Umfeld von HomeMatic die 4fach Batterieaktorplaine mit einem kleinen Netzteil und ner 4er Relaisleiste betreibe. Das ist von den Kosten her unwesentlich mehr, als der 4CH Pro kostet und ich kann nativ mit dem Gerät in der WebUI arbeiten sowie das Funkprotokoll nutzen. Die Anzahl der Schaltvorgänge hält sich dort bei mir auch in Grenzen, so dass es keine Probleme mit dem DutyCycle gibt (Stichwort Batterieaktor, Burst-Telegramm).

Zurück zum Thema:
Für erste Ergebnisse, solltest du dir die Funktion `void switchRelay(bool toState, bool transmitState)` anschauen und erweitern. Dort findet das eigentliche Schalten der Relais statt. 
Vielleicht erweiterst du diese Funktion um einen Parametern oder fügst noch eine überladene Funktion hinzu, um den Rest des Codes erstmal kompatibel zu halten.
Und das ist aus meiner Sicht auch das geringste Übel. Der größte Aufwand (für mich) ist das Drumherum... Erweitern des WiFiManagers und der Weboberfläche.

Wenn du noch mit an dem Projekt entwickeln möchtest, freue ich mich natürlich und es gibt sicher auch noch andere User, die den 4CH Pro nutzen möchten. Die habe ich bisher alle abgewimmelt... :) 
Ggf. kommt noch der Dual. Aber dessen Relais lässt sich nicht einfach mit H/L an GPIO Pins schalten. Das ist ne Baustelle für sich...

scotty24ffm, 2017-10-19:
Hallo,
super das es hier vielleicht in der Richtung weitergeht, bin auch begeisterter Sonoff und homematik und iobroker user, und würde das natürlich auch gerne nutzen können. Ich bin leider nur Arduino Anfänger, aber ich habe schon mal den 4ch Sonoff abgelichtet, sprich das man sich jetzt schon übers layout und verschaltung gedanken machen kann. es scheint mir nämlich direkt auf den io‘s angeschlossen zu sein, schaut gerne mal:
http://kabelsalat.biz 
vielleicht hilfts ja schon ein bisschen bis eure teile kommen :

jp112sdl, 2017-10-19:
Hallo,

die GPIO Belegung des 4CH Pro ist:
`GPIO_KEY1,        // GPIO00 Button 1`
`GPIO_USER,        // GPIO01 Serial RXD and Optional sensor`
`GPIO_USER,        // GPIO02 Optional sensor`
`GPIO_USER,        // GPIO03 Serial TXD and Optional sensor`
`GPIO_REL3,        // GPIO04 Sonoff 4CH Red Led and Relay 3 (0 = Off, 1 = On)`
`GPIO_REL2,        // GPIO05 Sonoff 4CH Red Led and Relay 2 (0 = Off, 1 = On)`
`GPIO_KEY2,        // GPIO09 Button 2`
`GPIO_KEY3,        // GPIO10 Button 3`
`GPIO_REL1,        // GPIO12 Red Led and Relay 1 (0 = Off, 1 = On)`
`GPIO_LED1_INV,    // GPIO13 Blue Led (0 = On, 1 = Off)`
`GPIO_KEY4,        // GPIO14 Button 4`
`GPIO_REL4,        // GPIO15 Red Led and Relay 4 (0 = Off, 1 = On)`

"Eure" Teile sind höchstens die Teile von @lmaertin :)

Ich mach für den 4CH (Pro) nichts weiter.

lmaertin, 2017-10-20:
Ich werde vermutlich eher versuchen die Tasmota-Firmware zu erweitern. Ähnlich wie das dortige MQTT-Modul, werde ich die STATES an die CCU2 zurückschicken. Dann habe ich zwar keine Timer oder so, aber der Aufwand sollte geringer sein.

jp112sdl, 2017-10-20:
Korrekt. 
Du brauchst die Tasmota Firmware nicht erweitern. Da ist schon alles drin. 
Die fehlen dann halt Timer und Zustandsübertragung an die CCU wenn man das Gerät am Taster bedient. Es sei denn du hast irgendwo einen MQTT Broker zu laufen, der das dann wieder umsetzt...

lmaertin, 2017-10-20:
Ich dachte auch schon daran das über einen Broker umzusetzen, aber diese Erweiterung hm2mqqt gefällt mir nicht und zudem möchte ich eigentlich keinen Broker dafür laufen lassen. Weiterhin nur die Kommandos via curl abzusetzen, liefert mir kein ACK vom Sonoff zurück. 

Wie ich es bisher richtig verstanden habe, hast du diese synchrone Kommunikation implementiert. Der State in HM ändert sich also nur, wenn ein gesendeter Schaltbefehl von der Sonoff auch empfangen wurde, oder? Das wäre mir wichtiger als die Konsistenz bei Bedienung via Button am Gerät.

jp112sdl, 2017-10-20:
`http://<ip>/1?ts=1` oder
`http://<ip>/0?ts=1`

->`ts = transmitState`

Wenn man von irgendeiner Stelle aus den Sonoff mit dem Parameter `transmitState` schaltet, ändert sich auch der Zustand in HM, ja.

Ist es das Verhalten, was du meinst?

lmaertin, 2017-10-20:
Ja, vermutlich meinen wir das gleiche, ich denke an deine Methoden setStateCUxD(...) und getStateCUxD(...).

jp112sdl, 2017-10-20:
Genau. Das sind die "Dreh- und Angelpunkte" für die CCU-Kommunikation.

Und beim Aufruf von `switchRelay(bool toState, bool transmitState)` übergibst du als ersten Parameter den Soll-Zustand (an oder aus) und ob dieser anschließend auch in Richtung CCU übertragen werden soll (transmitState = true)

mindhunt, 2017-11-22:
Hallo jp112sdl,
vielen Dank für den Code, läuft bei 3 Stück POW einwandfrei.
Hallo @lmaertin,
bist du bzgl. dem 4CH weitergekommen? Habe das Teil auch gerade hier liegen..

lmaertin, 2017-11-22:
ich hatte ein Lösung mit TASMOTA (Webhook-Command zum Schalten) umgesetzt und dann via die API von TASMOTA und json-parser den Status ausgelesen. Inzwischen bin ich aber auf iobroker als Integrationsplattform gegangen und muss mich mit der CCU nicht mehr unnötig rumärgern. Ich kann dir den Kram raussuchen, aber vermutlich hat das hier im Thread dann nichts zu suchen. Sag mir einfach, wie ich dich erreiche.

mindhunt, 2017-11-22:
Sehr gerne,
rukafaz@gmail.com

(Lösche ich nach Bestätigung wieder raus hier)

plehr, 2017-11-23:
@lmaertin hast du schon eine Lösung gefunden um die Firmware von @jp112sdl in iobroker zu integrieren?

jp112sdl, 2017-11-23:
@plehr Ohne HomeMatic dazwischen?
Mit HM brauchst du ja nur mit dem hm-rpc Adapter direkt aufs Device gehen...
[image]

plehr, 2017-11-23:
Stimmt... So geht es natürlich auch. War wohl zu tief in der Adapter-Denkweise drin...
Danke funktioniert einwandfrei 👍

Gelegenheitsbastler, 2017-11-24:
Auch ich habe Dein Projekt über die große Suchmaschine entdeckt und bin begeistert. Da ich nicht sonderlich gut programmieren kann, ist diese "Plug-and-play" Lösung eine wirklich tolle Alternative zu den Aktoren von HM, die mMn etwas überteuert sind. 

Aus gegebenem Anlaß würde mich interessieren, ob das Modell "Dual" in absehbarer Zeit in Dein Projekt integriert werden wird. Das Geld für die Hardware würde ich gerne spenden.

jp112sdl, 2017-11-24:
@Gelegenheitsbastler 
Moin! Vielen Dank für dein Angebot.
Ich habe einen Dual hier zu liegen. Der ist so ne Sache für sich. Warum auch immer (nur bei dem) auf die direkte GPIO-Ansteuerung der Relais verzichtet wurde und eine "Zwischenbaustufe" verwendet wird... Es macht die Ansteuerung komplizierter.
In absehbarer Zeit werden wohl wegen Zeitmangel leider keine weiteren Sonoff-Modelle folgen. :/
Da bleibt dir als Alternative nur ESPurna oder Tasmota mit dem ganzen MQTT-Kram (und ohne Rückkanal zu HomeMatic).

Gelegenheitsbastler, 2017-11-26:
Das ist schade. Auf eine andere Lösung möchte ich nicht wechseln, da Deine Lösung schnell und einfach funktioniert. 

Aber eine Frage noch: Ist es möglich, zusätzlich zum Schalten über die HomeMatic einen Taster anzuschließen, über den ebenfalls geschaltet werden kann, wobei der sich dann verändernde Zustand an die HomeMatic übertragen wird?

jp112sdl, 2017-11-26:
@Gelegenheitsbastler Klar, das geht. Schau mal ganz unten im [Wiki](https://github.com/jp112sdl/SonoffHMLOX/wiki/Homematic)

`GPIO14 Mode` auf `Taster` stellen und den Haken bei `GPIO14 nur Sender` **nicht** aktivieren.

Gelegenheitsbastler, 2017-11-26:
Genau das klappt leider nicht. Ich kann dann (im 230V Betrieb) zwar sowohl von der RaspberryMatic Oberfläche als auch vom Taster aus das Relais hörbar schalten. Aber eine Änderung der Anzeige (EIN / AUS) in der Oberfläche sehe ich nur, wenn ich über die Oberfläche geschaltet habe. Mache ich das per Taster, ändert sich dort nichts. 

Liegt es vielleicht daran, dass meine Hardware etwas anders ist? Eigentlich kann das aber kein Grund sein, denn das Schalten ansich funktioniert, nur der geänderte Zustand wird nicht übertragen. 

Fotos der etwas anderen Hardwareversion (es fehlt der 5. Anschluss für GPIO14) ist auf disen Bilder zu sehen:

https://www.dropbox.com/s/gbf1s8ipo5ni1ds/20171126_194008.jpg?dl=0
https://www.dropbox.com/s/aarv3vjgv1l9tjr/20171126_194014.jpg?dl=0
https://www.dropbox.com/s/qbnyrhhnlooqhpd/20171126_194037.jpg?dl=0

jp112sdl, 2017-11-26:
Dir fehlt also nur die Rückmeldung.
Das lag bisher bei 100% der Useranfragen am falschen Namen...
Der Sonoff Gerätename von [hier](https://github.com/jp112sdl/SonoffHMLOX/wiki/Sonoff_Konfiguration) muss **identisch** sein mit dem CUxD Gerätenamen [hier](https://raw.githubusercontent.com/jp112sdl/SonoffHMLOX/master/Images/CCU_Geraetebenennung.png) (grüne Box)

Gelegenheitsbastler, 2017-11-26:
Perfekt. Das war der Fehler. Nun funktioniert es so, wie es soll. Danke!

jp112sdl, 2017-11-26:
@Gelegenheitsbastler Perfekt.
P.S.: Ich hab angefangen, den Sonoff Dual mit einer [Rollladen-Firmware](https://github.com/jp112sdl/SonoffDualShutterHMLOX) zu versehen...
Ist aber alles noch ganz rudimentär.

[image]

Gelegenheitsbastler, 2017-11-26:
YMMD!

Ich würde den Aktor dann aber nicht (nur) zur Steuerung der Rolladen nutzen wollen. Da ich zusätzlich auf jeden Fall noch eine Möglichkeit benötige, beide Relais getrennt voneinander über zwei Taster zu steuern, habe ich den Schaltplan mal grob überflogen:

https://www.itead.cc/wiki/images/1/1d/Sonoff_Dual_View.pdf
https://user-images.githubusercontent.com/30664452/32122885-6dddaf1c-bb62-11e7-977a-314b51aa7b6f.png

Demnach sind einige GPIO ungenutzt. Wäre es möglich, dass Du in der Firmware Variablen für zwei Taster und zwei LEDs einbaust? Die Funktionalität soll genau so sein, wie beim Modell Switch. Relais aus = LED aus. Taster drücken, Relais ein und LED ein. Wenn ich keine 4 GPIO finde (schaue ich mir morgen in Ruhe an), bastel ich etwas für 2 GPIO. Das wären dann die Eingänge für die Taster.

Die Verbindungen und fehlende Hardware würde ich mir passend erstellen und die Dokumentation (Schaltplan, Fotos, evtl. Platinenlayout) für Dein Projekt zur Verfügung stellen.

plehr, 2017-11-26:
@Gelegenheitsbastler du hast ebenfalls die Möglichkeit im Projekt deinen eigenen Code einfließen zu lassen. Du kannst Änderungen am bestehenden Code im eigenen Fork vornehmen und @jp112sdl kann diese akzeptieren.
Dann kann @jp112sdl sich weiteren spannenden Projekten widmen und deine Wünsche kannst du mit allen Funktionen umsetzen...

jp112sdl, 2017-11-26:
Also ich weiß bis heute nicht, was den Hersteller Itead dazu bewegt hat, beim Dual die Relais und Buttons über den F330 anzusteuern. Ich find es mega nervig. Du kannst nicht mal eben Relais 1 AN oder Relais 2 AUS schalten... sondern musst immer setzen, was AN bleiben soll.
Also Relais 1 + 2 sind an und du möchtest Relais 1 ausschalten, musst du 
`Serial.write(0xA0);`
`Serial.write(0x04);`
 `Serial.write(0x02);`
 `Serial.write(0xA1);`
an den F330 schicken (nur Relais 2 an).
Durch den ganzen internen seriellen Kram, ist auch das serielle Debugging schon mal hinüber.
Deshalb ist in einem Code noch das UDP-Zeugs drin, damit ich die seriellen Ausgaben abschalten und an einen Syslog-Server schicken kann.

Und aufgrund dieser genannten Thematik fällt es mir sehr schwer, das bestehende SonoffHMLOX Projekt zu erweitern. 
Sogar das Drücken des Tasters am Dual sendet auch nur einen Char seriell. Nix mit gucken, ob prüfen LOW wird, oder einen Interrupt verwenden :D 

Man könnte für den Dual nun noch eine komplett eigene Firmware schreiben, aber dazu fehlt mir echt die Ambition. Da nehm ich lieber 2 einzelne Sonoff Switch, statt einen Dual. ;) 

@plehr 👍

Gelegenheitsbastler, 2017-11-27:
Euer Vorteil ist, dass ihr Software versteht und anwenden könnt. Ich leider nicht. Als ich den riesigen Berg an Dateien beim Auffinden Deines Projektes gesehen habe, habe ich einen Schreck bekommen und war sehr froh, als ich im Verlauf der Beschreibung die Möglichkeit fand, ein bin-File nutzen zu können. Das bekomme ich noch hin, aber was ich mit dem Berg an Dateien hätte anstellen sollen? Keine Ahnung.

Ich könnte aber eine Platine dazu beisteuern, die die beiden erwähnten LED ansteuert und aus den beiden Tastern zwei "Schalter" macht, so dass für die beiden (auf den ersten Blick) ungenutzten GPIO04 und GPIO05 ein eindeutiges HIGH oder LOW zur Verfügung steht, um zu erkennen, wie die Relais geschaltet werden müssen. Das wäre dann doch eigentlich relativ überschaubar zu integrieren. 

Wie beschrieben, kann ich nur Hardware und steuere mein bescheidenes Wissen gerne bei. Software (in diesem Umfang) kann ich leider nicht.

Aktuell mache ich es übrigens genau so, wie Du schreibst. Ich nutze zwei Switch statt einen Dual.

Gelegenheitsbastler, 2017-11-29:
Da es mit dem Modell "Dual" so problematisch ist, weil die Relais nicht direkt über die GPIO gesteuert werden, habe ich nach Alternativen gesucht. (Wieder)Gefunden habe ich dieses Projekt, dass ich mir vor einiger Zeit schon mal angeschaut habe: 
https://luani.de/projekte/esp8266-hvio/

Wäre es damit einfacher? Die Relais werden direkt über die GPIO gesteuert.
GPIO04 - Kanal 1
GPIO05 - Kanal 2
GPIO15 - LED
GPIO12 - Eingang Taster 1
GPIO13 - Eingang Taster 2

Ein großer Vorteil wäre, dass die Eingänge mit 230 V AC angesteuert werden können und man somit keine großen Klimmzüge machen muss, um vorhandene Schalter/Taster zu verwenden. Außerdem passt dieses Projekt in eine UP-Dose, was für viele User eine Erleichterung darstellen dürfte.

jp112sdl, 2017-11-29:
Es ist ja nicht nur die Ansteuerung der Relais. Der gesamte Code ist nicht auf Mehrkanalbetrieb ausgelegt... Es müsste das komplette Web umgebaut werden und auch die Funktionen für Homematic/Loxone.
Ich kenne HVIO. Der Nutzerkreis, der sich so ein HVIO nachbaut, ist aber auch eher gering. Und zudem wird Tasmota als Firmware sogar unterstützt.

jp112sdl, 2017-12-04:
@Gelegenheitsbastler 
Gute Nachrichten! Ich habe bei mir einen Anwendungsfall für einen Sonoff Dual als Schaltaktor 👍 
Ich werde jedoch nicht diese FW hier erweitern, sondern ein eigenes Projekt für den Dual als 2fach Aktor erstellen.

Gelegenheitsbastler, 2017-12-05:
Das hört sich super an. Dann warte ich mal geduldig ab. 
Wäre es denn möglich, die oben verlinkte Hardware gleich mit einzubinden? Wie gesagt würde ich Dir die dafür notwendige Hardware zukommen lassen.

jp112sdl, 2017-12-05:
> Wie gesagt würde ich Dir die dafür notwendige Hardware zukommen lassen.

Du würdest also eine HVIO Platine anfertigen, komplett bestücken und mir fürs Prototyping zur Verfügung stellen?

Jap - dann kommen wir ins Geschäft 👍

Gelegenheitsbastler, 2017-12-05:
Genau so hatte ich das gemeint. 
Das gute Stück sieht so aus:
https://www.dropbox.com/s/oij22klye05qlfp/20171205_080008.jpg?dl=0
https://www.dropbox.com/s/okao9mq9jotuv48/20171205_080017.jpg?dl=0
Den AD/DC Wandler habe ich absichtlich noch nicht bestückt, da es in der Entwicklungsphase nicht ratsam ist, 230V am Testobjekt anliegen zu haben. Aber ich gehe mal davon aus, dass Du die vier fehlenden Lötstellen hin bekommst. 

Wenn Du mir Deine Adressdaten zukommen lässt, könnte ich die Platine (und den AC/DC Wandler) heute noch in die Post geben.

jp112sdl, 2017-12-05:
Echt coole Sache! Wie kommts, dass du sowas schon zu liegen hast? Hast du HVIO schon mit anderem Arduino Code verwendet?

Ich habe sogar noch einen HLK-PM01 hier zu liegen ✌️ 

Wie/wo kann ich dir meine Daten mitteilen? E-Mail? Falls du bei Facebook bist - der Link zu meinem FB-Profil steht hier im Profil.

Gelegenheitsbastler, 2017-12-05:
Ich kann besser Hardware als Software. Daher habe ich das Projekt nachgebaut. Es ist ein Blinksketch installiert. Über den seriellen Monitor ist etwas Text zu lesen und die Relais schalten im Sekundentakt. Die Hardware funktioniert also soweit. In Verbindung mit meiner RaspberryMatic habe ich das gute Stück noch nicht dazu bewegen können, zu tun, was ich möchte. 
Du erreichst mich unter frsa2002 yahoo de

jp112sdl, 2017-12-05:
@Gelegenheitsbastler 
Hast ne Mail. :)

Gelegenheitsbastler, 2017-12-05:
Ich kann besser Hardware als Software. Daher habe ich das Projekt nachgebaut. Es ist ein Blinksketch installiert. Über den seriellen Monitor ist etwas Text zu lesen und die Relais schalten im Sekundentakt. Die Hardware funktioniert also soweit. In Verbindung mit meiner RaspberryMatic habe ich das gute Stück noch nicht dazu bewegen können, zu tun, was ich möchte.

63chris, 2017-12-15:
Vorab - Tolle Sache, klasse. Das schafft echte Alternativen zu den standard HM Geräten. Herzlichen Dank!
Zur Info - ich habe eben die FW mal auf einen Sonoff SV geflasht (Rx/Tx gekreuzt, enter flash mode wie beim Sonoff Basic). Funktioniert tadellos. Ich bin begeistert.
Man könnte dieses Gerät ja im WIKI in die Liste der unterstützten Geräte hinzufügen.

jp112sdl, 2017-12-15:
@63chris Vielen Dank! ...für beides :)
Habe ich hinzugefügt! 👍

closed 2018-04-06 by jp112sdl