jp112sdl/SonoffHMLOX #22: Zustandsrückmeldung

Java-Jim, 2017-10-25:
Hallo jp112sdl,

ich habe grad ein paar Sonoff Touch geliefert bekommen, mit deiner Firmware geflasht und als Schater in die Homematic CCU1 eingetragen. Das funktioniert soweit und ich bin mit der Firmware auf den ersten Blick ziemich zufrieden.
Ich habe im vorhergehenden Issue etwas von TransmitState gelesen. Habe ich das richtig verstanden, dass bei passender Einrichtung der Touch Schalter nach jedem Schaltvorgang (manuell wie auch per Netzwerk) ein StateUpdate an den CUxD sendet?
Nach dem Wiki Artikel kann ich derzeitig nämlich nur Schalten ohne das sich der Zustand in der HM ändert, wenn ich manuell den Schalter bediene. Es wäre schon gut, wenn ich den aktuellen Zustand sehen kann, um zu erkennen ob ich das Licht im Büro vergessen habe auszuschalten...
Was müsste dazu unternommen werden?

Gruß Java-Jim

jp112sdl, 2017-10-25:
Hallo @Java-Jim,

dein Fehler ist womöglich darauf zurückzuführen, dass das CUxD Device nicht denselben Namen hat ([Bild](https://github.com/jp112sdl/SonoffHMLOX/raw/master/Images/CCU_Geraetebenennung.png)) wie der Gerätename, den du bei der Sonoff-Einrichtung im Konfigurationsportal eingegeben ([Bild](https://github.com/jp112sdl/SonoffHMLOX/raw/master/Images/Sonoff-ConfigMode1.png)) hast.

Das wäre so auf Anhieb meine einzige Erklärung, weshalb der Zustand nicht übertragen wird.

Habs gerade noch mal bei mir getestet. Wenn ich den Touch durch Berührung schalte, ändert sich auch der Zustand des CUxD Schalters.

Java-Jim, 2017-10-25:
Oh ja das kann sein. Ich hab auch Leerzeichen drin. Werde es gleich mal anpassen.
Kann man im CUxD die Geräte nachträglich ändern? Ich habe nur die Option das Symbol zu ändern...
Sonst lege ich die Geräte eben neu an.

jp112sdl, 2017-10-25:
Ja kannst du. Brauchst nur den Kanal anklicken... wie man auch die Kanäle von allen HomeMatic Geräten umbenennen kann.

jp112sdl, 2017-10-25:
Es geht ja um den Kanal, auf dem der Touch läuft (im Bild vorhin "Sonoff1"). Nicht den CUxD Schalter als solches ("Sonoff").

Java-Jim, 2017-10-25:
Ich habe die Geräte im CUxD gelöscht und wieder neu angelegt. Jedoch steht beim ersten Kanal „nicht verknüpfbar“ und die zwei CMDs werden nicht gespeichert.

jp112sdl, 2017-10-25:
"Nicht verknüpfbar" steht bei mir bei allen Kanälen... Das ist wohl der Tatsache geschuldet, dass sich CUxD Geräte nicht direktverknüpfen lassen.

Warum die CMDs nicht gespeichert werden, weiß ich nicht. Vielleicht liegt am `?`.
Du brauchst den Parameter dort auch nicht
`http://<ip>/0` und `http://<ip>/1` reichen aus.

Der Parameter ts=1 veranlasst den Sonoff, seinen Zustand ans CUxD Gerät zurückzumelden. Beispielsweise, wenn du ihn per HTTP-Befehl von einer anderen Anwendung heraus schaltet.
Beim CUxD CMD brauchst du es nicht, da er ja selbst das Schalten ausführt und somit weiß, was er macht :)

Java-Jim, 2017-10-25:
Ich habe den ts Parameter und das ? im CMD weggelassen, so wird der Eintrag auch gespeichert.
Und was soll ich sagen, es funktioniert jetzt mit Zustandsrückmeldung. Grossartig!
Ich danke dir für deine Unterstützung und hoffe das es anderen hilft.

jp112sdl, 2017-10-25:
Prima, freut mich, dass es nun auch bei dir läuft!

closed 2017-10-25 by jp112sdl