jp112sdl/SonoffHMLOX #24: Externer Schalter

DoktorDennis, 2017-11-15:
Hallo Jerome,

erstmal ein Lob für deine Arbeit. Läuft Top wenn man sich an die Anleitung hält. Aber wie es ja immer ist einen kleinen Wunsch hätte ich 
Bei Tasmato kann man an den GPIO 14 einen Schalter aufklemmen und hat so auch die Möglichkeit per Schalter den Switch zu schalten. 
Man könnte jetzt ja auch schon an den Tasterleitungen anschliessen und somit mit einen Wandtaster auslösen aber Schalter wäre schöner.
Ich habe es schon versucht den Punkt im Quellcode zufinden aber ich muss erkennen ich bin ein Noob...

Gruss und Danke
   ###Dennis

jp112sdl, 2017-11-15:
Hallo Dennis,

erstmal vielen Dank! 👍 

mir war gar nicht bewusst, dass der GPIO Pin 14 bereits auf den Pad der Platine ausgeführt ist.
Ich hab den mal mit in die Firmware aufgenommen.
Wäre toll, wenn du es mal testen könntest.

https://github.com/jp112sdl/SonoffHMLOX/releases/tag/1.0.13a

DoktorDennis, 2017-11-15:
Hallo Jerome,

perfekt das ging aber schnell.
So aufgespielt und es klappt mit einen Taster. Super
In Tasmato kann man zwischen Taster und Schalter wählen.
Meinst du, dass es ähnlich einfach umzusetzen?

Gruss
Dennis

jp112sdl, 2017-11-15:
Was ich mich grade frage: Wie bist du auf meine Firmware gekommen, wenn Tasmota scheinbar den größeren Funktionsumfang bietet?

Zurück zur eigentlichen Frage. Was meinst du mit "Taster" und "Schalter"?
Du meinst, was man GPIO 14 anschließt - einen Taster oder Schalter?
Oder das Verhalten des Sonoff? Taster = Relais nur so lange an, wie man drückt; Schalter = Toggle An/Aus?

jenshavelberg, 2017-11-15:
@jp112sdl  @DoktorDennis  meint, dass bei Schalter (sein Feature Request) der PIN geschlossen bleibt und entgegen des Tasters eben nicht wieder geöffnet wird. Der nächste Schaltimpuls ist dann also nicht das folgende Tasten sondern das Öffnen des Schalters.

jp112sdl, 2017-11-15:
@jenshavelberg Ah, jetzt ja 👍 Ok, hab ich gerafft. War vorher auf ne Eltako-Schaltung hinaus...

@DoktorDennis Bitte testen, neue Firmware...
https://github.com/jp112sdl/SonoffHMLOX/releases/tag/1.0.13b

DoktorDennis, 2017-11-15:
Hallo,

> Was ich mich grade frage: Wie bist du auf meine Firmware gekommen, wenn Tasmota scheinbar den größeren Funktionsumfang bietet?

Was hilft der ganze Umfang von MQTT etc... wenn die Steuerung aus der Homematic nicht klappen will und auch keine Rückmeldung gibt.
Die Firmware findet man recht schnell wenn man Sonoff mods sucht. Also habe ich damit angefangen dann aber gemerkt das wird nix.
Dann bin ich auch Zufall auf deine Firmware gestossen und lief sofort. 
Nun hätte ich einfach an den original Taster ja die Leitung anlöten können fertig aber ich habe halt in der Wand einen Schalter keinen Taster daher die Frage.

So neuesten Stand getestet. Das läuft leider nicht. Geht an aber aus erst wenn wieder aus und wieder anschaltet. Also der Stromkreis muss einmal aufgehen und wieder geschlossen werden.

Gruss
Dennis

jp112sdl, 2017-11-15:
> Was hilft der ganze Umfang von MQTT etc... wenn die Steuerung aus der Homematic nicht klappen will und auch keine Rückmeldung gibt.

Genau aus diesem Grund hab ich mich rangesetzt und was Eigenes geschrieben. 👍 

> Das läuft leider nicht.

Oh je... das war offensichtlich 🤐 

Ich hoffe du hast noch Geduld... Ich hab nämlich keinen Sonoff zum Testen da. :) 

Please try again:
https://github.com/jp112sdl/SonoffHMLOX/releases/tag/1.0.13c

DoktorDennis, 2017-11-15:
Hallo 

So läuft, ich freu mich. Total cool... 
So soeben Donated damit du dir einen Sonoff besorgen kannst.

Gruss
Dennis

jp112sdl, 2017-11-15:
> So soeben Donated damit du dir einen Sonoff besorgen kannst.

Echt toll! Tausend Dank!

Mich interessiert noch der Use Case... 
Der Schalter representiert ja nicht immer den tatsächlichen Zustand. 
Du kannst zwar per Schalter beispielsweise einschalten, jedoch im Web/HTTP wieder ausschalten. 
Dann stimmt die Schalterposition nicht mehr mit den Zustand überein.

DoktorDennis, 2017-11-16:
Hallo 

so gestern abend habe ich nicht geschafft:

Use Case: Deckenspots die jetzt über Schalter geschaltet werden aber aus bequemlichkeit selten angemacht werden.
Nun sollen Sie in die in der Szene "Guck was ich kann"  :-) auch automatisiert und per alexa gesteuert werden. Aber es soll halt noch per Schalter schalter sein. Also für die Technikverweigerer.

Ja da hast du recht. Mit der Schalterposition stimmt. 
Das müsste jetzt eine deutliche komplexere Auswertung und da ist die Frage ob die Version A mit Taster es auch tut.

Um das Problem zu lösen müsste mit einen Ausschalten mehr WEB oder Taster am Sonoff der Wert 
des GPIO 14"closed" für den nächten Schaltvorgang per Schalter auf GPIO 14 "open" umgesetzt werden, und umgedreht jenach dem das den letzten Schaltvorgang zum Anschalten benutzt werden.

Also wie gesagt komplex vielleicht zu komplex und dann könnte es ja auch andere geben die es mit Taster gut finden, dann würde auch noch eine Auswahl on Taster oder Schalter notwendig macht.

Also ich kann mit Taster lösen.

Gruss
Dennis

jp112sdl, 2017-11-16:
Ich verstehe, was du meinst.

Wie eine Wechselschaltung - die Schalterposition ist nicht absolut, sondern sie toggelt beim Schalten einfach das Relais. 

Bau ich ein 👍 

Wenn du selbst kompilieren kannst - im Develop ist derzeit schon alles soweit drin, dass du den GPIO14-Mode im Web in der Konfiguration einstellen kannst (Aus / Taster / Schalter).

jp112sdl, 2017-11-16:
So, seit diesem [Commit](https://github.com/jp112sdl/SonoffHMLOX/commit/7b65a333f7808cbebfabc1e2ed1665a7f38ee2de) ist jetzt auch die Auswahlmöglichkeit drin, ob die Schalterposition absolut sein soll, oder eine Wechselschaltung (toggle) bedient

DoktorDennis, 2017-11-16:
> Wenn du selbst kompilieren kannst - im Develop ist derzeit schon alles soweit drin, dass du den GPIO14-Mode im Web in der Konfiguration einstellen kannst (Aus / Taster / Schalter).
ich dachte ich kann wenigsten das aber es will nicht und es macht auch keinen Sinn die Fehlermeldung..

jp112sdl, 2017-11-16:
@DoktorDennis [1.0.13d](https://github.com/jp112sdl/SonoffHMLOX/releases/tag/1.0.13d)

DoktorDennis, 2017-11-16:
Hallo und Danke

so habe es aufgespielt und es funktioniert solange nicht der Haken bei GPIO14 gesetzt wird. 
Mit dem Haken läuft nix mehr.

Und nur beim kompilieren bemängelt die Arduino IDE das die HL1802.h Datei und ESP8266.ping.h nicht zu finden sind. 
Wobei ich mir sicherbin das ich der Fehler bin.

Schönen Abend.

Dennis

jp112sdl, 2017-11-16:
> Mit dem Haken läuft nix mehr.

Das ist richtig. 
Wenn der Haken gesetzt ist, wird der GPIO14 als eigenständiger Schalt-Kontakt behandelt und an die CCU übertragen. Somit kann man den Sonoff noch als Wandsender nutzen, getrennt vom Relais. Eine genaue Beschreibung kommt dann noch zum Release.

Du lässt also den Haken raus und stellst `GPIO14 Mode` auf `Schalter (toggle)`

Die fehlende HLW8012 Lib findest du hier im Projekt-Ordner [libraries](https://github.com/jp112sdl/SonoffHMLOX/tree/master/libraries).
Und die Ping-Library gibts [hier](https://github.com/dancol90/ESP8266Ping).
Die Ordner kopierst du in dein Arduino-libraries Verzeichnis.

DoktorDennis, 2017-11-18:
Hallo jp112sdl,

ich bin wirklich begeistert, auch sehr cool wie du es jetzt noch erweiterst. Eine echte Alternative zu Tasmato, den perfekte Integration in die Homematic.
Brauchst du Fotos für die Dokumentation? 

Wegen mir kannst du das Ding schliessen, wenn was zu erproben ist sag Bescheid.

Gruss
Dennis

jp112sdl, 2017-11-18:
@DoktorDennis Ja, Fotos nehm ich sehr gern!

closed 2017-11-21 by jp112sdl

CenkHH, 2017-12-05:
Hallo Jerome,

Toll das man Loxone und Homematic mit Sonoff anbinden kann.
Ich habe den Sonoff TH10 mit Temperatur Sensor (TH11), weiterhin den Sonoff SC.
Beide möchte in Loxone anbinden und mir die Messdaten für die Heizsteuerung angeben lassen.
Würdest du bitte auch die SonoffHMLOX auf diese beide tollen Produkte erweitern.

Vielen Dank
Gruß
Cenk

jp112sdl, 2017-12-06:
@CenkHH Hallo Cenk,
die Unterstützung für weitere Sonoff Modelle ist leider nicht vorgesehen.

CenkHH, 2017-12-06:
Hallo Jerome,

 

danke für die schnelle Antwort.

Schade das die Software Limitiert auf zwit Sonoif Produkte ist.

 

Gruß Cenk

 
 

Gesendet: Mittwoch, 06. Dezember 2017 um 06:36 Uhr

jp112sdl, 2017-12-06:
Naja, es ist kein kommerzielles Produkt und ich programmiere (leider) auch allein.

Ursprünglich hatte ich für mich die Firmware geschrieben und wollte sie dann quelloffen für alle bereitstellen, damit auch andere davon profitieren. 
Dazu kam dann noch die Loxone-Unterstützung für einen engen Freund - das war es dann aber auch. 
Letztendlich kann jedoch jeder den Quelltext verwenden und anpassen oder ein ganz neues Projekt daraus entstehen lassen.