jp112sdl/SonoffHMLOX #26: Invalid Head of packet

FuXXz, 2017-12-16:
Hallo,

meine ersten beiden Sonoff haben super geklappt. Habe nun ein paar neue bekommen und dort bekomme ich beim flashen immer Invalid head of packet.
Was kann ich tun?

jp112sdl, 2017-12-17:
Moin!

Da fallen mir ein paar Dinge ein:
- steckt der FTDI auf 3.3V?
- evtl. RX/TX vertauscht?
- Sonoff nicht im FlashModus? (Taster beim Anschließen an Spannung gedrückt halten!)

Was jedoch bei dem Problem bisher die häufigste Ursache war:
- Leistung des USB-Anschlusses reicht nicht aus. Die Spannung bricht zusammen, sobald der Flashvorgang startet.

Hast du die Möglichkeit, das Flashen von einem anderen PC/Notebook aus durchzuführen?
Oder einen aktiven USB-Hub dazwischen zu klemmen?

FuXXz, 2017-12-18:
Hi,
danke für die Antwort.
Ich habe jetzt noch weitere Sonoffs bearbeitet, aber dort kam der Fehler nicht.
Also vermute ich einfach mal, dass ich den FlashModus nicht an hatte. Ich krame den besagten Sonoff gleich noch einmal raus und berichte.
Viele Grüße

jenshavelberg, 2017-12-30:
Hat das flashen jetzt bei dir geklappt?

FuXXz, 2017-12-30:
Hi,
jep, hat alles geklappt. Es lag wohl wirklich daran dass der Flashmodus nicht aktiv war

closed 2017-12-30 by jenshavelberg

FiegeU, 2017-12-31:
N'abend,
ich weiß ist closed, ich habe übrigens den selben Bug beim flashen einer Micropython FW auf einen Sonoff Switch.
Die SonoffHMLOX lässt sich problemlos draufspielen.
Hat jemand eine Idee dazu ?

Falshe unter Windows, mit verschiedenen Tools ausprobiert, fängt an zu flashen, dann bei etwa 20%  kommt aber immer zu diesem Bug... Es läuft dann weiter, aber lauffähig ist das nicht, was da rauskommt.

Ist aber nicht bricked danach.

Danke,
Uli

jp112sdl, 2017-12-31:
Ich glaub nicht, dass hier jemand schon jemals mit Micropython gearbeitet hat

sabbelhardy, 2018-10-17:
Moin. 
Ich hatte vor einiger Zeit ein Sonoff Basic in meiner Homematic integriert. Nun wollte ich noch weitere flashen und ich habe nur Probleme. 

Der ESP-Loader startet mit der "SonoffHMLOX.ino.generic.bin" bricht bei einem Sonoff immer mit der Meldung "Invalid head of packet" ab. Bei einem 2. Sonoff erfolgt mal der Abbruch mit gleicher Meldung und manchmal läuft der Loader bis 100% durch. Anschließend soll ja die Debug-Ausgabe erscheinen - leider bei mir nicht mit dem gewünschten Erfolg. Es erscheint folgendes als Liste: 

Wifi evt: 7 - das 10x untereinander. Anschließend:
WM. failed to connect an hit timeout
bcn Odel ifusl
ets Jan 8 2013, rst cause:1, boot mode: (1,7)  - das 2x untereinander. Anschließend noch:
wdt reset

Auf dem FTDI232 sind die 3,3V gesteckt.
RX-TX sind vertauscht. VCC und GND sind auch korrekt verbunden
Die Lötstellen der eingebauten Stifte wurden mehrmals geprüft (Kalte Lötstelle o.ä.)
Der Taster wurde beim Anschluss an den USB-Port gedrückt
Der USB-Anschluss wurde an unterschiedlichen Steckplätzen am PC - vorne/hinten versucht
Das USB-Kabel wurde gewechselt

Jetzt bin ich am Ende ;-)
Naja nicht wirklich - Aber vielleicht bekomme ich hier einen Tipp bzw. Hilfe, damit ich die "Dinger" endlich zum Laufen bekomme.

Viele Grüße
Hartmut

jp112sdl, 2018-10-17:
Hi!

Mir würde noch einfallen, dass der USB-Port evtl. zu wenig Strom liefert.
Das würde auch erklären, warum mal geht und mal nicht oder mal nur bis zur Hälfte.

Hast du die Möglichkeit, einen aktiven USB-Hub zwischen Rechner und FTDI zu stecken?

jp112sdl, 2018-10-18:
> WM: failed to connect and hit timeout

Das ist der WifiManager... da ist mit dem WLAN noch was nicht in Ordnung.
Kannst du mal das gesamte Debuglog von Beginn an posten?

jp112sdl, 2018-10-18:
> /config.json not found.

Das heißt, dass noch gar keine Konfiguration gespeichert wurde.
Die Konfiguration wird nur gespeichert (und die Konfigurationsdatei nur dann erstellt), wenn die WLAN-Verbindung erfolgreich hergestellt werden konnte. 
Ansonsten fällt der Sonoff immer wieder zurück in den AP-Modus.

sabbelhardy, 2018-10-18:
Hallo Jérome,
ich danke dir für Deine Hinweise. Ich habe es geschafft.
Mit der Mac-Adresse hatte die Sonoff im Router eingetragen. Das hat jedoch nichts genützt. Hatte übersehen, dass ich noch "WLAN-Zugang auf die bekannten WLAN-Geräte beschränken" eingestellt hatte. Nach Abänderung lief alles, wie in der Anleitung beschrieben.
Danke nochmals und einen schönen Donnerstag.
Hartmut

Dennis235, 2018-12-13:
´Hallo,

ich habe leider das gleiche Problem wie ober beschrieben.
Ich bekomme immer mehrfach die Mitteilung "Invalid head of packet"
Ich habe schon alles mögliche ausprobiert, aber nichts führt zum Erfolg.
Habe es auch schon per ESP Loader, Python, SonOTA, Atom, und noch ein zwei Varianten probiert, aber ich bekomme die Dinger einfach nicht geflasht, habe es mit Windows und Mac probiert.
Ich habe ach schon verschiedene Hardware ausprobiert mit einem s20, 3 verschiede Basic und 2 pow.
Mein Wunsch ist es die Dinger irgendwie in mein Loxone System zu integrieren da mir auf Dauer die nano io air zu teuer werden.
Ich wäre echt sehr glücklich wenn mir jemand bei meinem Problem weiterhelfen kann.
Gerne auch vor Ort wenn jemand aus der nähe von 51766 kommt.
Würde ca 100km fahren oder halt auch bei mir.
Würde mich auch finanziell erkenntlich zeigen.

Gruß Dennis

P.S Ich willl nicht das jemand Sie mir für Geld flasht, ich will es selber hin bekommen

PaulMerk, 2018-12-13:
Wenn du bis Mitte Januar keine Lösung gefunden hast, kann  ich versuchen zu helfen (wohne in der Nähe - 51427 - bin aber kein Spezialist !!!) 
Ich hatte auch das Problem - irgendwann hat es bei mir, auch durch Hinweise von Jérôme, geklappt.
Gruß
PM

BigJohnson68, 2018-12-13:
wow - so viele Programme habe ich noch nicht probiert -
hört sich nach Verbindungsproblem zwischen Sonoff und Software an
TX/RX vertauscht
TTL 5v statt 3,3v
kein Treiber hierfür installiert
richtige com-Port verwendet
hat es überhaupt schon mal funktioniert bzw. erfolgreich geflasht
GPIO 0 vor Flashvorgang gedrückt

habe sogar tote Sonoffs wieder zum Leben geflasht - musste zwar mehrmals ran und whipe

PaulMerk, 2018-12-14:
gute Auflistung, mir fällt noch ein:
Kontaktprobleme mit der Sonoff-Platine