jp112sdl/SonoffHMLOX #27: Sonoff Switch schaltet manchmal nicht

RealUli, 2017-12-17:
Moinsen Jérôme,
ich habe vier Sonoff Switch alle mit der aktuellen Firmware, die ich via NodeMCU (MicroPython) einschalte.
(Kann den Code bei Bedarf zeigen, nix spektakuläres.)

Manchmal ist das Einschalten verzögert und manchmal schlägt es ganz fehl. (Das Micropython der MCU gibt dann einfach nur OSError aus...)
Alle Geräte befinden sich im selben WLAN.

Was ich beobachtet habe ist, dass die Switche auch manchmal recht "lange" brauchen um auf Pings zu antworten. Es scheint so, als würden sie in einen Schlafmodus gehen.

Das hier ist noch gut:

C:\Users\XXX>ping -t 192.168.99.201

Ping wird ausgeführt für 192.168.99.201 mit 32 Bytes Daten:
Antwort von 192.168.99.201: Bytes=32 Zeit=152ms TTL=128
Antwort von 192.168.99.201: Bytes=32 Zeit=18ms TTL=128
Antwort von 192.168.99.201: Bytes=32 Zeit=33ms TTL=128
Antwort von 192.168.99.201: Bytes=32 Zeit=11ms TTL=128
Antwort von 192.168.99.201: Bytes=32 Zeit=59ms TTL=128
Antwort von 192.168.99.201: Bytes=32 Zeit=11ms TTL=128
Antwort von 192.168.99.201: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.99.201: Bytes=32 Zeit=9ms TTL=128
Antwort von 192.168.99.201: Bytes=32 Zeit=6ms TTL=128

Oder auch mal so:

C:\Users\XXX>ping -t 192.168.99.201

Ping wird ausgeführt für 192.168.99.201 mit 32 Bytes Daten:
Antwort von 192.168.99.201: Bytes=32 Zeit=96ms TTL=128
Zeitüberschreitung der Anforderung.
Antwort von 192.168.99.201: Bytes=32 Zeit=90ms TTL=128
Antwort von 192.168.99.201: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.99.201: Bytes=32 Zeit=112ms TTL=128
Antwort von 192.168.99.201: Bytes=32 Zeit=12ms TTL=128
Antwort von 192.168.99.201: Bytes=32 Zeit=21ms TTL=128
Antwort von 192.168.99.201: Bytes=32 Zeit=28ms TTL=128
Antwort von 192.168.99.201: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.99.201: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.99.201: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.99.201: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.99.201: Bytes=32 Zeit=3ms TTL=128

Wenn es einmal läuft, antwortet er auch recht zügig. Daraufhin hatte ich die Idee, einfach jede Sekunde einmal Port 80 von der NodeMCU aus auf dem Switch zu öffnen und ohne Request wieder zu schließen.
Das hat die Sache noch verschäft, dann war das Einschalten noch verzögerter...

Hättest Du eine Idee dazu ?

An ein Problem im WLAN mag ich nicht so recht glauben, Downloads und das Videostreaming von der SAT-Box funktionieren gut.

Danke,
Uli

Gelegenheitsbastler, 2017-12-17:
Sehr ähnliche Probleme habe ich mit einem S20. 
Mit welcher Hardware läuft Dein Wlan? Bei mir ist es eine 7490 und ein 1750E mit aktuellster Firmware im Mesh.

jp112sdl, 2017-12-17:
Moin!

Das Verhalten entspricht dem [#2330](https://github.com/esp8266/Arduino/issues/2330) und ist auf einen noch nicht gelösten Fehler in der ESP SDK zurückzuführen. Man sucht schon eine Weile...

jp112sdl, 2017-12-17:
Komischerweise habe ich solche Aussetzer gar nicht... *3x auf Holz klopf*

RealUli, 2017-12-17:
Ich habe eine 7390 mit der EWE Firmware drauf:
Version installiert:
FRITZ!OS: 06.83

jp112sdl, 2017-12-17:
Ich nutze eine Apple TimeCapsule als AP

RealUli, 2017-12-17:
@jp112sdl Denkst Du das es am Switch liegt, oder eher an der MCU ? 
(Die Switche tauschen magst Du vermutlich nicht, oder ? :) )

jp112sdl, 2017-12-17:
Nein, never change a running system ;) 

Aber ich denke auch nicht, dass es an der ESP-Hardware liegt. 

Es wird irgendeine Kombination zwischen AP-Modell und der LWIP Implementation in der ESP SDK liegen. Ist echt schwer... Das Ticket gibts auch schon über 1 Jahr.

RealUli, 2017-12-17:
War auch nicht so ernst gemeint mit dem tauschen.

OK, dank Dir trotzdem !

Was ich überlegt hatte, war ob man nicht direkt einen AP auf dem Switch aufmachen kann und dann direkt die Verbindung P2P macht. Meinst Du das macht Sinn ?

jp112sdl, 2017-12-17:
Lies dir mal die 3 Kilometer Comments zu dem Issue durch :) 

Es gab wohl auch solche Versuche, die auch nicht zielführend waren.

Es hilft leider auch nicht, vom Sonoff aus bspw. das Gateway zyklisch anzupingen, um die Verbindung aufrecht zu erhalten. Alles schon versucht, weils bei nem Kumpel auch nicht astrein läuft.

jp112sdl, 2017-12-17:
@FiegeU 
Eher nicht. Das hilft ja auch nicht... angenommen, er läuft bei mir, dann wissen wir noch nicht, warum er bei dir zickt. Und wenn er bei mir auch zickt... Haben wir auch nix gewonnen ;)

FiegeU, 2017-12-17:
@jp112sdl 
Was wir aber wüssten, ist, ob es bei der Time Capsule als AP auch auftritt. 
Das war eigentlich die Idee dahinter...

jp112sdl, 2017-12-17:
Oh man... ich werd alt!!!

Ich hab das Problem auch **gehabt**!

Hab das [hier](https://github.com/esp8266/Arduino/issues/2330#issuecomment-338921334) auch kommentiert.

Da ich jetzt nen Dauerping (nicht _vom_ - sondern _zum_ Sonoff)  laufen hab, tritt die Verzögerung nicht mehr auf.

Ist schon fast 2 Monate her... Deshalb war es mir wohl nicht mehr so bewusst.

Mea culpa :)

FiegeU, 2017-12-17:
Danke für den Hinweis.
Ich habe jetzt mal ein Ping-Script auf meinem Server eingerichtet und alle 5 Minuten in der Aufgabenplanung laufen lassen.
Ist jetzt nicht Nagios, aber das möchte ich jetzt nicht extra installieren.

Falls es jemand sehen möchte:

date /t > c:\tmp\sonoffLog.txt
time /t >> c:\tmp\sonoffLog.txt
ping 192.168.99.201 >> c:\tmp\sonoffLog.txt
ping 192.168.99.203 >> c:\tmp\sonoffLog.txt
ping 192.168.99.204 >> c:\tmp\sonoffLog.txt

Vielleicht noch ein paar Eckdaten. Hab im (W)-LAN eine Homematic auf einem Raspi3 und 6 oder 7 Clients. Notebooks, iPhones, Mini PC in der Küche.
Die Homematic funkt auch noch im 8xx Mhz Band. Der Raspi ist physikalisch direkt neben der Fritzbox an die Wand geschraubt (50 cm Abstand).
Vielleicht kann jemand eine Regel ableiten :)

Ich werd es mal ein paar Tage beobachten.

jp112sdl, 2017-12-18:
Ja, Nagios brauchst dafür nicht extra. Bei mir läuft der Ping mittlerweile über CUxD.

Ansonsten ist es bei mir ähnlich. AP und HM-Raspi sowie ioBroker-Raspi sitzen dicht beieinander.

Sonoffs hab ich 7 Stück im Einsatz, dazu kommen noch 15 Wemos D1 mini, die ja auch nen ESP8266 µC besitzen.

FiegeU, 2017-12-22:
N'abend zusammen (oder moinsen),
ich hatte jetzt ein paar Tage das Ping-Script mitlaufen lassen.
Aussetzer gibt es immer noch :( , aber ich habe das Gefühl, die Peaks sind ein bisschen raus.
Vor ein paar Tagen kam ich heim, das Licht ging an und nach 60 Sekunden wieder aus, wie gewollt. Sofort danach ging ich wieder in den Bereich des Bewegungsmelder und das Licht hätte wieder einschalten müssen, aber da dauerte es 10 Sekunden bis es sich wieder einschaltete.
70 Sekunden vorher ging es ohne spürbare Verzögerung.

Aber ansonsten "gefühlt" besser, danke !

jp112sdl, 2017-12-22:
> Sofort danach ging ich wieder in den Bereich des Bewegungsmelder und das Licht hätte wieder einschalten müssen

Ja, bei so kurzer Zeitspanne schon! Ich kenne die Aussetzer von früher nur, wenn ich einen Sonoff mal einen halben Tag lang nicht geschaltet hatte. Kurz hintereinander ging immer.
Schon echt kurios.

FiegeU, 2017-12-22:
Ist aber nicht nur einer, der diess Verhalten hat. Bei mir im Schlafzimmer ist ein Sensortouchfeld mit einem 8266 dahinter, der hat auch einen Morgen erst mit 10 Sekunden Verzöderung eingeschaltet, das war aber vor dem Pingscript...
Bin mir aber auch nicht sicher, ob es nicht doch Störungen durch den Homematic (8xxMHz ?) Sender des Raspis gibt...
Den Ferritkern habe ich aber um die Zuleitung gelegt.

jp112sdl, 2017-12-22:
> Bin mir aber auch nicht sicher, ob es nicht doch Störungen durch den Homematic (8xxMHz ?) Sender des Raspis gibt...
Den Ferritkern habe ich aber um die Zuleitung gelegt.

Nee, da stört nix das WLAN

FiegeU, 2017-12-22:
Ich würd das Ding jetzt zumachen wollen, oder meinst Du es gibt da noch Progress ?

jp112sdl, 2017-12-22:
Von meiner Seite aus nicht. Zumal es auch nicht direkt mit dem Projektcode zusammenhängt.
Mach ich zu - frohes Fest :)

closed 2017-12-22 by jp112sdl