jp112sdl/SonoffHMLOX #28: HomematicTouchAsSender Tastendruck nicht erkannt

Eshco93, 2017-12-23:
Hallo,

die im Wiki beschriebene Funktion "HomematicTouchAsSender" funktioniert bei mir leider nicht. Alles andere konnte ich erfolgreich testen.
Den 1. Schaltkanal meines Sonoffs habe ich "Sonoff1" genannt. Den ersten Senderkanal entsprechend "Sonoff1:1".
Die HomeMatic erkennt die Betätigung des Tasters leider nicht.

Im Syslog von CUxD erhalte ich bei Betätigung des Tasters folgende Meldung:

Dec 23 22:46:36 homematic-ccu2 local0.err ReGaHss: Error: IseESP::ScriptRuntimeError: ret=dom.GetObject(".PRESS_LONG").State(true) [iseESPexec.cpp:12447]

jp112sdl, 2017-12-24:
Hallo,

hast du nach dem Benennen des CUxD Tasterkanals den Sonoff Touch noch mal neu gestartet?
Die Gerätekonfiguration wird nur beim Booten des Sonoff einmalig eingelesen.

Die Ausgabe `dom.GetObject(".PRESS_LONG").State(true)` deutet darauf hin, dass beim Booten des Sonoff die Seriennummer des CUxD Tasterkanals nicht ermittelt werden konnte.

Bitte noch mal die CUxD Konfiguration überprüfen (evtl. versehentlich ein Leerzeichen hinter "Sonoff1:1"?).

Hilfreich wäre auch, mit angeschlossenem FTDI Adapter und einem seriellen Terminal (115200 Baud), die Logausgaben des Sonoffs zu beobachten.
Da muss dann die Seriennummer des Tasterkanals zu sehen sein.

[image]

P.S.: In der Wifi-Konfigurationsoberfläche des Sonoff hast du bei Gerätename auch "Sonoff1" eingetragen?

Eshco93, 2017-12-24:
Hallo,

vielen Dank für die schnelle Antwort! Es lag wohl daran, dass ich das Sonoff nach der Änderung der Konfiguration nicht noch einmal neu gestartet habe. Ich habe es jetzt nochmal getestet und nun funktioniert der Tastereingang ;-)

jp112sdl, 2017-12-24:
Perfekt! Dann ist ja das Weihnachtsfest gerettet 👍

closed 2017-12-24 by jp112sdl