jp112sdl/SonoffHMLOX #29: Anzeige des Gesamtverbrauches

gentoo79, 2017-12-27:
Moin,

Ich habe da mal eine Frage und zwar ist es möglich beim POW eine Systemvariable zu erstellen, wo der Gesamtverbrauch angezeigt wird und nicht wie jetzt die aktuelle Leistung ?

Schöner wäre es wenn man es so darstellen könnte wie beim HM-ES-PMSw1-Pl .

LG
Gentoo79

jp112sdl, 2017-12-27:
Hi @gentoo79,

eigentlich ist keine Erweiterung vorgesehen, denn du kannst den Energiezähler theoretisch auch in HomeMatic mit einem Skript ausrechnen.
Die Leistung (W) wird ja in einem konfigurierbaren

 Intervall an die CCU geschickt.
Nun brauchst du nur die Leistung kumulieren (also immer addieren) und durch die Zeiteinheit dividieren, dann hast du die Wh.

Ich habs jetzt mal im Code hinzugefügt. 
Du kannst ja mal schauen, ob die Werte plausibel sind, denn ich hab gerade wenig Zeit zum Testen.

[SonoffHMLOX.ino.generic.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/1589284/SonoffHMLOX.ino.generic.bin.zip)

gentoo79, 2017-12-28:
Moin,
ja ich habe die Firmware mal hochgeladen. Mal schauen ob es soweit passt. 
Aber als SV wird dieses jetzt noch nicht angezeigt oder ?. 
Also das mit Script schreiben hört sich gut an  (wenn man es kann). Ich warte da den mal auf eine Lösung :)

jp112sdl, 2017-12-28:
> Aber als SV wird dieses jetzt noch nicht angezeigt oder ?.

Nein, im Moment nur im Web bzw. wenn du `/getPower` oder `/getPowerJSON` aufrufst.

> Also das mit Script schreiben hört sich gut an (wenn man es kann). Ich warte da den mal auf eine Lösung :)

An einem HomeMatic Skript werd ich mich aber hier nicht weiter beteiligen.  ;)

jp112sdl, 2017-12-28:
@gentoo79 
Moin....
Die Berechnung in der ersten Version ist falsch. Teste bitte diese hier.

Es ist jetzt auch möglich, unter "Konfigurieren" eine SV einzutragen, um den Energiezählerwert an HomeMatic zu übertragen.

[SonoffHMLOX.ino.generic.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/1590781/SonoffHMLOX.ino.generic.bin.zip)

gentoo79, 2017-12-28:
Moin,

Ok danke ich werde die jetzt mal installieren und mal schauen 
Danke für deine schnelle Arbeit.

LG
Gentoo79

gentoo79, 2017-12-28:
So einen Fehler habe ich gefunden. Sofer der Zähler bei 265 wh ist schaltet er auf 0 zurück 
[image]

jp112sdl, 2017-12-28:
Hi @gentoo79,
sehe ich es richtig, dass nach der 264,51 erstmal eine größere Pause ist?
15:15:32 -> 15:16:46
Da ist ja 1:15min. lang gar keine Übertragung.
Es sieht also eher so aus, als würde der Sonoff abstürzen und neustarten... Ich schau es mir an.

jp112sdl, 2017-12-29:
@gentoo79 
Ich kann den Fehler leider nicht reproduzieren...
Hab bei 337.0 Wh jetzt abgebrochen.

[image]

gentoo79, 2017-12-29:
Moin,
Also bei mir kommt es öfters vor. Habe mal mit das Diagramm mit hinzugefügt. 

[image]

jp112sdl, 2017-12-29:
Kann es sein, dass der Sonoff in bestimmten Zeitintervallen - alle 4h - durchbootet?

jp112sdl, 2017-12-29:
Nicht ganz 4h... auch mal nach 5 oder schon nach 2... Sehr sehr seltsam... Da lässt sich ja keine Regelmäßigkeit ableiten.

Weder nach einer konkreten Zeit noch nach einen konkreten Wert.

jp112sdl, 2017-12-30:
@gentoo79 
Mir fällt jetzt eigentlich nur noch eine Möglichkeit ein: Die WLAN Verbindung bricht ab.
Wenn beim setStateCUxD() die WLAN Verbindung nicht mehr verbunden ist, mache ich einen Neu-Verbindungsversuch. Wenn dieser fehlschlägt, erfolgt ein Reboot des Gerätes.

jenshavelberg, 2017-12-30:
wird der reboot irgendwo geloggt/protokolliert? hätte was, wenn man in der webausgabe sehen würde, wann der letzte, vom system erfolgte, gewünschte reboot war. nur mit datetime hat das ganze system bisher nix zu tun, oder?

jp112sdl, 2017-12-30:
Nein, wird nix geloggt (außer seriell).
Mit Uhrzeit hat der Sonoff auch nix am Hut.

jp112sdl, 2017-12-30:
@gentoo79 
Logging per UDP an einen Syslog-Server kann ich noch anbieten. 
Das hab ich für meine Entwicklungsumgebung hier schon implementiert. Müsstest mir nur die IP deines Syslog-Servers nennen, dann kommt die mit in ne Custom Firmware hier.

gentoo79, 2017-12-30:
meine ip wäre 192.168.178.21 port: 514

jp112sdl, 2017-12-30:
Ok, dann versuchs mal mit der hier:
[SonoffHMLOX.ino.generic.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/1594491/SonoffHMLOX.ino.generic.bin.zip)

gentoo79, 2017-12-30:
So ich teste mal

gentoo79, 2017-12-30:
Leider kommt es hier wirklich zu einem reboot
[image]

jp112sdl, 2017-12-30:
Hmm... Ja, wenn die WLAN Verbindung tatsächlich abbricht, stehts auch nicht im Syslog. Wie sollte auch der Syslog-Server noch erreicht werden. Es spricht also alles dafür.

Ich bin dann auch mit meinem Latein am Ende. Bei mir läuft das Ding durch.

Gelegenheitsbastler, 2017-12-31:
Das ist jetzt hier etwas OT. Aber ich gehe mal davon aus, dass es ein Hardwareproblem ist. Meine These: Wenn der ESP das Wlan einschaltet, kommt es zu Stromspitzen. Wenn diese zu stark für die verbauten C sind, dann bricht die Spannung so weit ein, dass der ESP neu bootet. 

Aus der Ferne ist eine Diagnose natürlich schwierig, daher kann ich nur allgemeine Tipps geben. Wenn Du die Möglichkeit hast, dann verbaue möglichst nahe am Vcc Pin des ESP einen 100nF KerKo und einen "fetten" ElKo von z. B. 470µF. Wenn Du nur 100µF da hast, ist das für einen Test auch erst mal okay. Du kannst auch mehrere parallel schalten, wenn es der Platz zulässt. Da am ESP nur 3V3 anliegen, dürfte bezüglich der Spannung nahezu jeder ElKo passen.

jp112sdl, 2017-12-31:
@Gelegenheitsbastler 
> Wenn der ESP das Wlan einschaltet, kommt es zu Stromspitzen

WLAN wird ja einmalig beim Starten des Geräts eingeschaltet und bleibt dann dauerhaft an

Gelegenheitsbastler, 2017-12-31:
Ist die Übertragung fortwährend oder erfolgt sie in Intervallen?
Falls es Intervalle sind: Ist es anhand des Logs möglich, zu prüfen, ob der Reset dann erfolgt, wenn gemäß Intervall gesendet werden sollte?

Aber wie auch immer. Einen Versuch wäre es wert, oder nicht?

gentoo79, 2017-12-31:
ich weiss nicht ob das wirklich weiter hilft. Aber das steht an Fehler drinnen. Habe auch nochmal die syslog angehängt. 
[syslog-Sonoff.log](https://github.com/jp112sdl/SonoffHMLOX/files/1595091/syslog-Sonoff.log)

```
Search "HTTP Error Code -1" (24 hits in 1 file)
  \\WDMYCLOUD\Public\Homematic\syslog\syslog-Sonoff.log (24 hits)
	Line 940: Dec 30 17:20:59 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Strom failed with HTTP Error Code -1
	Line 943: Dec 30 17:21:05 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Gesamtverbrauch failed with HTTP Error Code -1
	Line 2542: Dec 30 19:15:32 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Strom failed with HTTP Error Code -1
	Line 3521: Dec 30 20:25:02 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Strom failed with HTTP Error Code -1
	Line 5392: Dec 30 22:40:32 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Strom failed with HTTP Error Code -1
	Line 6583: Dec 31 00:06:53 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Strom failed with HTTP Error Code -1
	Line 6586: Dec 31 00:06:59 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Gesamtverbrauch failed with HTTP Error Code -1
	Line 9605: Dec 31 03:44:23 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Strom failed with HTTP Error Code -1
	Line 10123: Dec 31 04:21:24 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Strom failed with HTTP Error Code -1
	Line 10126: Dec 31 04:21:30 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Gesamtverbrauch failed with HTTP Error Code -1
	Line 10693: Dec 31 05:02:24 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Strom failed with HTTP Error Code -1
	Line 10696: Dec 31 05:02:30 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Gesamtverbrauch failed with HTTP Error Code -1
	Line 10847: Dec 31 05:13:24 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Strom failed with HTTP Error Code -1
	Line 10850: Dec 31 05:13:30 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Gesamtverbrauch failed with HTTP Error Code -1
	Line 11141: Dec 31 05:34:24 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Strom failed with HTTP Error Code -1
	Line 11144: Dec 31 05:34:30 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Gesamtverbrauch failed with HTTP Error Code -1
	Line 11148: Dec 31 05:34:54 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Strom failed with HTTP Error Code -1
	Line 11151: Dec 31 05:35:00 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Gesamtverbrauch failed with HTTP Error Code -1
	Line 11372: Dec 31 05:50:54 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Strom failed with HTTP Error Code -1
	Line 11375: Dec 31 05:51:00 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Gesamtverbrauch failed with HTTP Error Code -1
	Line 13275: Dec 31 08:06:56 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Strom failed with HTTP Error Code -1
	Line 13278: Dec 31 08:07:02 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Gesamtverbrauch failed with HTTP Error Code -1
	Line 14571: Dec 31 09:35:26 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Strom failed with HTTP Error Code -1
	Line 14574: Dec 31 09:35:32 Sonoff Heizung1 setStateCUxD() HTTP Heizung%20Gesamtverbrauch failed with HTTP Error Code -1
```
Komisch finde ich nur das aber nur 6x ein reboot gemacht wurden ist. zur Zeit läuft das ganze.
```
Search "Boot" (6 hits in 1 file)
  \\WDMYCLOUD\Public\Homematic\syslog\syslog-Sonoff.log (6 hits)
	Line 1: Dec 30 16:17:52 Sonoff Heizung1 Setup Sonoff Heizung1 - Boot abgeschlossen, SSID = FRITZ!Box 7490, IP = 192.168.178.15, RSSI = -65, MAC = 60:01:94:5E:14:63
	Line 3973: Dec 30 20:57:41 Sonoff Heizung1 Setup Sonoff Heizung1 - Boot abgeschlossen, SSID = FRITZ!Box 7490, IP = 192.168.178.15, RSSI = -66, MAC = 60:01:94:5E:14:63
	Line 5130: Dec 30 22:21:26 Sonoff Heizung1 Setup Sonoff Heizung1 - Boot abgeschlossen, SSID = FRITZ!Box 7490, IP = 192.168.178.15, RSSI = -69, MAC = 60:01:94:5E:14:63
	Line 5395: Dec 30 22:40:46 Sonoff Heizung1 Setup Sonoff Heizung1 - Boot abgeschlossen, SSID = FRITZ!Box 7490, IP = 192.168.178.15, RSSI = -69, MAC = 60:01:94:5E:14:63
	Line 8609: Dec 31 02:32:03 Sonoff Heizung1 Setup Sonoff Heizung1 - Boot abgeschlossen, SSID = FRITZ!Box 7490, IP = 192.168.178.15, RSSI = -69, MAC = 60:01:94:5E:14:63
	Line 9088: Dec 31 03:06:47 Sonoff Heizung1 Setup Sonoff Heizung1 - Boot abgeschlossen, SSID = FRITZ!Box 7490, IP = 192.168.178.15, RSSI = -67, MAC = 60:01:94:5E:14:63

```

jp112sdl, 2017-12-31:
HTTP Error Code -1 (`HTTPC_ERROR_CONNECTION_REFUSED`, siehe auch [hier](https://github.com/esp8266/Arduino/issues/1368#issuecomment-171408375)) ist schon mal ein Hinweis, dass was im Netzwerk faul ist.

jp112sdl, 2017-12-31:
Ich habe mal `http.setReuse(true);` gesetzt.

Teste bitte mal mit dieser Firmware
[SonoffHMLOX.ino.generic.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/1595110/SonoffHMLOX.ino.generic.bin.zip)

gentoo79, 2017-12-31:
Jup ich teste mal 

hier nochmal ein log vom WLAN . WLAN an anderen Geräten läuft hier aber optimal ohne Ausfälle .

> 31.12.17
> 12:29:30
> WLAN-Gerät angemeldet (2,4 GHz), 72 Mbit/s, Heizung, IP 192.168.178.15, MAC 60:01:94:5E:14:63.
> 31.12.17
> 12:29:30
> WLAN-Gerät hat sich abgemeldet (2,4 GHz), Heizung, IP 192.168.178.15, MAC 60:01:94:5E:14:63.
> 31.12.17
> 12:29:24
> WLAN-Gerät angemeldet (2,4 GHz), 72 Mbit/s, Heizung, IP 192.168.178.15, MAC 60:01:94:5E:14:63.
> 31.12.17
> 12:29:10
> WLAN-Gerät hat sich abgemeldet (2,4 GHz), Heizung, IP 192.168.178.15, MAC 60:01:94:5E:14:63.
> 31.12.17
> 12:27:02
> WLAN-Gerät angemeldet (2,4 GHz), 72 Mbit/s, Heizung, IP 192.168.178.15, MAC 60:01:94:5E:14:63.
> 31.12.17
> 12:27:02
> WLAN-Gerät hat sich abgemeldet (2,4 GHz), Heizung, IP 192.168.178.15, MAC 60:01:94:5E:14:63.
> 31.12.17
> 12:25:42
> WLAN-Gerät angemeldet (2,4 GHz), 72 Mbit/s, Heizung, IP 192.168.178.15, MAC 60:01:94:5E:14:63.
> 31.12.17
> 12:25:42
> WLAN-Gerät hat sich abgemeldet (2,4 GHz), Heizung, IP 192.168.178.15, MAC 60:01:94:5E:14:63.
> 31.12.17
> 12:25:34
> WLAN-Gerät angemeldet (2,4 GHz), 72 Mbit/s, Heizung, IP 192.168.178.15, MAC 60:01:94:5E:14:63.
> 31.12.17
> 12:25:34
> WLAN-Gerät hat sich abgemeldet (2,4 GHz), Heizung, IP 192.168.178.15, MAC 60:01:94:5E:14:63.
> 31.12.17
> 12:25:30
> WLAN-Gerät angemeldet (2,4 GHz), 72 Mbit/s, Heizung, IP 192.168.178.15, MAC 60:01:94:5E:14:63.
> 31.12.17
> 12:25:30
> WLAN-Gerät hat sich abgemeldet (2,4 GHz), Heizung, IP 192.168.178.15, MAC 60:01:94:5E:14:63.

jp112sdl, 2017-12-31:
Hmm, ich kenn das Log von den Fritten nicht...
@jenshavelberg meldet sich dein Sonoff auch so oft im WLAN ab-/an?

jenshavelberg, 2017-12-31:
Hab bei meiner Fritte ja das WLAN aus, weil ich damit überhaupt nicht zufrieden war. Hab auf Unifi AP-LC Access Points umgestellt. In dem log sehe ich, dass die Verbindung dauerhaft besteht und nicht abgebaut wird.

gentoo79, 2017-12-31:
ok dann werde ich mal gucken warum der Sonoff sich abmeldet. 
Aber danke erstmal für eure Hilfe.  

Ich wünschen Euch einen guten Rutsch ins neue Jahr.

jp112sdl, 2017-12-31:
@gentoo79 Wir werden die Quelle des Übels schon noch irgendwann finden :) 

Dir auch einen guten Rutsch! 🎆

jp112sdl, 2017-12-31:
Kurze Zwischenmeldung: 
Ich konnte den Fehler rekonstruieren und den Error-Stack seriell abgreifen!

```Decoding 23 results
0x402443ad: tcp_pcb_purge at core/tcp.c line 1446
0x402446c7: tcp_pcb_remove at core/tcp.c line 1468 (discriminator 1)
0x402448a4: tcp_abandon at core/tcp.c line 372
0x40244902: tcp_abort at core/tcp.c line 406
0x4020137a: delay at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/cores/esp8266/core_esp8266_wiring.c line 54
0x40211984: ClientContext::abort() at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/libraries/ESP8266WiFi/src/WiFiClient.cpp line 341
0x40211d02: ClientContext::connect(ip_addr*, unsigned short) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/libraries/ESP8266WiFi/src/WiFiClient.cpp line 341
:  (inlined by) WiFiClient::connect(IPAddress, unsigned short) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/libraries/ESP8266WiFi/src/WiFiClient.cpp line 135
0x402117f8: WiFiClient::connect(char const*, unsigned short) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/libraries/ESP8266WiFi/src/WiFiClient.cpp line 341
0x40212a07: TransportTraits::create() at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 51
0x40212e94: HTTPClient::connect() at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 51
0x4021ab27: String::changeBuffer(unsigned int) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/cores/esp8266/WString.cpp line 518
0x40213664: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 51
0x401004d8: malloc at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/cores/esp8266/umm_malloc/umm_malloc.c line 1668
0x4010020c: _umm_free at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/cores/esp8266/umm_malloc/umm_malloc.c line 1291
0x40213712: HTTPClient::GET() at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 51
0x4020b25c: setStateCUxD(String, String) at /Users/user/Documents/Arduino/ECHT-Einsatz/InUse/SonoffHMLOX/HomeMatic.ino line 11
0x4021ae42: String::String(float, unsigned char) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/cores/esp8266/WString.cpp line 518
0x4021abf2: String::String(char const*) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/cores/esp8266/WString.cpp line 518
0x4020b73a: handleHLW8012() at /Users/user/Documents/Arduino/ECHT-Einsatz/InUse/SonoffHMLOX/HLW8012.ino line 122
0x4010068c: free at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/cores/esp8266/umm_malloc/umm_malloc.c line 1737
0x4020e674: loop at /Users/user/Documents/Arduino/ECHT-Einsatz/InUse/SonoffHMLOX/SonoffHMLOX.ino line 456 (discriminator 10)
0x4021b870: loop_wrapper at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/cores/esp8266/core_esp8266_main.cpp line 57
0x4010070c: cont_norm at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc2/cores/esp8266/cont.S line 109
```

Was meiner Meinung nach passiert:
Der HTTP-Client will seinen GET loswerden: `HTTPClient::sendRequest` und eine Verbindung zum Ziel aufbauen `HTTPClient::connect()`.
Die WLAN-Verbindung ist (aus welchem Grund auch immer) nicht mehr verfügbar, so dass sie erneut aufgebaut wird: `WiFiClient::connect`
Der Verbindungsversuch wird abgebrochen `ClientContext::abort()` und das Programm läuft in die Exception.

jp112sdl, 2018-01-01:
@gentoo79 
Bitte testen! (P.S.: Ich UDP Debugging zu deinem Syslog Server ist da erstmal wieder raus)
[SonoffHMLOX.ino.generic.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/1595805/SonoffHMLOX.ino.generic.bin.zip)

gentoo79, 2018-01-01:
Erstmal Frohes neues Jahr. Ich habe jetzt mal installiert. Mal schauen

jp112sdl, 2018-01-01:
Stimmt... Frohes neues Jahr! :)

jp112sdl, 2018-01-01:
Ich hab mal Ticket [#4060](https://github.com/esp8266/Arduino/issues/4060) zu dem Problem erstellt.

Habe den Code, wie dort vorgeschlagen, jetzt mit der neuesten SDK direkt ausm Git kompiliert.

Kann ich dich noch mal nötigen, diese Firmware einzuspielen? :) 
Vielen Dank für deine Geduld! 

[SonoffHMLOX.ino.generic.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/1596193/SonoffHMLOX.ino.generic.bin.zip)

gentoo79, 2018-01-01:
So habe ich installiert. Mal schaun :) die davor lief übrigens bisher auch durch

gentoo79, 2018-01-01:
ähm jetzt kommt garkein connect zu stande

jp112sdl, 2018-01-01:
Kannst du bitte noch mal in den WiFi Config Modus gehen und die WLAN Daten neu eintragen?

jp112sdl, 2018-01-01:
Ist irgendwie voll nervig... Bei mir ging es auch erst nachm 3. Mal oder so.

Die .bin davor war mit der SDK Version 2.3.0 kompiliert. Ich konnte mir schon sowas denken... dass es damit geht. Die hat aber die WPA2 KRACK Schwachstelle nicht gefixt. Das wurde erst in der 2.4.0-rc2 gemacht, die nach wie vor noch nicht stable ist.

Nun habe ich den latest Code ausm Git genommen

jp112sdl, 2018-01-01:
Oder mach den Sonoff mal komplett stromlos.

Ich hab jetzt auf meinen noch mal die .bin per OTA geflasht. WiFi wird verbunden.

jp112sdl, 2018-01-01:
Ahhhh! 
Asche auf mein Haupt!!! 
Hab dir die falsche Datei geschickt.
Hier jetzt aber :) 

[SonoffHMLOX.ino.generic.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/1596208/SonoffHMLOX.ino.generic.bin.zip)

gentoo79, 2018-01-01:
hmm ich komme nicht in den Konfigurationsmodus :(. Muss ich den jetzt komplett neu flashen (ausbauen) ?

jp112sdl, 2018-01-01:
Leider ja. Ist es mit viel Aufwand verbunden?

Ich hatte mir bei meinem damals extra die 4 Pins an eine Buchsenleiste nach außen gelegt, um im Notfall schnell ranzukommen zum Flashen...

[image]

gentoo79, 2018-01-01:
nö nicht viel aufwand, da er bei mir an der Heizung dran ist. Muss nur gleich erstmal meine Tochter ins bett bringen :). Solange gibt es halt keine Heizung :P

gentoo79, 2018-01-01:
So habe mal schnell Kabel angelötet, damit das demnächst schneller geht und Software ist drauf . Dann will ich mal sehen ob er die Nacht durchhält

gentoo79, 2018-01-01:
Ach so bei dir startet er auch neu ? Hätte gedacht das es nur bei mir ist .

jp112sdl, 2018-01-01:
Den Reboot hab ich eingebaut, wenn der WiFi-Status nicht mehr `WL_CONNECTED` ist und ein Wiederverbindungsversuch fehlschlägt. 
Sonst würde zwar der Sonoff weiterlaufen, ab nie wieder im WLAN erreichbar sein - bis er halt neu gestartet wird.

jenshavelberg, 2018-01-01:
Kann man nicht den letzten kumulierten Strom Wert immer persistieren und so nach einem eventuellen Neustart auf den alten Wert weiter draufrechnen? (abgesehen davon, dass so oder so eine Lücke bei der Stromverbrauchsmessung durch den Neustart entsteht)

jp112sdl, 2018-01-01:
Die Schreibzyklen des Flash Speichers sind begrenzt. Man müsste einen Kompromiss finden, zwischen Genauigkeit und Lebensdauer... Entweder ein größerer Gap im Energiezählerwert bei nem unerwarteten Reboot oder eine kürzere Lebensdauer des Flashs.
Ziel ist auf jeden Fall, die Reboots zu verhindern.
Ich hoffe, dass sich da demnächst was im SDK tut... Hab meinen Issue [#4060](https://github.com/esp8266/Arduino/issues/4060) dort noch mal ergänzt, nachdem ich nun mit der allerneusten ESP8266-Core Version kompiliert habe.
Wenn ich in den Issues mal nach `bcn_timeout` suche, dann gibts da schon recht aktuelle Treffer.

jp112sdl, 2018-01-02:
@gentoo79 
Mit der Version läufts bei mir seit gestern 21:48 Uhr bis jetzt (6:10 Uhr).

[SonoffHMLOX.ino.generic.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/1596866/SonoffHMLOX.ino.generic.bin.zip)

gentoo79, 2018-01-02:
Moin,
also bisher läuft die Version von Gestern super. 
[image]

jp112sdl, 2018-01-02:
Danke für dein Feedback.
Nach intensiver Suche konnte ich noch einige Tickets mit demselben Problem finden - und auch einen Workaround, der zu 99% funktioniert.

Das restliche Prozent ist der Moment, wenn man versucht, eine HTTP-Verbindung zu einem Client aufzubauen, wenn die WLAN Verbindung unterbrochen, der `WiFi.status()` jedoch noch nicht `!= WL_CONNECTED` ist.

Hab das in einem kleinen Sketch reproduzieren können und mit an das Ticket [#4060](https://github.com/esp8266/Arduino/issues/4060) angehängt.

Die Version, die du momentan hast, hat diesen Workaround noch nicht.
Ich werde sie heute am Nachmittag hier einstellen - und dich bitten, diese dann zu testen.

Ist echt eine große Hilfe! Danke!

jp112sdl, 2018-01-02:
@gentoo79 
Bitte mal testen... :) 

Wenn die jetzt durchläuft, dann wirds die finale Version.

[SonoffHMLOX.ino.generic.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/1597907/SonoffHMLOX.ino.generic.bin.zip)

gentoo79, 2018-01-02:
so habe ich jetzt installiert. :)

gentoo79, 2018-01-02:
Bisher läuft es 
[image]

jp112sdl, 2018-01-03:
Top - Das sieht schon mal gut aus!

jp112sdl, 2018-01-03:
Lief knapp 24h, dann isser wieder abgestürzt

gentoo79, 2018-01-03:
Also noch läuft er bei mir. Haben ja auch noch 2 Std. Ca 
[image]

gentoo79, 2018-01-03:
So bei mir läuft es immernoch :)

[7 more replies]

closed 2018-01-05 by jp112sdl