jp112sdl/SonoffHMLOX #30: Sicherheit und Status

dirk99, 2017-12-27:
Super Sache, diese Software. Ich habe noch 2Punkte zur Verbesserung.
1. Security Nur Befehle von der HM akzeptieren, sobald die HM konfiguriert ist.
2. immer den Status an die HM übertragen, auch wenn der Status über die Taste verändert wird oder der alte Status nach Stromausfall wieder Hergestellt wird.

Ich hoffe die Vorschläge finden Anklang.

Viele Grüße
Dirk

FiegeU, 2017-12-28:
Moinsen,

@jp112sdl : Du hattest gescheieben:
>> ist keine Bedienung (auch nicht am Taster) möglich, da der µC kein Multitasking kann
Könnte man das nicht über einen IRQ im Eingang lösen ? Dann könntest Du den An-Befehl loswerden und gleichzeitig doch auf die HM warten.
Nur so meine 2 Cents...

LG
Uli

P.S. Was ich heut zufällig entdeckt hatte, war, dass man auf dem Sonoff auch eine MicroPython FW ziehen kann. Wenn ich mal ein bisschen Luft habe, mache ich mal einen Prototypen um mal zu testen, ob unter Python auch diese Zeitverzögerung manchmal auftritt, oder hat das schon jemand gemacht ?

jp112sdl, 2017-12-28:
@FiegeU :
> Könnte man das nicht über einen IRQ im Eingang lösen ? Dann könntest Du den An-Befehl loswerden und gleichzeitig doch auf die HM warten.

Jap. Dann hab ich den Befehl im Puffer (in einer Variable) und kann aber trotzdem erst dann schalten, wenn das Warten auf HM beendet ist.

> ob unter Python auch diese Zeitverzögerung manchmal auftritt
Meinst du mit Zeitverzögerung das Warten auf einen HTTP-Timeout? 
Ich denke ja... denn der Prozessor kann nur Single-Threading. Ganz gleich was auf ihm läuft.

Aber ich lasse mich gern auch vom Gegenteil überzeugen!

FiegeU, 2017-12-31:
Moinsen,

klappt leider nicht scheitere schon beim flashen ...
@jp112sdl :
Warum kann man denn nicht sofort schalten? Vielleicht denken wir auch in veschiedene Richtungen...

Die Zeitverzögerung: Ja, dieser Bug, den wir mit dem Pingen lösen wollten....

Bei mir ist es übrigens zwischendurch wieder zu Einschaltproblemen gekommen. So langsam habe ich den Eindruck, dass es mit der Temperatur zusammenhängt. Kann aber Quatsch sein. Schlechter war es an dem Tag, als es so kalt war. Heute ist es ja überraschend warm, da trat der Fehler auch gar nicht auf, bisher zumindest....

jp112sdl, 2017-12-31:
> Warum kann man denn nicht sofort schalten? Vielleicht denken wir auch in veschiedene Richtungen...

Man kann sofort schalten. Dann geht der Sonoff an und wartet auf die Rückmeldung von der CCU. Die ist noch nicht wieder hochgefahren. Also stehste da und wartest auf den HTTP Timeout. 
Währenddessen ist keine Bedienung möglich.

jp112sdl, 2017-12-31:
Trotz aller Ideen und Vorschläge, handelt es sich ja hier ursprünglich "nur" um die Veröffentlichung eines privaten Projekts, an dem ich bis auf ein paar Web-/JS-Funktionen (Danke @jenshavelberg), komplett allein programmiere.

Jeder darf den Code gern forken, Änderungsvorschläge und Pull Requests einreichen.

Vielleicht baut ja jemand die ursprünglich vom Issue-Verfasser (der sich hier leider noch nicht wieder zu Wort gemeldet hat) genannten Features noch ein :)

ZXBnu, 2018-01-16:
Hi jp112sdl,

sorry, daß ich mich hier ranhänge, PN hab ich nicht gefunden.

Beim Einrichten der SonoffHMLOX übernimmt der Sonoff nicht die von mir eingetragene feste IP.

Deshalb von mir Anfängerfragen:

Mit welcher IDE / Compiler kann ich die Dateien verarbeiten?

Mit Atom konnte ich nur Tasmota editieren und kompilieren, nicht den SonoffHMLOX Code. Erst mäkelte er wg PlatformIO, nachdem ich das gelöst hatte, ging builden nicht. Alles ggf meine Unfähigkeit.

Mit der Arduino IDE kam ich ggf auch wg. persönlichem Unwissen nicht zurecht. Deshalb möchte ich sicher sein, bevor ich mehr try and error Zeit aufwende: Welches ist die Plattform, in der ich den Code öffne und bearbeiten kann?

Wo in den Code-Dateien kann ich die feste IP und ggf die SSID vor dem kompilieren editieren?

Falls ich hier das Thema sprenge bitte Hinweis, wie ich Dich dazu fragen kann.

Danke und Grüße

jp112sdl, 2018-01-17:
Moin @ZXBnu ,

das von dir geschilderte Problem konnte von mir grundsätzlich erstmal nicht nachvollzogen werden.
Was mir jedoch beim Testen aufgefallen war, ist das Fehlen des Eingabefeldes für das "Gateway".
Arbeitest du mit mehreren Subnetzen, so dass das evtl. schon die Ursache war?
Ich habe es soeben in #35 gefixt (Release ab >= [1.0.19](https://github.com/jp112sdl/SonoffHMLOX/releases/latest)).

Mit Atom/PlatformIO habe ich noch nicht gearbeitet. Es mag viele tolle Features bieten (zu Methoden springen, Syntax Highlighting), aber ich habe mich an den guten Arduino IDE Compiler gewöhnt. :) 

In der Arduino IDE öffnest du irgendeine der vielen .ino-Dateien. 
Es werden dann alle weiteren Files in Tabs mit geöffnet.

Die statische IP wird (wie alle anderen Parameter auch), in der Konfigurationsdatei im "Dateisystem" abgelegt. 
Um einen fixen Wert zu nutzen, müsstest du die IP nach Zeile [240](https://github.com/jp112sdl/SonoffHMLOX/blob/master/SonoffHMLOX.ino#L240) setzen.
Um eine fixe SSID zu nutzen, müsstest du sie in der WifiManager.ino ganz oben eintragen (Zeile 1+2).

Wenn du den Code selbst kompilieren willst, gibt es noch einige wichtige Dinge zu beachten:
1. Du benötigst die [Arduino/esp8266 Core](https://github.com/esp8266/Arduino) Unterstützung in der Arduino IDE. Und zwar **nicht das Release**, da sind ein paar schlimme Bugs drin! Sondern den letzten Stand direkt aus dem Github Repo. 
2. Du benötigst die ArduinoJSON Library
3. Du benötigst die HLW8012 Library (oder du entfernst die Code-Teile für den Sonoff POW, wenn du sie nicht benötigst)
4. Du kompilierst den Sketch als Board "Generic ESP8285 Module": Flash:1M, 64k SPIFFS, lwip v2 MSS=536, crystal frequency 26MHZ, CPU frequency 80MHz

Ich glaub, für nen Anfänger alles nicht so einfach!

Teste bitte erstmal die neueste Firmware Version, vielleicht klappt es ja dann schon.
Sehr hilfreich wäre sonst auch, wenn du die seriellen Debug-Ausgaben (mit angeschlossenem FTDI Adapter und einem seriellen Terminal mit 115200 Baud) mal anhängen könntest.

Viel Erfolg!

ZXBnu, 2018-01-17:
Nabend @jp112sdl,

erstmal wirklich vielen Dank für Deine ausführliche Antwort.

Das Gatewayfeld fehlte bei IMHO der .15 bis .17, ich hatte am Ende die .14 genommen.

Ich hab bisher keine Subnetze, 192.168.1.1. - 254, /24, aber bis zu 5 WLANs gleichzeitig je nach Ort im Haus.

Mit dem Sonoff Thema hab ich gerade erst angefangen, hatte es bei Herrn Back in der Sendung gesehen und dann nie Zeit.

Zu allem Elend war noch mein erster FDTI Adapter teil?- defekt, er lief nur an nem XP Rechner, nicht an mehreren Win10. Jetzt egal, aber es ging schon mal  blöd los. Mein Hintergrund ist Dipl.-Ing. IT vor 20 Jahren, dann lange berufsfremd unterwegs, mit Arduino und Raspi vor 3 Jahren rumgespielt, aber eher betreutes Coden, mehr löten...

Stand der Dinge:

Egal welche FW ich auf dem Sonoff Switch habe, ich verbinde mein Smartphone (weil dort jedes WLAN ne andere IP Ragen haben kann, beom Lappi müßte ich immer umstellen) mit dem Sonoff WLAN, gebe alle sein incl der IP für die CCU2 und den festen IP Daten unten, IP 192.168.1.83, Maske255.55.255.0, Gateway 192.168.1.7, das ist mein KD Router, n anderen hab ich auch (1.1) , soll aber über den 1.7 gehen.

Dann speichere ich, auch nach Neustart des Sonoff seh ich dessen eigenes WLAN, kein Zugriff über die SSID und die IP, die ich eingetragen hatte, nur über die 4.1.

OK, dann hab ich widerwillig DHCP auf einer Fritz aktiviert, dann über 4.1. onfiguriert außer die drei Felder unten, dann verbindet er sich mit der eingetragenen SSID und der Fritz und ich hab Zugriff auf die SonoffHMLOX-eigene Maske.

Mit Eintrag der festen IP hab ich gefühlte 20x versucht zu speichern, kein Erfolg.

Frage: müßte nach dem speichern nicht das Sonoff-eigene WLAN verschwinden oder 'schmeißt' er das trotzdem in die Umgebung?

Ich versteh das so, daß ich im sonoffeigenen WLAN unter ...4.1 in der Maske die neuen Daten hinterlege, dann speichere, ab sofort connected er zur hinterlegten SSID und ist unter 4.1 nicht mehr ansprechbar. Falsch?

BTW: Ist der Resetbutton in der Maske totalreset incl aller Daten oder nur Reset mit Daten aus der Maske bleiben?

Mein anderes Thema: Wenn ich it der Arduino IDE eine xyz.ino Datei öffne, öffnen sich die anderen *.ino nicht in Tabs mit, sondern er sagt mir, "die Datei xyz.ino muss sich in einem Sketch-Ordner "xyz" befinden und ob er den Ordner anlegen und sie dorthin verscheiben" soll und dann wird fortgefahren. Das soll ich bestätigen. Hab ich da irgendwas falsch konfiguiert? Hab aber noch nicht viel gesucht dazu.

Oder muß ich ein Projekt erstellen? Hatte schon ab und zu Begriffsschwierigkeiten mit solchen IDEs, DOS Oberfläche und C oder TP waren für mein schlichtes Gemüt besser :-)

Wenn ich das laufen habe, solte ich die festen IPs für mich konfigurieren können, dann brauche ich die Einrichtung über die Maske nicht mehr bzw. muß da nicht mehr zwingend suchen (ich hab hier sicher n Einzelproblem)

Danke und Grüße

jp112sdl, 2018-01-17:
Nabend!

> Zu allem Elend war noch mein erster FDTI Adapter teil?- defekt, er lief nur an nem XP Rechner, nicht an mehreren Win10. Jetzt egal, aber es ging schon mal blöd los.

Meistens ist nicht der FTDI defekt. Bei 98% aller Fälle, bei denen das Flashen von einem Windows-PC/Notebook nicht funktioniert hat, war ein zu schwacher USB Port Ursache. Mal klappte es, mal nicht, mal brache es mittendrin ab.
Wurde ein aktiver USB-Hub dazwischen geklappt, ging es immer. Zuverlässig.
Oder es wurde eine dünn-drahtige USB Verlängerung benutzt, bei der die Spannung dann auch gern schnell absackt, wenn man etwas mehr Last fordert.
Ich arbeite ausschließlich unter Mac. Da hatte ich das Phänomen noch nie beobachten können.

> Dann speichere ich, auch nach Neustart des Sonoff seh ich dessen eigenes WLAN, kein Zugriff über die SSID und die IP, die ich eingetragen hatte, nur über die 4.1.

Das bedeutet, dass der Sonoff aus dem WiFi-Konfigurationsmodus nicht raus kommt. Und das passiert nur, wenn mit deinen eingegebenen WLAN Konfigurationsdaten (SSID / Passwort) keine Verbindung zu deinem AP hergestellt werden kann

>BTW: Ist der Resetbutton in der Maske totalreset incl aller Daten oder nur Reset mit Daten aus der Maske bleiben?

Da wird nix gelöscht, es wird nur der Sonoff neu gestartet.

>Frage: müßte nach dem speichern nicht das Sonoff-eigene WLAN verschwinden oder 'schmeißt' er das trotzdem in die Umgebung?

Ja, das verschwindet dann. Je nach Häufigkeit der (Hintergrund-)Suche nach WLANs in der Umgebung, mal früher, mal später. Hängt vom Betriebsystem ab. 

Aber warum es immer wieder Probleme gibt, gerade mit der AP-Modus-Endlosschleife lässt sich nur genau herausfinden, wenn ich die seriellen Debugmeldungen sehe. Alles andere ist Glaskugelmodus.

> Wenn ich it der Arduino IDE eine xyz.ino Datei öffne, öffnen sich die anderen *.ino nicht in Tabs mit, sondern er sagt mir, "die Datei xyz.ino muss sich in einem Sketch-Ordner "xyz" befinden und ob [...]

Ja, alle Dateien müssen sich im Ordner befinden, der genau so heißt, wie die Main-.ino des Projekts. In diesem Fall also `SonoffHMLOX`.

>Wenn ich das laufen habe, solte ich die festen IPs für mich konfigurieren können, dann brauche ich die Einrichtung über die Maske nicht mehr bzw. muß da nicht mehr zwingend suchen (ich hab hier sicher n Einzelproblem)

Ich habe bei mir im DHCP-Server für die Sonoff-MAC-Adressen extra Reservierungen angelegt, so dass diese immer dieselben IPs zugewiesen bekommen und ich habe **eine** zentrale Stelle, an denen ich meine festen IPs pflege. Halte ich für die sinnvollste Variante, sofern WLAN-AP und DHCP-Server von ein und demselben Gerät bereitgestellt werden.

Viel Erfolg dann noch weiterhin!

ZXBnu, 2018-01-18:
Danke für Deine Ratschläge.

Jetzt konnte ich alles in der Arduino IDE öffnen, hätte das ohne Deinen Hinweis nicht hinbekommen.

Ich schaue es mir in Ruhe an, wenn ich mehr Zeit habe.

Hab jetzt auch verstanden, wann das Sonoff WLAN verschwindet. Komischerweise funktionierte es mit einem Switch und Deiner .19 sofort, mit dem anderen nicht. Hab sicher irgendwas falsch gemacht und schau in Ruhe.

Auch habe ich am Ende des Flashens in der Konsole Meldungen gesehen, die ich sicher irgendwie auch selbst aufrufen kann, muß nur schauen wie ich mich verbinde, hab ssh lange nicht benutzt und irgendwas war da auch mit putty.

Der sich weigernde FTDI hing an einem Aktiv-(USB3)-HUB mit nem normalen Kabel und allerdings auch direkt an zwei verschiednene Laptops. Kann da hinkommen mit zu wenig Strom.

Aber die Fehlermeldung in der Konsole war immer die gleiche, daß "eine andere Anwendung den COM Port belegt". Aber das ist jetzt egal, ich hab FTDIs die funktionieren und ich such das nicht weiter.

Danek nochmal, Grüße

ZXBnu, 2018-01-19:
NAbend nochmal @jp112sdl ,

das konnte ich mit putty sehen s.u..

Dazu ist zu sagen, daß ich bereits mit unterschiedlichen FWs x-mal konfiguruert habe, SSID Jonathat9, das WLAN merkt er sich. Auch die 1.221 für die CCU2 IP.IP, Maske, Gateway bleiben 0

Kann es sein daß beim FW flashen die Parameter wie SSID und CCU2 IP stehen bleiben? 
 
Jetzt habe ich testweise mal die IP für die CCU auf 1.223 gesetzt , das übernimmt er nicht.

Nachdem er den AP mode in der Konsole anzeigt tut sich nichts nach dem speichern. Erst nach aus/ein seh ich wieder Einträge.

Jute Nacht, bis demnächst

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v074402e2
~ld

   Sonoff 5C:CF:7F:68:BE:42 startet...
Config-Modus durch bootConfigMode aktivieren?
-> bootConfigModeFilename mounted file system
-> bootcfg.mod existiert NICHT
Config-Modus mit Taster aktivieren?
Config-Modus nicht aktiviert.
loadSystemConfig mounting FS...
loadSystemConfig mounted file system
loadSystemConfig reading config file
loadSystemConfig opened config file
Content of JSON Config-File: /config.json
{"ip":"0.0.0.0","netmask":"0.0.0.0","gw":"0.0.0.0","ccuip":"192.168.1.221","sono                         ff":"Switch4","restoreOldState":false,"ledDisabled":false,"backendtype":0,"loxud                         pport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":                         0,"gpio14mode":0,"gpio14assender":false}

JSON OK
ssid = Jonathan9, psk = **************
Connecting WLAN the classic way...
..............................
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v074402e2
~ld

   Sonoff 5C:CF:7F:68:BE:42 startet...
Config-Modus durch bootConfigMode aktivieren?
-> bootConfigModeFilename mounted file system
-> bootcfg.mod existiert NICHT
Config-Modus mit Taster aktivieren?
Config-Modus nicht aktiviert.
loadSystemConfig mounting FS...
loadSystemConfig mounted file system
loadSystemConfig reading config file
loadSystemConfig opened config file
Content of JSON Config-File: /config.json
{"ip":"0.0.0.0","netmask":"0.0.0.0","gw":"0.0.0.0","ccuip":"192.168.1.221","sono                         ff":"Switch4","restoreOldState":false,"ledDisabled":false,"backendtype":0,"loxud                         pport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":                         0,"gpio14mode":0,"gpio14assender":false}

JSON OK
ssid = Jonathan9, psk = **************
Connecti

jp112sdl, 2018-01-19:
> Kann es sein daß beim FW flashen die Parameter wie SSID und CCU2 IP stehen bleiben?

Ja,
- Die CCU2 IP wird in einer Config-Datei im SPIFFS abgelegt. Das bleibt beim Flashen erhalten.
- Die SSID/PSK werden im EEPROM gespeichert. Auch die bleiben beim Flashen erhalten.

> Jetzt habe ich testweise mal die IP für die CCU auf 1.223 gesetzt , das übernimmt er nicht.

```
Connecting WLAN the classic way...
..............................
ets Jan 8 2013,rst cause:1, boot mode:(3,7)
```
Es kommt keine WLAN-Verbindung zustande.  Hast du beim Ändern der CCU IP auf .223 auch das WLAN Passwort noch mal eingegeben? Vielleicht war das falsch? Dann wird auch die Änderung der IP nicht übernommen.

> Nachdem er den AP mode in der Konsole anzeigt tut sich nichts nach dem speichern. Erst nach aus/ein seh ich wieder Einträge.

Auch ein Indiz für falsche WLAN Zugangsdaten. Wobei nach dem Flashen per Kabel der erste Software-Reboot erst nach einem Power Off erfolgen kann. Es empfiehlt sich daher, nach dem Flashen immer mindestens 1x den Strom zu trennen.

Mehr fällt mir auch grad nicht ein, wie ich weiterhelfen könnte.

ZXBnu, 2018-01-20:
hilft mir schon gut weiter, danke.

Wie resette ich den kompletten Sonoff Switch incl. aller hinterlegter Parameter und Einträge?

Versucht habe ich nach einschalten Taster für bis zu 60 sek. und vor einschalten Taster drücken, dann ein und bis zu 60 sek. beides löschte z.B. nicht die CCU IP

jp112sdl, 2018-01-20:
> Wie resette ich den kompletten Sonoff Switch incl. aller hinterlegter Parameter und Einträge?

Gar nicht 👀 Hat bisher noch niemand nach gefragt... 😁 
Wäre wohl auch nur sinnvoll, wenn du den Sonoff mal verkaufen willst. Sonst eigentlich nicht.
Die Config wird auch beim Speichern immer komplett neu geschrieben, so dass auch keine "Fragmente" hängen bleiben können.

PaulMerk, 2018-01-22:
Hallo Jérôme,
die ArduinoJSON Library hatte ich bereits; HLW8012 habe ich hier gefunden:
https://bitbucket.org/xoseperez/hlw8012/downloads/?tab=downloads

Ich habe versucht Arduino/esp8266 Core einzubinden:
https://github.com/esp8266/Arduino
(Arduino-master.zip 20,477 MB) 
Die Arduino-IDE meldet jedoch dass diese zip-Datei keine gültige Bibliothek ist.

Wo kann man die benötigte Bibliothek bekommen (Bibliothekverwalter Arduino / Githiub ) ?
Gruß
PM

jp112sdl, 2018-01-23:
Moin,

>Die Arduino-IDE meldet jedoch dass diese zip-Datei keine gültige Bibliothek ist.

Das ist richtig, es handelt sich hierbei um ein `Board`.
Die Installation des esp8266/Arduino Core ist auf der [Startseite (README) des Repository](https://github.com/esp8266/Arduino#installing-with-boards-manager) erklärt:
- Arduino IDE Einstellungen
- Additional Boards Manager URLs (in DE steht da glaub ich "zusätzliche Boardverwalter")
- `http://arduino.esp8266.com/stable/package_esp8266com_index.json` hinzufügen
- Werkzeuge->Boards-> Boardverwalter: nach esp8266 suchen und installieren
- Werkzeuge->Boards-> **ESP8285** auswählen, **Flash 1M (64k SPIFFS)**

PaulMerk, 2018-01-24:
Hallo Jérôme,

Sonoff POW schalten mit CCU geht prima.

Leistung und Energie fehlt mir noch.
CUX-Gerätename für Leistung und Energie sind angelegt und auch in der Sonoff POW-Konfiguration eingetragen.
Werden diese Informationen zeitgesteuert vom POW übertragen oder muss man sie abrufen ?
Wie lauten ggf. die Befehle ?

Gruß
Paul

jp112sdl, 2018-01-24:
> CUX-Gerätename für Leistung und Energie sind angelegt 

Du brauchst 2 **Systemvariablen** (SV) vom Typ "Zahl" in der CCU.
Den Namen der SV trägst du in der Sonoff Config ein.
Der Name der SV muss systemweit **eindeutig** sein (und darf nicht identisch sein mit dem CUxD-Gerätenamen bzw -Kanalnamen).
Der Wert wird im Intervall, der in der Sonoff Config eingetragen ist, übertragen.

Siehe "zusätzliche Option bei Sonoff POW" in der [Konfiguration](https://github.com/jp112sdl/SonoffHMLOX/wiki/Sonoff_Konfiguration).

closed 2018-04-06 by jp112sdl