jp112sdl/SonoffHMLOX #33: Übertragung der Leistungswerte des Sonoff POW bricht unregelmäßig ab (duplicated from #29)

jp112sdl, 2018-01-05:
Unter Verwendung der aktuellen Arduino Core Version 2.4.0 kommt es zu Problemen mit dem Verbindungsaufbau nach einem WLAN Disconnect, teilweise auch schon nach dem WPA2 Re-Keying.

Bislang eingestellte Issues:
[#4060](https://github.com/esp8266/Arduino/issues/4060)
[#4078](https://github.com/esp8266/Arduino/issues/4078)
[#4091](https://github.com/esp8266/Arduino/issues/4091)
[#4095](https://github.com/esp8266/Arduino/issues/4095)

gentoo79, 2018-01-06:
Also bei mir läuft bis jetzt alles super 
[image]
[image]

jp112sdl, 2018-01-06:
@gentoo79 Ja, denn deine .bin ist noch mit LwIP 1.4 kompiliert. Da gabs den Bug noch nicht

gentoo79, 2018-01-06:
Also ist die Firmnware 1.0.15 noch die lwIP1.4 ?ich hatte das update ausgeführt, auf die neue Version

jp112sdl, 2018-01-06:
Leider ja - es tut sich aber gerade was in Sachen Bug-Behebung der lwip2

gentoo79, 2018-01-06:
ok dann warte ich noch mit dem nächsten update :)

jp112sdl, 2018-01-07:
Fixed mit PR [#4105](https://github.com/esp8266/Arduino/pull/4105)

jp112sdl, 2018-01-07:
Published [@1.0.17](https://github.com/jp112sdl/SonoffHMLOX/releases/tag/1.0.17)

closed 2018-01-07 by jp112sdl