jp112sdl/SonoffHMLOX #34: Keine Verbindung zum Webinterface ESP8266

TomT1965, 2018-01-11:
Hallo,

es gab ein Closed Issue bezogen auf den Fehler https://github.com/esp8266/Arduino/issues/2330 bei dem das Webinterface nicht erreichbar ist.

Du pingst den ESP extern regelmäßig an, damit dieser Fehler nicht bei dir auftritt.

Gelöst werden kann das Problem aber auch anders.
Im Code mittels eines Timers mit dem einfachen Befehl
`    WiFi.reconnect();
`
Aufrufen tue ich das alle 30 sek.

Give it a try :-)

Grüße
Tom

jp112sdl, 2018-01-11:
Hallo Tom, 
vielen Dank für den Hinweis.

Ein `WiFi.reconnect();` führte leider auch zu den Problemen:
[#4060](https://github.com/esp8266/Arduino/issues/4060)
[#4078](https://github.com/esp8266/Arduino/issues/4078)
[#4091](https://github.com/esp8266/Arduino/issues/4091)
[#4095](https://github.com/esp8266/Arduino/issues/4095)
[#4101](https://github.com/esp8266/Arduino/issues/4101)  
die zwar momentan in der Git-Version gefixt sind, jedoch noch nicht im Release erschienen.

Nutzt du die 2.4.0 aus dem Release oder den master ausm Git?

jp112sdl, 2018-01-11:
P.S.: Ich hatte das Problem mit dem Verbindungsverlust öfter mit einem Wemos D1. 
Seitdem ich den Code mit der latest master ausm Git und lwip2 (536) kompiliert habe, ist auch dieser dauerhaft erreichbar. 

Ich möchte ungern ein zyklisches `WiFi.reconnect();` einbauen.
Bei 1-2 Sonoffs mag das noch gehen. Ich kenne Installation von > 20 Geräten. 
Ein Reconnect erzeugt eine Menge "Schmutz". Das hat auch Auswirkungen auf andere Geräte im heimischen WLAN. 

In [#2330](https://github.com/esp8266/Arduino/issues/2330) schrieb auch jemand, der mit 60-sekündlichem Reconnect seinen ESP nach einer gewissen Zeit gegen die Wand gefahren hat.
Da finde ich deine 30 Sekunden schon echt sportlich!

Sollte es tatsächlich darauf hinaus laufen (ich verfolge den Issue schon sehr lange), dass ein `WiFi.reconnect();` die einzige Möglichkeit ist, den Sonoff am Netzwerk zu erhalten und sollten sich (bisher waren es nur 2-3 Leute) noch mehr Anwender melden, dann würde ich es einbauen. 
Aber dann auch nur zyklisch alle 30 Minuten.

TomT1965, 2018-01-11:
Ich nutze die 2.4.0-rc2 ..
Kommt von hier : https://github.com/esp8266/Arduino/releases/download/2.4.0-rc2/package_esp8266com_index.json

jp112sdl, 2018-01-11:
Ja ok. Die ist noch mega buggy. 
Ich nutze die nicht mehr.

FiegeU, 2018-01-11:
Moinsen,
ich kann direkt dazu jetzt nichts beitragen, nur am Rande:
Ich hatte jetzt mal einen Test gemacht, die MicroPython FW auf dem Sonoff Switch zu installieren.
Nach längerem Hin- und Her (musste auch erst noch ein anderes Projekt abschließen..) ist die jetzt drauf und es gibt einen Prototypen, der per WebRequest eine Lampe ein- und ausschaltet.
Ist aber noch buggy, soll ja auch nur ein Testballon werden, gebe da aber noch mal Feedback.

Falls es jemanden interessiert, das MP auf den Sonoff zu spielen hat bei mir nur vom Mac aus geklappt, Windows mochte nicht.

Auch auf dem Mac müssen die passenden Parameter mitgegeben werden, die Anleitungen im Netz passen so nicht, so klappt es aber:

Zum Löschen des Flashs:
sudo esptool.py --port /dev/cu.usbserial-A50285BI erase_flash

Zum aufspielen der FW:
sudo esptool.py --port /dev/cu.usbserial-A50285BI --baud 115200 write_flash --flash_size=detect -fm dout 0x0 ./Downloads/esp8266-20171101-v1.9.3.bin 

Zumindest bei mir hat es so geklappt.

Gruß
Uli

jp112sdl, 2018-01-12:
@FiegeU 
Beschäftigst du dich mit MP im Zusammenhang mit Sonoff nur wegen der Netzwerkstabilität oder siehst du irgendwelche anderen Vorteile gegenüber der "klassischen" Arduino Programmierung?

FiegeU, 2018-01-12:
N'abend :)
@jp112sdl 
Also ich mag Python eigentlich nicht besonders, aber die Buildchain ist viel schmaler.
Alles was ich brauche ist Notepad++, tippe den Quelltext und lade ihn hoch.
Ich muss nix kompilieren und viel wichtiger, die Laufzeitumgebung räumt hinter mir auf. (Vielleicht tut es die Arduino auch, hab da nur einmal ganz kurz drauf geguckt.) Soll heißen, keine Pointer, Objektstrukturen (Arrays und Listen) die ich kenne, komplexe Datentypen und der schon erwähnte Garbage Collector. Ich hatte das MP nur gewählt, weil es mir in meinen anderen Projekten schnell ein Ergebnis lieferte. (Bewegungsmelder auf 8266 Basis mit Temperaturüberwachung und Sensortastenfeldsteuerung. BTW, das letzte Projekt, aber auf Raspibasis, war ein Türöffner, der die Distanz zu einem Bluetoothsender misst und dann bei einer gewissen Nähe die Tür öffnet. Das war dann für den 8266 eine Nummer zu groß. Der Große hat zwar BT, wird aber von MP nicht unterstützt. Das Projekt ist gerade in der Abschlussphase, also bei 97%..)

Ich hatte auch mal mit dem uPyCraft rumgespielt, das kann gleich auf den 8266 hochladen, das ist mir aber zu buggy, das zerhackt häufiger den Quelltext. 

Sorry, bin ins schwafeln gekommen...
Also der Punkt ist, man kann es auch wie damals auf dem Z80 in Assembler machen, aber MP ist einfach einfacher :) , das Ergebnis ist schneller da.

Gruß
Uli

jp112sdl, 2018-01-13:
> Sorry, bin ins schwafeln gekommen...

Passt schon :) Brainstorming ist immer gut und bringt einen manchmal ab vom engstirnigen Denken und auf neue Idee! 
Aber speziell hier bzgl. des Sonoff-Repos nicht wirklich passend. Da könnte man sich in einem Forum o.ä. mal austauschen.

Ich mach den Issue hier erstmal zu. 👍

closed 2018-01-13 by jp112sdl

FiegeU, 2018-01-14:
Moinsen,
ich weiß, der Issue ist closed, war aber noch den Quelltext für das MP schuldig, da sind jetzt die größten Bugs raus und ich habe sie bei mir auf dem Flur am laufen (mit deaktiviertem Ping-Script). Schauen wir mal was passiert. Vielleicht mag ihn ja mal jemand begutachten, poste was passiert, hier haben wir dann alles komplett archiviert (auch für mich ;) )...

Habe die FW geflasht, WebRepl aktiviert und diese beiden Dateien hochgeladen, wobei boot.py nur das WebRepl zur Laufzeit einschaltet und die Verbindung zum Netzwerk herstellt. Der Rest passiert im main.py, ist aber auch nicht viel 
Einfach die ip/0 zum ausschalten, ip/1? zum dauerhaften einschalten (dass das  ? nötig ist, ist ein Bug, ist aber gerade erst eben aufgefallen, keine Lust auf einen schnellen fix) oder ip/1?t=5 für ein Einschalten von t Sekunden.

boot.py: 
```
# This file is running on every boot 
# (c) 2018  U. Fiege
# Licensed under the GPL v3.0 or later

import gc
import webrepl
webrepl.start()
gc.collect()

def do_connect():
    import network
    sta_if = network.WLAN(network.STA_IF)
    if not sta_if.isconnected():
        print('connecting to network...')
        sta_if.active(True)
        sta_if.connect('SSID', 'pass')
        while not sta_if.isconnected():
            pass
    print('Current network config:', sta_if.ifconfig())

```

main.py:
```
# (c) 2018  U. Fiege
# Licensed under the GPL v3.0 or later
import network
import socket
import machine
from machine import Pin
import utime  as time
import machine

pinNo = 13 # 13 = LED, 12 = Relay
pinNoRelay = 12
timerRunning = False
def log(msg):
    print(msg)

def handleInterrupt(timer):
  global timerRunning 
  pin = machine.Pin(pinNo)
  pin = machine.Pin(pinNo, machine.Pin.OUT)
  pin.value(1)
  pin = machine.Pin(pinNoRelay, machine.Pin.OUT)
  pin.value(0)
  timerRunning = False

    
def start_server():
    global timerRunning 
    pin = machine.Pin(pinNo)
    pin = machine.Pin(pinNo, machine.Pin.OUT)
    pin.value(1)
    pinR = machine.Pin(pinNoRelay, machine.Pin.OUT)
    pinR.value(0)
    
    log('start server method')
    addr = socket.getaddrinfo('0.0.0.0', 80)[0][-1]
    s = socket.socket()
    s.bind(addr)
    s.listen(1)
    log('listening on'+str( addr))
    while True:
        print (time.time())
      
        swOn = False
        t = 0
        cl, addr = s.accept()
        log('client connected from'+str(addr))
        request = str(cl.recv(1024))
        request = str(request)
        try:
          i = request.upper().find("HTTP")
          parm = request[:i]
          print ("parm:" , parm)
          if (parm.find("/1?")>=0):
            swOn = True
            i = parm.find("t=")
            if (i >= 0):
              t = int(parm[(i+2):])
              # delete any running timer 
              if (timerRunning == True):
                try:
                  timer.deinit()
                #endTry
                except Exception as err:
                  print ("Error clearing timer:", str(err))
                #endExcept   
              #endif
              timer = machine.Timer(0)
              timer.init(period=t * 1000, mode=machine.Timer.ONE_SHOT, callback=handleInterrupt)
              timerRunning = True
            #endif
          #endif
        except Exception as err:
          print ("Error:" , str(err))
        cl.send(html)
        cl.close()
        if (swOn == True):
          pin.value(0)
          pinR.value(1)          
        else:
          pin.value(1)
          pinR.value(0)
        #endif
    #endwhile
#main part

html = """<!DOCTYPE html>
<html><head> <title>ESP8266</title> </head>
    <body>Sonoff switch prototype</body>
</html>
"""
start_server()
```

Schönen Abend noch zusammen,
Gruß
Uli

FiegeU, 2018-01-27:
Moinsen,

ich weiß das Ding ist zu, wollte aber noch kurz Feedback liefern.
Nach knapp zwei Wochen ohne dem Pingscript kann ich sagen, der Sonoffswitch auf MP-Basis hat bei weitem nicht so starke Ausfälle wie die anderen beiden. Da ist höchstens mal eine Verzögerung von 1-2 Sekuden drin, wobei einer der Anderen schon mal 15 gebraucht hat.
Was mir (wieder) aufgefallen ist, wenn die Temperatur runtergeht und das Gerät tagsüber oder nachtsüber nicht angehen braucht, ist es oft so, dass der erste Switch enorm dauert....
Ich meine vorgestern ist er auch in den vollen Timeout gelaufen. (>15 Sekunden).

Ich habe mal die MP Software etwas aufgepeppt, primär, weil ich die Lampe mit dem Handy auch einschalten wollen konnte, daher hat sie jetzt Befehlsschaltflächen dafür.
Ich hänge sie mal an, dann ist sie wenigstens gleich archiviert. Sie hat natürlich nicht im entferntesten den Umfang wie Deine Version.

Viele Grüße
Uli

Code start  main.py --->
`

# (c) 2018  U. Fiege
# Changelog:
# 27.01.2018 Added web UI, ignore favicon & Apple icon requests
# Licensed under the GPL v3.0 or later
import network
import usocket as socket
import machine
from machine import Pin
import utime  as time
import machine

pinNo = 13 # 13 = LED, 12 = Relay
pinNoRelay = 12
onTime = 60 # number of seconds the web ui button should switch it on
title = "Licht Bad"
timerRunning = False
def log(msg):
    print(msg)

def handleInterrupt(timer):
  global timerRunning 
  pin = machine.Pin(pinNo)
  pin = machine.Pin(pinNo, machine.Pin.OUT)
  pin.value(1)
  pin = machine.Pin(pinNoRelay, machine.Pin.OUT)
  pin.value(0)
  timerRunning = False

    
def start_server():
    global timerRunning 
    pin = machine.Pin(pinNo)
    pin = machine.Pin(pinNo, machine.Pin.OUT)
    pin.value(1)
    pinR = machine.Pin(pinNoRelay, machine.Pin.OUT)
    pinR.value(0)
    
    log('start server method')
    addr = socket.getaddrinfo('0.0.0.0', 80)[0][-1]
    s = socket.socket()
    s.bind(addr)
    s.listen(1)
    log('listening on'+str( addr))
    while True:
        print (time.time())
      
        swOn = False
        t = 0
        cl, addr = s.accept()
        log('client connected from'+str(addr))
        request = str(cl.recv(4096))
        request = str(request)
        try:
            i = request.upper().find("HTTP")
            parm = request[:i]
            print ("parm:" , parm)           
            if ( (parm.lower().find("/favicon.ico")==-1) and (parm.lower().find("/apple-touch-icon")==-1) ):
                if (parm.find("/1?")>=0):
                    swOn = True
                    i = parm.find("t=")
                    if (i >= 0):
                        t = int(parm[(i+2):])
                        # delete any running timer 
                        if (timerRunning == True):
                            try:
                                timer.deinit()
                            #endTry
                            except Exception as err:
                                print ("Error clearing timer:", str(err))
                            #endExcept   
                        #endif
                        timer = machine.Timer(0)
                        timer.init(period=t * 1000, mode=machine.Timer.ONE_SHOT, callback=handleInterrupt)
                        timerRunning = True
                    #endif
                #endif
                if (swOn == True):
                    pin.value(0)
                    pinR.value(1)          
                elif (parm.find("/0")>=0): # switch off
                    pin.value(1)
                    pinR.value(0)
                    if (timerRunning == True):
                        try:
                            timer.deinit()
                        #endTry
                        except Exception as err:
                            print ("Error clearing timer:", str(err))
                        #endExcept   
                    #endif                    
                #endif
                if (parm.lower().find("get /") == 2): # only send an answer to GET requests
                    cl.sendall(html)               
                #endif
            else:
                print ("Ignoring favicon or touch icon request...")
            #endif
            
        except Exception as err:
            print ("Error:" , str(err))
        cl.close()
    #endwhile
#main part

html = """HTTP/1.0 200 OK

<!DOCTYPE html>
<html><head> <title>"""+ title + """</title> </head>
    <body>
        <form>
            <input style="width:100%;padding:20px;cursor:pointer;box-shadow: 6px 6px 5px; #999; -webkit-box-shadow:6px 6px 5px #999; -moz-box-shadow:6px 6px 5px #999;font-weight:bold; background: #00ff00; color: #000; border-radius: 10px; border: 1px solid #999; font-size: 550%;" type="button" value="ON (""" +  str(onTime) + """ sec )" onclick="window.location.href=location.protocol + '//' + location.host +  '/1?t=""" + str(onTime) + """'" />
            <BR><BR>
            <BR><BR>
            <input style="width:100%; padding: 20px; cursor: pointer; box-shadow: 6px 6px 5px; #999; -webkit-box-shadow: 6px 6px 5px #999; -moz-box-shadow: 6px 6px 5px #999; font-weight: bold; background: #ff0000; color: #000; border-radius: 10px; border: 1px solid #999; font-size: 550%;" type="button" value="OFF" onclick="window.location.href=location.protocol + '//' + location.host +  '/0'" />       
            <BR><BR>
            <BR><BR>
            <input style="width: 100%; padding: 20px; cursor: pointer; box-shadow: 6px 6px 5px; #999; -webkit-box-shadow: 6px 6px 5px #999; -moz-box-shadow: 6px 6px 5px #999; font-weight: bold; background: #00ff00; color: #000; border-radius: 10px; border: 1px solid #999; font-size: 550%;" type="button" value="PERMANENT-ON" onclick="window.location.href=location.protocol + '//' + location.host +  '/1?'" />
        </form>     
    </body>
</html>

"""
start_server()

`
<-- Code end

jp112sdl, 2018-01-27:
Hab den mDNS Responder rausgeworfen. Seither auch Ausfälle mehr.

FiegeU, 2018-01-27:
Auf dem Server, oder auf dem AP-Rechner  ?
Sprich, das Bonjour komplett aus dem Netz verbannt ?

Was mir noch aufgefallen ist, ist dass die Fritzbox alle paar Minuten einen Connect zum Sonoff (zumindest zu dem mit MicroPython) aufbaut:
[image]

jp112sdl, 2018-01-27:
Es arbeiten ja doch irgendwie alle nur mit IP-Adressen... 
Hab den mDNS aus dem Sonoff verbannt.

Es hat sich aber auch ein Dauer-Ping bewährt, um die Verbindung aufrecht zu erhalten.
Mittels CUxD-Ping hat es den Charme, dass man gleich noch eine Überwachung für eine "Kommunikation gestört"-Meldung hat.

Mit dem esp8266-Core direkt aus dem Git läuft es momentan sehr gut.