jp112sdl/SonoffHMLOX #36: Von Tasmota zu SonoffHMLOX per OTA?

dan11hh, 2018-01-20:
Tolles Projekt! Danke für Dein Engagement Jérôme!!!
Ich hab bisher mit Tasmota rumgespielt, auch weil ich per OTA flaschen wollte. Kann ich eigentlich die Tasmota FW Update Funktion nutzen um SonoffHMLOX einzuspielen? Möchte meinen Sonoff Basic doch gerne an die CCU2 hängen. Geht das? Oder MUSS ich per Kabel flaschen?

jp112sdl, 2018-01-20:
Hallo @dan11hh,
ich kann dir deine Frage leider nicht beantworten, da ich noch nie mit Tasmota gearbeitet habe.

Versuch es 👍 

Notfalls musst du dann jedoch mit nem Kabel ran, falls es schief läuft.

dan11hh, 2018-01-20:
Ok. Hab mich getraut. Geht aber wohl leider nicht. Er führt keinen Update Vorgang durch. Wunder mich jetzt aber auch nicht wirklich. Schade. Hab jetzt immer noch Tasmota drauf. Hmmm dann muss ich mir wohl doch ein Kabel kaufen...

jp112sdl, 2018-01-20:
Ja, dann bist du auf der sicheren Seite. Auch für künftige Geräte. Soweit ich gelesen habe, geht SonOTA nur bis zu einer Werks-Firmware Version < 1.6.

closed 2018-02-01 by jp112sdl