jp112sdl/SonoffHMLOX #38: Timer im Einschaltbefehl

elbman, 2018-01-31:
Hi,

vielen Dank für deine Arbeit und das ausführliche Wiki!

Ich möchte an meiner S20 einen Drucker schalten, der nach 5min abschalten soll

leider wird der Anschalt Befehl mit Timer in der Homematic CCU2 unter SWITCH|CMD_LONG nicht akzeptiert
/usr/local/addons/cuxd/curl -s http://192.168.2.58/1?t=300

es liegt vermutlich am Gleichheitszeichen

gibt es da einen Trick?

Gelegenheitsbastler, 2018-02-01:
Genau das gleiche Problem hatte ich auch. Daher habe ich beide Zeichen umgewandelt.
Versuch es mal mit:
/usr/local/addons/cuxd/curl -s http://192.168.2.58/1~3ft~3d300

Jerome schrieb dazu:
"Es liegt nur am = Zeichen.
Steht auch im CUxD Handbuch (Version 1.12) auf Seite 89"

jp112sdl, 2018-02-01:
@elbman:
Ja, die Vermutung ist richtig.
Siehe [CUxD Handbuch Version 1.12](https://homematic-forum.de/forum/download/file.php?id=44701&sid=18923a374529a6d828fc89a7a4acd979) auf Seite **86**

@Gelegenheitsbastler: 
Ich dachte auch, es war 89... Hab gerade noch mal nachgeschaut, Seite 86 ist richtig.

[image]

closed 2018-02-01 by jp112sdl

elbman, 2018-02-01:
Danke ;-)