jp112sdl/SonoffHMLOX #40: IP adresse not configurable

grobiballon, 2018-02-10:
I have flashed 2 Sonoff S20 devices. On first device everything worked out well. The second device hase some issues. I have set it up once. Somehow I entered a wrong IP-Adress - or the device just decided for one by himself.
If I enter the config mode again I get into the confic screen and it seems that I could change the adress.  But it is not saving this. Also an OTA reflash and making it creating his own config wifi (press button after startup) did not helped.
What are I am doing wrong. It seems the IP is burned into some frozen or write protected area on the device.
If I remember right I have tried to install Tasmato on the device affected before and then flashed it with ESP loader. Finally I now reflashed it again with ESP loader and the config is still the same as before and not reset...

Probably you could help me.

jp112sdl, 2018-02-10:
The config file is completely recreated when saving the settings in configuration mode.
Can you create a log of the debug output?
You have to use 115200 baud.

Otherwise it is hard to understand or reproduce the problem.

elbman, 2018-03-21:
Hi, wie du siehst gebe ich in der Konfiguration die IP 192.168.2.11 ein
er nimmt aber 192.168.1.62 ???

grobiballon, 2018-03-25:
Hi, ich schreibe mal nun auf Deutsch, da hier anscheined ale Deutsch verstehen...
Ich habe vorhin die neue Firmware mit der Factory Reset funktion auf den betreffenden Adapter gespielt. Wenn ich den Factory-Reset aufrufe, kommt eine Seite, dass dieser im Gange ist, aber keine Meldung, dass er fertig ist.
Ich habe dann trotzdem den Adapter nochmals neu mit fester IP konfiguriert. Leider zieht er diese hier aber wieder per DHCP.
Bin gerne bereit weitere Tests durchzuführen. Zu erwähnen ist vielleicht noch, dass ich auf diesem Adapter vorger  Tasmota laufen hatte. Alle meine anderen Adapter funktionieren einwandfrei...

grobiballon, 2018-03-25:
Update:
Ich habe gerade nochmals geflascht. Ich habe den Adapter dann - ohne ihn stromlos zu machen - konfiguriert und er hat die IP gefressen. Nach dem ich aber einmal die Versorgungsspannung getrennt habe, gab es wieder nur DHCP...

jp112sdl, 2018-03-26:
Moin.

> Wenn ich den Factory-Reset aufrufe, kommt eine Seite, dass dieser im Gange ist, aber keine Meldung, dass er fertig ist.

Ja, eine Meldung im Web kommt nicht, das sollte ich mal noch dokumentieren.
Ggf. könnte die LED am Sonoff blinken, wenn er fertig ist.
<hr>
Das mit der statischen IP bringt mich noch ins Grab. 😠 
Bei mir funktioniert es immer. 🤔 

> Nach dem ich aber einmal die Versorgungsspannung getrennt habe, gab es wieder nur DHCP...

Wenn du den Konfigurationsmodus aufrufst, steht aber die statische IP noch im Eingabefeld? Oder ist das dann auch wieder leer?

> Bin gerne bereit weitere Tests durchzuführen. 

Danke!
Kannst du mal das serielle Log vom Bootvorgang schicken?

grobiballon, 2018-03-26:
Hi, logs schicke ich dir, aber schaffe das erst kommendes Wochenende... Gruß Andreas

grobiballon, 2018-03-30:
**Direkt nach dem neuen Flashen:**

Config-Modus nicht aktiviert.
loadSystemConfig mounting FS...
loadSystemConfig mounted file system
loadSystemConfig reading config file
loadSystemConfig opened config file
Content of JSON Config-File: /config.json
{"ip":"192.168.1.202","netmask":"255.255.255.0","gw":"192.168.1.1","ccuip":"192.168.1.200","sonoff":"Sonoff S20 Nr2","restoreOldState":false,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}

JSON OK
ssid = grobi@home, psk = *************
Connecting WLAN the classic way...
Wifi Connected

WLAN erfolgreich verbunden!
SSID: grobi@home
IP Address: 192.168.1.202
Gateway Address: 192.168.1.1
signal strength (RSSI):31 dBm

Sonoff Modell = Switch / S20
getStateFromCUxD url: http://192.168.1.200:8181/cuxd.exe?ret=dom.GetObject(%22Sonoff%20S20%20Nr2%22).Address()
HTTP Sonoff%20S20%20Nr2 fail
result: 
HomeMaticConfig.ChannelName = CUxD.
Switch Relay to 0 with transmitState = 1
setStateCUxD url: http://192.168.1.200:8181/cuxd.exe?ret=dom.GetObject(%22CUxD..SET_STATE%22).State(0)
HTTP CUxD..SET_STATE failed with HTTP Error Code -1
result: 
Starte OTA-Handler...
Starte UDP-Handler an Port 6676...
Sonoff S20 Nr2 - Boot abgeschlossen, SSID = grobi@home, IP = 192.168.1.202, RSSI = -63, MAC = DC:4F:22:11:65:93

grobiballon, 2018-03-30:
**Zugriff über 192.168.1.202 möglich:**

Sending Web-Reply: {"state": 0, "timer": 0, "resttimer": 0, "fw": "1.0.22"}
Sending Web-Reply: {"state": 0, "timer": 0, "resttimer": 0, "fw": "1.0.22"}
webSwitchRelayOn(), Parameter, aber mit TimerSeconds = 0
Switch Relay to 1 with transmitState = 1
setStateCUxD url: http://192.168.1.200:8181/cuxd.exe?ret=dom.GetObject(%22CUxD..SET_STATE%22).State(1)
HTTP CUxD..SET_STATE success
result: null
Sending Web-Reply: {"state": 1, "timer": 0, "resttimer": 0, "fw": "1.0.22"}
Sending Web-Reply: {"state": 1, "timer": 0, "resttimer": 0, "fw": "1.0.22"}
Switch Relay to 0 with transmitState = 1
setStateCUxD url: http://192.168.1.200:8181/cuxd.exe?ret=dom.GetObject(%22CUxD..SET_STATE%22).State(0)
HTTP CUxD..SET_STATE success
result: null
Sending Web-Reply: {"state": 0, "timer": 0, "resttimer": 0, "fw": "1.0.22"}
Sending Web-Reply: {"state": 0, "timer": 0, "resttimer": 0, "fw": "1.0.22"}

grobiballon, 2018-03-30:
**Danach habe ich den Config Boot aufgerufen:**

setBootConfigMode mounted file system
Boot to ConfigMode requested. Restarting...

 ets Jan  8 2013,rst cause:1, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

grobiballon, 2018-03-30:
Und nun ist er nicht mehr erreichbar. IP Scanner findet ihn nicht und es gibt auch kein Sonoff Wlan. Nach einem PowerOff ist er nun auf einmal unter der 192.168.1.202 erreichbar (ging vorher nicht). Nach erneutem trennen der Spannungsversorgung ging es wieder unter der 192.168.1.202. Und jetzt kommts:
Abgesteckt  vom Programmer und ab in die Steckdose: Direkt nach dem Bootvorgang konnte ich noch ein Schaltvorgang im Webinterface deiner Firmware unter der 192.168.1.202 machen. Danach nicht mehr. Dann habe ich es unter der 192.168.1.114 versucht, und es hat funktioniert. Also hat das Teil nach dem Booten die IP Adresse gewechselt.
Ich habe gerade extra auch nochmal geguckt: im Router ist keine vordefinierte IP für die MAC Adresse des Adapters vorgegeben, und der S210 wird auch nicht als DHCP vergeben angezeigt. Somit macht der Adapter das selber...
Und jetzt kommst du ;-)

jp112sdl, 2018-03-30:
Ich werfe an der Stelle echt das Handtuch.

Hatte noch nen Bekannten gefragt, der auch ne Fritzbox hat, und alles durchspielen lassen (keine Reservierung, feste IP, DHCP IP, hin - her - zurück, Strom ran, Strom weg)... und der Sonoff verhielt sich immer wie erwartet.

Bei mir kann ich das Verhalten auch nicht reproduzieren.

Echt mega strange... Aber mir fehlt jetzt jeglicher Punkt, wo ich noch ansetzen könnte.

Trag am besten ne Reservierung im DHCP (zusätzlich) ein.

grobiballon, 2018-03-30:
Hast du einen  Plan, wie man so ein Teil komplett dumm bekommt - also heißt alles gelöscht? Ich vermute immernoch das das mit der anderen Firmware zusammenhängt die da vorher drauf war...

jp112sdl, 2018-03-30:
Mit der 1.0.22 gibts im WifiManager nen Factory Reset Button.
Der löscht Wifi-Zugangsdaten und das SPIFFS

grobiballon, 2018-04-01:
Hi, ich meinte eigentlich so richtig dumm - also alles auf der Kiste weg. Ich habe nochmals herumexperimentiert. Wenn die Kiste an der Steckdose hängt, macht er kurzzeitig die .202 und dann die .114
Wenn ich das Dingen per Programmer mit Spannung versorge immer die richtige .202

jp112sdl, 2018-04-01:
Es gibt getrennte Speicherbereiche, somit gibt es auch nicht "alles weg" Schalter.
Du kannst die Arduino IDE nehmen und einen leeren Sketch flashen. Dann wäre der Runcode-Bereich auch einmal überschrieben

grobiballon, 2018-04-02:
Moin, habe gestern nochmals Stunden in das Dingen investiert. Habe ein leeres Image drauf geflasht, mit den Speicherbereichen gespielt, zwischendurch mal Tasmota und am Ende wieder dein Image. Gleiches Problem...

jp112sdl, 2018-04-02:
Und mit Tasmota ist alles i.O.?

grobiballon, 2018-04-02:
Hatt ich natürlich nicht getestet. Gerade hab ich amer Tasmota nochmal geflasht und es funktioniert alles wie es soll...

grobiballon, 2018-04-02:
Gerade dann deine FW per OT rüber gespielt (vorher in tasmota die .202 als static IP gelassen) und das gleiche Problem.

jp112sdl, 2018-04-02:
Echt seltsam. Hab keinen Plan, wo ich da noch ansetzen könnte

jp112sdl, 2018-04-06:
Könnte hiermit zusammenhängen: esp8266/arduino#4476

jp112sdl, 2018-04-08:
Ich hab noch mal mit dem neuesten esp8266/Arduino Core kompiliert... 
Kannst ja mal probieren, ob es hiermit klappt:
[SonoffHMLOX.ino.generic.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/1887921/SonoffHMLOX.ino.generic.bin.zip)

grobiballon, 2018-04-09:
Hi, zu spät gelesen. Schaffe ich nun erst am Wochenende. Aber danke schon mal... Gruß Andreas

elbman, 2018-04-09:
Guten Morgen, habe es soeben getestet. Bei mir leider immer noch das Selbe

jp112sdl, 2018-04-09:
Ich habe das Phänomen nun 1x selbst beobachten können und hatte zum Glück auch das Debugging für WIFI eingeschaltet. Leider (bei mir) mit wenig informativem Inhalt. :/ 

@elbman hast du die Möglichkeit kurzfristig noch Tests durchzuführen?
Hier ist eine Version mit ganz viel Debugging:
[SonoffHMLOX.ino.generic.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/1888590/SonoffHMLOX.ino.generic.bin.zip)

elbman, 2018-04-09:
```
setBootConfigMode mounted file system
Boot to ConfigMode requested. Restarting...
state: 5 -> 0 (0)
                 rm 0
                     pm close 7
                               del if0
                                      usl
                                         wifi evt: 1
                                                    STA disconnect: 8

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v438d3f1d
~ld

SDK:2.2.1(cfd48f3)/Core:2.4.1-42-g438d3f1/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1)

Sonoff EC:FA:BC:13:1F:A7 startet... (FW: 1.0.22)
Config-Modus durch bootConfigMode aktivieren?
wifi evt: 2
           -> bootConfigModeFilename mounted file system
-> bootcfg.mod existiert, starte Config-Modus
loadSystemConfig mounting FS...
loadSystemConfig mounted file system
loadSystemConfig reading config file
loadSystemConfig opened config file
Content of JSON Config-File: /config.json
{"ip":"192.168.2.11","netmask":"255.255.255.0","gw":"192.168.2.1","ccuip":"192.168.2.4","sonoff":"Fire","restoreOldState":true,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}

JSON OK
ssid = Embedded-VPN, psk = *************
*WM: Adding parameter
*WM: sonoffmodel
*WM: Adding parameter
*WM: ccu
*WM: Adding parameter
*WM: loxudpport
*WM: Adding parameter
*WM: sonoff
*WM: Adding parameter
*WM: hmpowervariable_pow
*WM: Adding parameter
*WM: hmecvariable_pow
*WM: Adding parameter
*WM: custom_powermeasureinterval_pow
*WM: Adding parameter
*WM: restorestate
*WM: Adding parameter
*WM: leddisabled_switch
*WM: Adding parameter
*WM: gpio14mode_switch
*WM: Adding parameter
*WM: custom_gpio14assender_switch
*WM: Adding parameter
*WM: backendtype
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: custom_ip
*WM: Adding parameter
*WM: custom_netmask
*WM: Adding parameter
*WM: custom_gw
scandone
        del if0
[... 274 lines cut ...]
[hostByName] Host: 192.168.2.4 is a IP!
                                       result: null
HomeMaticConfig.ChannelName = CUxD.null
getLastState mounted file system
laststat.txt existiert
getLastState FileContent = 1
Switch Relay to 1 with transmitState = 1
setLastState mounted file system
setLastState (1) saved.
Fire - Boot abgeschlossen, SSID = Embedded-VPN, IP = 192.168.2.11, RSSI = -36, MAC = EC:FA:BC:13:1F:A7
ip:192.168.2.71,mask:255.255.255.0,gw:192.168.2.1
                                                 wifi evt: 3
                                                            pm open,type:0 0
                                                                            Sending Web-Reply: {"state": 1, "timer": 0, "resttimer": 0, "fw": "1.0.22"}
Sending Web-Reply: {"state": 1, "timer": 0, "resttimer": 0, "fw": "1.0.22"}
Sending Web-Reply: {"state": 1, "timer": 0, "resttimer": 0, "fw": "1.0.22"}
Sending Web-Reply: {"state": 1, "timer": 0, "resttimer": 0, "fw": "1.0.22"}
Sending Web-Reply: {"state": 1, "timer": 0, "resttimer": 0, "fw": "1.0.22"}
Sending Web-Reply: {"state": 1, "timer": 0, "resttimer": 0, "fw": "1.0.22"}
Sending Web-Reply: {"state": 1, "timer": 0, "resttimer": 0, "fw": "1.0.22"}
Sending Web-Reply: {"state": 1, "timer": 0, "resttimer": 0, "fw": "1.0.22"}
Sending Web-Reply: {"state": 1, "timer": 0, "resttimer": 0, "fw": "1.0.22"}
```

Klar, hoffe das hilft weiter

jp112sdl, 2018-04-10:
Btw.: Ihr könntet evtl. mal die [1.0.21](https://github.com/jp112sdl/SonoffHMLOX/releases/download/1.0.21/SonoffHMLOX.ino.generic.bin) oder [1.0.20](https://github.com/jp112sdl/SonoffHMLOX/releases/download/1.0.20/SonoffHMLOX.ino.generic.bin) testen. 

Ist der Bug dort auch vorhanden?

elbman, 2018-04-10:
Puhhh, hab jetzt alle Releases bis zur 17 runterprobiert. Alle haben bei mir den selben Bug. (vereinzelt kam es dazu, dass die gewünschte IP 192.168.2.11 gewählt wurde, die Freude war aber von kurzer Dauer, nach dem Neustart immer wieder die 71 via DHCP...)

jp112sdl, 2018-04-10:
Vielen Dank für deine Mühe.
Werde das Projekt dann hier wohl einstellen.
Oder die Möglichkeit der statischen IP-Konfiguration entfernen.
Ich denk die Tage mal drüber nach.

Vielen Dank nochmal!

elbman, 2018-04-11:
Hey, Kopf hoch, wegen eines kleinen Bugs dieses geniale Projekt einzustellen ist doch Blödsinn...
Ich vermute, dass es nur bei einer ganz bestimmten Serie / Hardwareversion der Adapter auftritt
(habe / hatte 6 S20, 1xTouch, 1xSwitch und nur einer hat diesen Bug)

grobiballon, 2018-04-11:
@jp112sdl Ich finde das Projekt auch cool und du hast schon viel zu viel erreicht - also bitte nicht eingestampfen. Lass uns versuchen gemeinsam das Problem zu lösen :-)
Am Sonntag have ich wahrscheinlich zeit meinen S20 zu testen. Bei mir macht auch nur einer Probleme.
@elbman: Glaubst du es würde Sinn machen mal die Innereien zu vergleichen, ob wir irgendwelche Gleichheiten oder Unterschiede zwischen den problematischen Adaptern finden (Chip-Typen, etc.)?
@jp112sdl: Glaubst du das Problem liegt alleine in dem esp8266/Arduino code? Nutzt Tasmota den Selben esp8266 code für das Wifi? Vielleicht kann man vergleichen, was die Jungs von Tasmota anders machen?
Wenn ich Dir auf den Nerv gehe, sag aber bescheid!
Gruß
Andreas

jp112sdl, 2018-04-11:
> Glaubst du das Problem liegt alleine in dem esp8266/Arduino code?

Ich würde sagen ja. 

> Glaubst du es würde Sinn machen mal die Innereien zu vergleichen, ob wir irgendwelche Gleichheiten oder Unterschiede zwischen den problematischen Adaptern finden (Chip-Typen, etc.)?

Das könnte man machen. Manchmal sind ESP8266 verbaut, manchmal ESP8285.
Der Code ist für ESP8285 kompiliert und läuft aber auch auf ESP8266. Es geht dabei hauptsächlich um die Flash-Methode DOUT.

> Nutzt Tasmota den Selben esp8266 code für das Wifi? Vielleicht kann man vergleichen, was die Jungs von Tasmota anders machen?

Das kann ich dir nicht sagen, da ich keine Zeit habe, mich mit Tasmota auseinanderzusetzen.
Es ist ein sehr umfangreicher Code, der kontinuierlich weiter entwickelt wird.

Mein Plan war ursprünglich nur, die Sonoffs irgendwie mit Rückmeldung in Homematic und für einen Freund in Loxone reinzubekommen.
Ich programmiere ganz allein an dem Ding, habe leider keine 4, 6 oder mehr Augen, die evtl. einen Fehler schnell entdecken könnten.

Irgendwann tauchte die WPA2-Krack-Schwachstelle auf, die mit Lwip2 gefixt wurde. Und seither gibts nur Trödel. 

Schon allein esp8266/Arduino#2330 ist zum k*tzen. :/ 

Derzeit rückt die ganze ESP-Geschichte in den Hintergrund, weil es mit der [pa-pa/AskSinPP](https://github.com/pa-pa/AskSinPP) Library möglich ist, mittels 328P und CC1101 Geräte zu bauen, die nativ 868MHz HomeMatic BidCos sprechen und somit auch direkt in der CCU anlernbar sind. 
Kein WLAN / LAN, kein CUxD, direktverknüpfbar...

Hab vorgestern erst einen Sonoff 4CH auf Homematic Funk umgebaut...
[image]

jp112sdl, 2018-04-13:
@elbman Ich hatte noch eine Idee.
Bitte mal testen.

[SonoffHMLOX.ino.generic.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/1906047/SonoffHMLOX.ino.generic.bin.zip)

jp112sdl, 2018-04-16:
Oh, es scheint sich was zu tun, bzgl. des "versehentlichen" DHCP trotz statischer IP!

https://github.com/d-a-v/esp82xx-nonos-linklayer/commit/c9bfc85832fd1e9ccb17e25fa9b8c9f3be6605a4

grobiballon, 2018-04-16:
Uh, mir fällt gerade ein, dass ich am Wochenende vergessen habe zu testen... Und bin schon wieder unterwegs und habe den Adapter nicht mitgenommen, wie ich es vorhatte. Sorry...

jp112sdl, 2018-04-17:
Einen hab ich noch :) 

[SonoffHMLOX.ino.generic.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/1919705/SonoffHMLOX.ino.generic.bin.zip)

elbman, 2018-04-18:
leider kein Erfolg, Verhalten unverändert

jp112sdl, 2018-04-21:
Und nur bei einem Gerät? Ich verstehs nicht...

fastsonic, 2018-04-23:
Hi, ich habe gestern auch 2 weitere POW in Betrieb genommen.
Hier habe ich das Problem das sie sich auch die feste IP nicht nehmen wollen.
Allerdings bin ich noch auf dem standart Firmware und habe die letzte hier noch nicht getestet. Werde ich heute Abend nachholen. 
Bei mir funktioniert ein POW einwandfrei und 3 haben jetzt das Problem. Teilweise kommen sie frisch resettet einmal mit der festen hoch, aber spätestens nach einem reboot sind sie wieder auf DHCP. In der Config steht aber weiterhin die Manuell hinterlegte. 
Wenn ich noch etwas zur Fehlerbereinigung beitragen kann?

jp112sdl, 2018-04-23:
Leider nicht :/ 

Am besten eine Reservierung am Router einrichten, so dass der Pow immer dieselbe IP bekommt.

fastsonic, 2018-04-23:
leider aufräumtechnisch ziemlich ungünstig... Meine Fritzbox verteilt ( standartnetz) 10.0.0.100-250; 255.255.254 Meine Automatisierung läuft aber ab 10.0.1.1 ;255.255.254. 
Ich teste erstmal deine neuere Version hier. Ansonsten muß ich mir wohl einen anderen Weg suchen.
Trotzdem Danke für das Projekt.

elbman, 2018-05-02:
Gern ;-) 
Bei Vergabe einer festen IP kann sich mein Sonoff nicht in das WLAN einbuchen, bei leeren Feldern geht es mit ner über DHCP zugewiesenen Adresse.

jp112sdl, 2018-05-02:
Das wird ja immer bunter. -.-

Hier noch mal mit anderen Optionen kompiliert:
[SonoffHMLOX.ino.generic.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/1968885/SonoffHMLOX.ino.generic.bin.zip)

jp112sdl, 2018-05-03:
@elbman Ich habe jetzt mal an einem AP getestet, ohne DHCP-Server dahinter.
Der Sonoff hat trotz konfigurierter statischer IP keine Verbindung hergestellt, weil immer noch der DHCP Client im Hintergrund hoch kommt..
Habe daraufhin noch ein bisschen was am Code geändert, so dass es bei mir zumindest jetzt ging.

Please try & test:
[SonoffHMLOX.ino.generic.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/1969696/SonoffHMLOX.ino.generic.bin.zip)

Danke für die Ausdauer! 👍

fastsonic, 2018-05-03:
Danke das du doch noch dran bist.
Ich wollte ja die Tage testen. 2 Pow mit der 1.1 sind plötzlich mit ihrer statischen Adresse online. Auch nach dem booten bleiben sie jetzt da. Sie mussten es sich scheinbar  nur ein paar Tage überlegen. :)

elbman, 2018-05-03:
YES! 

Es läuft, feste IP wird genommen!

Zum ersten mal bleibt sie auch wirklich nach dem Neustart!!!

Sehr gute Arbeit, Danke!!!!

grobiballon, 2018-05-06:
Moin, habe es nun endlich geschafft zu testen. Mein "kränkelnder" S20 behält mit der letzten hier geposteten Firmware die statische IP. Dementsprechend ist aus meiner Sicht dieses Issue gelöst!

Vielen Dank für den Einsaz!

jp112sdl, 2018-05-11:
Vielen Dank für eure Mit- und Zuarbeiten! 

Ich mach den Issue dann hier erstmal mit der [1.0.25](https://github.com/jp112sdl/SonoffHMLOX/tree/1.0.25) zu :)

closed 2018-05-11 by jp112sdl