jp112sdl/SonoffHMLOX #42: Einstellungen löschen

elbman, 2018-03-19:
Gibt es eine Möglichkeit alle Einstellungen zu löschen? Reset auf "Werkseinstellungen"

Wenn ich versuche zu resetten (bootConfigMode oder neu flashe) werden die Einstellungen immer behalten. Eine Weitergabe z.B. an einen Freund ist so nicht möglich

Ein weiteres Problem, einer meiner Sonoff S20 beharrt hartnäckig auf einer festen IP (192.168.2.68), egal was ich in der Konfiguration eigegebe die Einstellung wird nicht übernommen, sie scheint wie eingebrannt.

Einzige Lösung die ich gefunden habe ist den Sonoff mit einem leeren Image (z.B. Blank1MB.bin) flashen, dann die SonoffHMLOX.bin 

geht das evtl. auch einfacher?

jp112sdl, 2018-03-20:
> Wenn ich versuche zu resetten (bootConfigMode oder neu flashe) werden die Einstellungen immer behalten. Eine Weitergabe z.B. an einen Freund ist so nicht möglich

Warum ist eine Weitergabe dann nicht möglich?
Derjenige braucht nur bei Inbetriebnahme den Knopf zu drücken und die Konfiguration durchzuführen.

> Ein weiteres Problem, einer meiner Sonoff S20 beharrt hartnäckig auf einer festen IP (192.168.2.68), egal was ich in der Konfiguration eigegebe die Einstellung wird nicht übernommen, sie scheint wie eingebrannt.

Hast du schon mal das Feld leer gelassen (oder 0.0.0.0) eingetragen, gespeichert und neu gestartet?
Was steht dann drin?

_Sofern das Problem mit der 1.0.21 besteht, bitte mal mit [Version 1.0.20](https://github.com/jp112sdl/SonoffHMLOX/releases/download/1.0.20/SonoffHMLOX.ino.generic.bin) probieren._

> geht das evtl. auch einfacher?

Der Reset? Oder das Setzen der festen IP?

elbman, 2018-03-20:
Moin

> Warum ist eine Weitergabe dann nicht möglich?

Bei mir wählt er sich der Adapter halt direkt mit den noch gespeicherten, also mit den ohne weiteres nicht löschbaren Zugangsdaten, direkt ins Wlan ein. Daher würde ich mir sowas wie einen Parameter "bootConfigModeClean" wünschen ;-), der auch den Bereich löscht, indem die Config Daten geschrieben werden. Oder ein Image das diesen Bereich mit überschreibt
(beim initialen Flashen mit ESPLoader sieht man in der Console auch dass das Passwort und die Wlan SSID bereits mit Werten gefüllt sind. Offenbar Datenmüll vom StockImage)

> Hast du schon mal das Feld leer gelassen (oder 0.0.0.0) eingetragen, gespeichert und neu gestartet?
> Was steht dann drin?

das habe ich natürlich probiert, leider ohne Erfolg. Die einmal fest vergebene IP lässt sich nicht mehr ändern.
Ich habe diesen Adapter direkt mit der 1.0.21 bestückt. Bei der 1.0.20 (andere Adapter) habe ich bisher nicht das Verlangen gehabt die feste IP zu ändern, ich gehe aber davon aus, dass das Problem auch dort 
unter gewissen Umständen besteht (denke hier hat jemand das selbe Problem, wohl noch mit der 1.0.20 gehabt https://github.com/jp112sdl/SonoffHMLOX/issues/40)

> geht das evtl. auch einfacher?

Damit war natürlich nur das Setzen auf Werkseinstellungen gemeint ;-)

jp112sdl, 2018-03-20:
Ich kanns irgendwie nicht nachvollziehen...
Hatte vor dem Release der 1.0.21 genau das noch mal getestet: feste IP setzen und wieder löschen.
Das klappte auch wunderbar. Natürlich nur, wenn auch die WLAN-Zugangsdaten korrekt sind.
Kann beim Speichern die Verbindung zum Router nicht hergestellt werden (z.B. falsches Passwort eingegeben), wird auch die Config nicht geschrieben.

Die WLAN-Zugangsdaten liegen in einem anderen Speicherbereich als die Config.
Das sind 2 paar Schuhe... Und ja - selbst der frisch ausgelieferte Original-Sonoff versucht erstmal, sich mit dem Werks-WLAN zu verbinden. 
Aber das ist doch eigentlich auch nicht weiter schlimm!?

Falls du die Möglichkeit hast, schau dir das serielle Log mal an.
Beim Starten und auch beim Speichern der Config wird der Inhalt der config.json ausgegeben und auch die feste IP.

Ich teste es die Tage sonst auch noch mal selbst.

jp112sdl, 2018-03-20:
Ich kann es leider nicht nachvollziehen.

Habe eben mal meinem S20 eine statische IP verpasst und wieder gelöscht und wieder gesetzt... geht sofort. Mit der 1.0.21.

[image]

Hier noch der komplette Auszug:
```
e system
loadSystemConfig reading config file
loadSystemConfig opened config file
Content of JSON Config-File: /config.json
{"ip":"0.0.0.0","netmask":"0.0.0.0","gw":"0.0.0.0","ccuip":"192.168.1.253","sonoff":"S20_2","restoreOldState":false,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}

JSON OK
ssid = Apple Network, psk = *************
AP-Modus ist aktiv!
Should save config
Wifi Connected
CUSTOM STATIC IP: 0.0.0.0 Netmask: 0.0.0.0 GW: 0.0.0.0
Custom IP Address is set!
saving config
{"ip":"192.168.1.244","netmask":"255.255.255.0","gw":"192.168.1.1","ccuip":"192.168.1.253","sonoff":"S20_2","restoreOldState":false,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3ce888e8
~ld

Sonoff EC:FA:BC:12:E9:B0 startet...
Config-Modus durch bootConfigMode aktivieren? 
-> bootConfigModeFilename mounted file system
-> bootcfg.mod existiert NICHT
Config-Modus mit Taster aktivieren?
Config-Modus nicht aktiviert.
loadSystemConfig mounting FS...
loadSystemConfig mounted file system
loadSystemConfig reading config file
loadSystemConfig opened config file
Content of JSON Config-File: /config.json
{"ip":"192.168.1.244","netmask":"255.255.255.0","gw":"192.168.1.1","ccuip":"192.168.1.253","sonoff":"S20_2","restoreOldState":false,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}

JSON OK
ssid = Apple Network, psk = *************
Connecting WLAN the classic way...
......Wifi Connected

WLAN erfolgreich verbunden!
SSID: Apple Network
IP Address: 192.168.1.244
Gateway Address: 192.168.1.1
signal strength (RSSI):-74 dBm

Sonoff Modell = Switch / S20
getStateFromCUxD url: http://192.168.1.253:8181/cuxd.exe?ret=dom.GetObject(%22S20_2%22).Address()
result: null
HomeMaticConfig.ChannelName = CUxD.null
Switch Relay to 0 with transmitState = 1
Starte OTA-Handler...
Starte UDP-Handler an Port 6676...
S20_2 - Boot abgeschlossen, SSID = Apple Network, IP = 192.168.1.244, RSSI = -76, MAC = EC:FA:BC:12:E9:B0
Should save config
Wifi Connected
CUSTOM STATIC IP: 192.168.1.244 Netmask: 255.255.255.0 GW: 192.168.1.1
saving config
{"ip":"0.0.0.0","netmask":"0.0.0.0","gw":"0.0.0.0","ccuip":"192.168.1.253","sonoff":"S20_2","restoreOldState":false,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}
[... 165 lines cut ...]
loadSystemConfig reading config file
loadSystemConfig opened config file
Content of JSON Config-File: /config.json
{"ip":"0.0.0.0","netmask":"0.0.0.0","gw":"0.0.0.0","ccuip":"192.168.1.253","sonoff":"S20_2","restoreOldState":false,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}

JSON OK
ssid = Apple Network, psk = *************
Connecting WLAN the classic way...
......Wifi Connected

WLAN erfolgreich verbunden!
SSID: Apple Network
IP Address: 192.168.1.32
Gateway Address: 192.168.1.1
signal strength (RSSI):-73 dBm

Sonoff Modell = Switch / S20
getStateFromCUxD url: http://192.168.1.253:8181/cuxd.exe?ret=dom.GetObject(%22S20_2%22).Address()
result: null
HomeMaticConfig.ChannelName = CUxD.null
Switch Relay to 0 with transmitState = 1
Starte OTA-Handler...
Starte UDP-Handler an Port 6676...
S20_2 - Boot abgeschlossen, SSID = Apple Network, IP = 192.168.1.32, RSSI = -73, MAC = EC:FA:BC:12:E9:B0
```

elbman, 2018-03-21:
Boot to ConfigMode requested. Restarting...

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3ce888e8
~ld

   Sonoff EC:FA:BC:13:1F:A7 startet...
Config-Modus durch bootConfigMode aktivieren?
-> bootConfigModeFilename mounted file system
-> bootcfg.mod existiert, starte Config-Modus
loadSystemConfig mounting FS...
loadSystemConfig mounted file system
loadSystemConfig reading config file
loadSystemConfig opened config file
Content of JSON Config-File: /config.json
{"ip":"0.0.0.0","netmask":"0.0.0.0","gw":"0.0.0.0","ccuip":"192.168.2.4","sonoff                    ":"Feuer","restoreOldState":true,"ledDisabled":false,"backendtype":0,"loxudpport                    ":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gp                    io14mode":0,"gpio14assender":false}

JSON OK
ssid = Embedded-VPN, psk = *************
AP-Modus ist aktiv!
Should save config
Wifi Connected
CUSTOM STATIC IP: 0.0.0.0 Netmask: 0.0.0.0 GW: 0.0.0.0
Custom IP Address is set!
saving config
{"ip":"192.168.2.11","netmask":"255.255.255.0","gw":"192.168.2.1","ccuip":"192.168.2.4","sonoff":"Feuer","restoreOldState":true,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3ce888e8
~ld

   Sonoff EC:FA:BC:13:1F:A7 startet...
Config-Modus durch bootConfigMode aktivieren?
-> bootConfigModeFilename mounted file system
-> bootcfg.mod existiert NICHT
Config-Modus mit Taster aktivieren?
Config-Modus nicht aktiviert.
loadSystemConfig mounting FS...
loadSystemConfig mounted file system
loadSystemConfig reading config file
loadSystemConfig opened config file
Content of JSON Config-File: /config.json
{"ip":"192.168.2.11","netmask":"255.255.255.0","gw":"192.168.2.1","ccuip":"192.168.2.4","sonoff":"Feuer","restoreOldState":true,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}

JSON OK
ssid = Embedded-VPN, psk = *************
Connecting WLAN the classic way...
.Wifi Connected

WLAN erfolgreich verbunden!
SSID: Embedded-VPN
IP Address: 192.168.2.62
Gateway Address: 192.168.2.1
signal strength (RSSI):-33 dBm

Sonoff Modell = Switch / S20
getStateFromCUxD url: http://192.168.2.4:8181/cuxd.exe?ret=dom.GetObject(%22Feuer%22).Address()
result: null
HomeMaticConfig.ChannelName = CUxD.null
getLastState mounted file system
laststat.txt existiert
getLastState FileContent = 0
Switch Relay to 0 with transmitState = 1
setLastState mounted file system
setLastState (0) saved.
Starte OTA-Handler...
Starte UDP-Handler an Port 6676...
Feuer - Boot abgeschlossen, SSID = Embedded-VPN, IP = 192.168.2.62, RSSI = -34, MAC = EC:FA:BC:13:1F:A7

elbman, 2018-03-21:
Hi, wie du siehst gebe ich in der Konfiguration die IP 192.168.2.11 ein
er nimmt aber 192.168.2.62 ??? 
es ist im übrigen egal, was ich ihm vorgebe. Ohne Vorgabe für DHCP gibt es auch die 192.168.2.62, die Konfiguration scheint irgendwie festgesetzt

jp112sdl, 2018-03-21:
Hmm, ja, das sehe ich.
Erschreckend... Ich hab nicht mal die geringste Idee, wo ich ansetzen könnte.
Warum gehts bei mir, aber bei dir nicht? 🤐 

Kannst du es bitte mal mit angehängter Firmware testen?

[SonoffHMLOX.ino.generic.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/1835328/SonoffHMLOX.ino.generic.bin.zip)

elbman, 2018-03-22:
Config-Modus nicht aktiviert.
loadSystemConfig mounting FS...
loadSystemConfig mounted file system
loadSystemConfig reading config file
loadSystemConfig opened config file
Content of JSON Config-File: /config.json
{"ip":"192.168.2.11","netmask":"255.255.255.0","gw":"192.168.2.1","ccuip":"192.168.2.4","sonoff":"Feuer","restoreOldState":true,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}

JSON OK
ssid = Embedded-VPN, psk = *************
Connecting WLAN the classic way...
.Wifi Connected

WLAN erfolgreich verbunden!
SSID: Embedded-VPN
IP Address: 192.168.2.62
Gateway Address: 192.168.2.1
signal strength (RSSI):-32 dBm

Sonoff Modell = Switch / S20
getStateFromCUxD url: http://192.168.2.4:8181/cuxd.exe?ret=dom.GetObject(%22Feuer%22).Address()
result: null
HomeMaticConfig.ChannelName = CUxD.null
getLastState mounted file system
laststat.txt existiert
getLastState FileContent = 0
Switch Relay to 0 with transmitState = 1
setLastState mounted file system
setLastState (0) saved.
Starte OTA-Handler...
Starte UDP-Handler an Port 6676...
Feuer - Boot abgeschlossen, SSID = Embedded-VPN, IP = 192.168.2.62, RSSI = -32, MAC = EC:FA:BC:13:1F:A7

elbman, 2018-03-22:
Should save config
Wifi Connected
CUSTOM STATIC IP: 192.168.2.11 Netmask: 255.255.255.0 GW: 192.168.2.1
Custom IP Address is set!
saving config
{"ip":"192.168.2.15","netmask":"255.255.255.0","gw":"192.168.2.1","ccuip":"192.168.2.4","sonoff":"Feuer","restoreOldState":true,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v6d3f0b40
~ld

   Sonoff EC:FA:BC:13:1F:A7 startet...
Config-Modus durch bootConfigMode aktivieren?
-> bootConfigModeFilename mounted file system
-> bootcfg.mod existiert NICHT
Config-Modus mit Taster aktivieren?
Config-Modus nicht aktiviert.
loadSystemConfig mounting FS...
loadSystemConfig mounted file system
loadSystemConfig reading config file
loadSystemConfig opened config file
Content of JSON Config-File: /config.json
{"ip":"192.168.2.15","netmask":"255.255.255.0","gw":"192.168.2.1","ccuip":"192.168.2.4","sonoff":"Feuer","restoreOldState":true,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}

JSON OK
ssid = Embedded-VPN, psk = *************
Connecting WLAN the classic way...
.Wifi Connected

WLAN erfolgreich verbunden!
SSID: Embedded-VPN
IP Address: 192.168.2.62
Gateway Address: 192.168.2.1
signal strength (RSSI):-36 dBm

Sonoff Modell = Switch / S20
getStateFromCUxD url: http://192.168.2.4:8181/cuxd.exe?ret=dom.GetObject(%22Feuer%22).Address()
result: null
HomeMaticConfig.ChannelName = CUxD.null
getLastState mounted file system
laststat.txt existiert
getLastState FileContent = 0
Switch Relay to 0 with transmitState = 1
setLastState mounted file system
setLastState (0) saved.
Starte OTA-Handler...
Starte UDP-Handler an Port 6676...
Feuer - Boot abgeschlossen, SSID = Embedded-VPN, IP = 192.168.2.62, RSSI = -36, MAC = EC:FA:BC:13:1F:A7

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v6d3f0b40
~ld

   Sonoff EC:FA:BC:13:1F:A7 startet...
Config-Modus durch bootConfigMode aktivieren?
-> bootConfigModeFilename mounted file system
-> bootcfg.mod existiert NICHT
Config-Modus mit Taster aktivieren?
Config-Modus nicht aktiviert.
loadSystemConfig mounting FS...
loadSystemConfig mounted file system
loadSystemConfig reading config file
loadSystemConfig opened config file
Content of JSON Config-File: /config.json
{"ip":"192.168.2.15","netmask":"255.255.255.0","gw":"192.168.2.1","ccuip":"192.168.2.4","sonoff":"Feuer","restoreOldState":true,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}

JSON OK
ssid = Embedded-VPN, psk = *************
Connecting WLAN the classic way...
.Wifi Connected

WLAN erfolgreich verbunden!
SSID: Embedded-VPN
IP Address: 192.168.2.15
Gateway Address: 192.168.2.1
signal strength (RSSI):-42 dBm

Sonoff Modell = Switch / S20
getStateFromCUxD url: http://192.168.2.4:8181/cuxd.exe?ret=dom.GetObject(%22Feuer%22).Address()
result: null
HomeMaticConfig.ChannelName = CUxD.null
getLastState mounted file system
laststat.txt existiert
getLastState FileContent = 0
Switch Relay to 0 with transmitState = 1
setLastState mounted file system
setLastState (0) saved.
Starte OTA-Handler...
Starte UDP-Handler an Port 6676...
Feuer - Boot abgeschlossen, SSID = Embedded-VPN, IP = 192.168.2.15, RSSI = -41, MAC = EC:FA:BC:13:1F:A7

elbman, 2018-03-22:
Sonoff EC:FA:BC:13:1F:A7 startet...
Config-Modus durch bootConfigMode aktivieren?
-> bootConfigModeFilename mounted file system
-> bootcfg.mod existiert, starte Config-Modus
loadSystemConfig mounting FS...
loadSystemConfig mounted file system
loadSystemConfig reading config file
loadSystemConfig opened config file
Content of JSON Config-File: /config.json
{"ip":"192.168.2.15","netmask":"255.255.255.0","gw":"192.168.2.1","ccuip":"192.1           68.2.4","sonoff":"Feuer","restoreOldState":true,"ledDisabled":false,"backendtype           ":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"son           offmodel":0,"gpio14mode":0,"gpio14assender":false}

JSON OK
ssid = Embedded-VPN, psk = *************
AP-Modus ist aktiv!
Should save config
Wifi Connected
CUSTOM STATIC IP: 192.168.2.15 Netmask: 255.255.255.0 GW: 192.168.2.1
Custom IP Address is set!
saving config
{"ip":"192.168.2.11","netmask":"255.255.255.0","gw":"192.168.2.1","ccuip":"192.168.2.4","sonoff":"Feuer","restoreOldState":true,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v6d3f0b40
~ld

   Sonoff EC:FA:BC:13:1F:A7 startet...
Config-Modus durch bootConfigMode aktivieren?
-> bootConfigModeFilename mounted file system
-> bootcfg.mod existiert NICHT
Config-Modus mit Taster aktivieren?
Config-Modus nicht aktiviert.
loadSystemConfig mounting FS...
loadSystemConfig mounted file system
loadSystemConfig reading config file
loadSystemConfig opened config file
Content of JSON Config-File: /config.json
{"ip":"192.168.2.11","netmask":"255.255.255.0","gw":"192.168.2.1","ccuip":"192.168.2.4","sonoff":"Feuer","restoreOldState":true,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}

JSON OK
ssid = Embedded-VPN, psk = *************
Connecting WLAN the classic way...
.Wifi Connected

WLAN erfolgreich verbunden!
SSID: Embedded-VPN
IP Address: 192.168.2.62
Gateway Address: 192.168.2.1
signal strength (RSSI):-35 dBm

Sonoff Modell = Switch / S20
getStateFromCUxD url: http://192.168.2.4:8181/cuxd.exe?ret=dom.GetObject(%22Feuer%22).Address()
result: null
HomeMaticConfig.ChannelName = CUxD.null
getLastState mounted file system
laststat.txt existiert
getLastState FileContent = 0
Switch Relay to 0 with transmitState = 1
setLastState mounted file system
setLastState (0) saved.
Starte OTA-Handler...
Starte UDP-Handler an Port 6676...
Feuer - Boot abgeschlossen, SSID = Embedded-VPN, IP = 192.168.2.62, RSSI = -35, MAC = EC:FA:BC:13:1F:A7
Sending Web-Reply: {"state": 0, "timer": 0, "resttimer": 0, "fw": "1.0.21"}
Sending Web-Reply: {"state": 0, "timer": 0, "resttimer": 0, "fw": "1.0.21"}

elbman, 2018-03-22:
Guten Morgen,

habe es heute morgen mit der neuen Firmware getestet, anfangs das gleiche Verhalten, dann hat es offenbar mal kurz geklappt (192.168.2.15), danach aber wieder nicht... 
(sorry für die langen Postings oben)

Was mir allerdings grad überraschend aufgefallen ist (ich hoffe es hilft dem Phänomen auf den Grund zu gehen / vielleicht ist das auch so normal): 

Wenn ich den Sonoff in den Config Modus schicke ist dieser immer noch mit dem Wlan verbunden und auch erreichbar (Das Webinterface des Konfigmodus ist auch unter 192.168.2.62 zu erreichen, ohne das ich mich in das dann offene Netz des AP-Modus einwähle)

spawn21de, 2018-03-22:
Moin zusammen, ich hatte auch das Problem mit der IP Adresse, nach Wechsel von Fritzbox zu Connect Box.
Fritzbox hatte die feste statische IP im Gerät immer genommen (ohne Einstellung  an der Fritzbox) die Connect Box nicht. Musste der Connect Box mit der Mac adresse ebenfalls die feste IP vergeben, erst dann hat der Sonoff auch immer die feste IP Adresse bekommen. Vieleicht hilft der Tipp.... Gutes gelingen

jp112sdl, 2018-03-22:
> Wenn ich den Sonoff in den Config Modus schicke ist dieser immer noch mit dem Wlan verbunden und auch erreichbar (Das Webinterface des Konfigmodus ist auch unter 192.168.2.62 zu erreichen, ohne das ich mich in das dann offene Netz des AP-Modus einwähle)

Ja, das ist normal, weil die WiFiManager Lib in den AP_STA Mode geht - also im AP und STA arbeitet. Nur so kann "live" geprüft werden, ob die eingegebenen Wifi Credentials korrekt sind.

>Moin zusammen, ich hatte auch das Problem mit der IP Adresse, nach Wechsel von Fritzbox zu Connect Box.
Fritzbox hatte die feste statische IP im Gerät immer genommen (ohne Einstellung an der Fritzbox) die Connect Box nicht. Musste der Connect Box mit der Mac adresse ebenfalls die feste IP vergeben, erst dann hat der Sonoff auch immer die feste IP Adresse bekommen. Vieleicht hilft der Tipp.... Gutes gelingen

Vielleicht hängt es tatsächlich damit zusammen. Du bist der Erste, der dieses Phänomen beobachtet hat. Und ich glaube, 95% der User nutzen eine FritzBox, mit der es geht.
Ich selbst nutze eine Apple TimeMachine und habe zum Testen einen alten Linksys WAP54G genommen. Da geht es auch.
Andererseits ist es auch seltsam, weshalb nun der AP ausschlaggebend sein sollte. 
Bei statischer IP ist ja nach dem "WLAN-Handshake" dann ohne DHCP alles Client-Sache.

In den Issues der esp8266/arduino Core Lib, auf die das ganze WiFi-Zeugs baut, habe ich auch ein paar Tickets gefunden, aber auch alles recht unspezifisch.

Ich hab nun noch mal mit Lwip 1.4 kompiliert... bitte mal testen.

[SonoffHMLOX.ino.generic.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/1836328/SonoffHMLOX.ino.generic.bin.zip)

spawn21de, 2018-03-22:
War bei mir auch nur zufällig wegen Anbieterwechsel ;-).

elbman, 2018-03-22:
Nutze die Fritzbox 3490 & 7580 mit aktueller Firmware, glaube weniger dass es daran liegt. 

> ich hab nun noch mal mit Lwip 1.4 kompiliert... bitte mal testen.
> 
> SonoffHMLOX.ino.generic.bin.zip
> 
 
mach ich heute Abend und melde mich dann

elbman, 2018-03-22:
```

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v6d3f0b40
~ld

   Sonoff EC:FA:BC:13:1F:A7 startet...
Config-Modus durch bootConfigMode aktivieren?
-> bootConfigModeFilename mounted file system
-> bootcfg.mod existiert NICHT
Config-Modus mit Taster aktivieren?
Config-Modus nicht aktiviert.
loadSystemConfig mounting FS...
loadSystemConfig mounted file system
loadSystemConfig reading config file
loadSystemConfig opened config file
Content of JSON Config-File: /config.json
{"ip":"192.168.2.16","netmask":"255.255.255.0","gw":"192.168.2.1","ccuip":"192.168.2.4","sonoff":"Feuer","restoreOldState":true,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}

JSON OK
ssid = Embedded-VPN, psk = *************
Connecting WLAN the classic way...
..............................
 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v6d3f0b40
~ld

   Sonoff EC:FA:BC:13:1F:A7 startet...
Config-Modus durch bootConfigMode aktivieren?
-> bootConfigModeFilename mounted file system
-> bootcfg.mod existiert NICHT
Config-Modus mit Taster aktivieren?
Config-Modus nicht aktiviert.
loadSystemConfig mounting FS...
loadSystemConfig mounted file system
loadSystemConfig reading config file
loadSystemConfig opened config file
Content of JSON Config-File: /config.json
{"ip":"192.168.2.16","netmask":"255.255.255.0","gw":"192.168.2.1","ccuip":"192.168.2.4","sonoff":"Feuer","restoreOldState":true,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}

JSON OK
ssid = Embedded-VPN, psk = *************
Connecting WLAN the classic way...
..............................
 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v6d3f0b40
~ld

   Sonoff EC:FA:BC:13:1F:A7 startet...
Config-Modus durch bootConfigMode aktivieren?
-> bootConfigModeFilename mounted file system
-> bootcfg.mod existiert NICHT
Config-Modus mit Taster aktivieren?
Config-Modus nicht aktiviert.
loadSystemConfig mounting FS...
loadSystemConfig mounted file system
loadSystemConfig reading config file
loadSystemConfig opened config file
Content of JSON Config-File: /config.json
{"ip":"192.168.2.16","netmask":"255.255.255.0","gw":"192.168.2.1","ccuip":"192.168.2.4","sonoff":"Feuer","restoreOldState":true,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}

JSON OK
ssid = Embedded-VPN, psk = *************
Connecting WLAN the classic way...
....................

```

leider connected er mit dieser firmware gar nicht

jp112sdl, 2018-03-22:
vorletzter Versuch...

[SonoffHMLOX.ino.generic.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/1838811/SonoffHMLOX.ino.generic.bin.zip)

elbman, 2018-03-22:
```
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v42f824b2
~ld

   Sonoff EC:FA:BC:13:1F:A7 startet...
Config-Modus durch bootConfigMode aktivieren?
-> bootConfigModeFilename mounted file system
-> bootcfg.mod existiert, starte Config-Modus
loadSystemConfig mounting FS...
loadSystemConfig mounted file system
loadSystemConfig reading config file
loadSystemConfig opened config file
Content of JSON Config-File: /config.json
{"ip":"192.168.2.11","netmask":"255.255.255.0","gw":"192.168.2.1","ccuip":"192.168.2.4","sonoff":"Feuer","restoreOldState":true,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}

JSON OK
ssid = Embedded-VPN, psk = *************
AP-Modus ist aktiv!
Should save config
Wifi Connected
CUSTOM STATIC IP: 192.168.2.11 Netmask: 255.255.255.0 GW: 192.168.2.1
Custom IP Address is set!
saving config
{"ip":"192.168.2.111","netmask":"255.255.255.0","gw":"192.168.2.1","ccuip":"192.168.2.4","sonoff":"Feuer","restoreOldState":true,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v42f824b2
~ld

   Sonoff EC:FA:BC:13:1F:A7 startet...
Config-Modus durch bootConfigMode aktivieren?
-> bootConfigModeFilename mounted file system
-> bootcfg.mod existiert NICHT
Config-Modus mit Taster aktivieren?
Config-Modus nicht aktiviert.
loadSystemConfig mounting FS...
loadSystemConfig mounted file system
loadSystemConfig reading config file
loadSystemConfig opened config file
Content of JSON Config-File: /config.json
{"ip":"192.168.2.111","netmask":"255.255.255.0","gw":"192.168.2.1","ccuip":"192.168.2.4","sonoff":"Feuer","restoreOldState":true,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}

JSON OK
ssid = Embedded-VPN, psk = *************
Connecting WLAN the classic way...
.......Wifi Connected

WLAN erfolgreich verbunden!
SSID: Embedded-VPN
IP Address: 192.168.2.111
Gateway Address: 192.168.2.1
signal strength (RSSI):-39 dBm

Sonoff Modell = Switch / S20
getStateFromCUxD url: http://192.168.2.4:8181/cuxd.exe?ret=dom.GetObject(%22Feuer%22).Address()
result: null
HomeMaticConfig.ChannelName = CUxD.null
getLastState mounted file system
laststat.txt existiert
getLastState FileContent = 0
Switch Relay to 0 with transmitState = 1
setLastState mounted file system
setLastState (0) saved.
Starte OTA-Handler...
Starte UDP-Handler an Port 6676...
Feuer - Boot abgeschlossen, SSID = Embedded-VPN, IP = 192.168.2.111, RSSI = -39, MAC = EC:FA:BC:13:1F:A7
```
Du wirst mit für verrückt halten... oh man: optisch ist jetzt alles i.O. soweit passt es, aber in Wirklichkeit verbindet sich der Adapter aber unter der IP 192.168.2.62 und auch nur unter dieser ist das Webinterface erreichbar.

jp112sdl, 2018-03-22:
Kann es sein, dass es mit dem ARP-Cache auf deinem Rechner zu tun hat?
Unter Win: `arp -d -a` als Administrator ausführen

elbman, 2018-03-22:
denke nicht, mit dem Iphone ist der Adapter ja auch nur unter 192.168.2.62 erreichbar...

elbman, 2018-03-22:
in der Fritzbox erscheint er in der Netzwerkübersicht auch mit der 192.168.2.62.
Habe den "bekannten Adapter" schon mehrfach dort gelöscht, er verbindet sich jedoch immer wieder mit seiner 192.168.2.62

jp112sdl, 2018-03-22:
Und du bist dir 100% sicher, keine Reservierung für die MAC angelegt zu haben? 

Ich hatte zwar vorletzten Versuch geschrieben - aber es war eigentlich der letzte.
Mehr Möglichkeiten, dem Interface eine statische IP zu vertickern, hab ich leider nicht...

elbman, 2018-03-22:
ja bin ich. 

Ich habe allerdings etwas Entscheidendes festgestellt. 
Ich habe den Adapter soeben komplett mit 00en beschrieben, dann die letzte Sonoff FW aufgespielt, konfiguriert und IP leer gelassen....
welche IP erhält der Adapter ? natürlich wieder die 192.168.2.62
Daraufhin habe ich in der Fritzbox den DHCP Bereich für die automatische Adressvergabe verändert (Adressen ab 192.168.2.70)
und siehe da nach dem Reboot erhalte ich die 192.168.2.71 !!!!

Jetzt stellt sich mir nur noch die Frage, wie es sein kann, dass trotz fest konfigurierter IP eine DHCP zugewiesene Adresse genommen wird?

Ich habe es gerade geschafft 192.168.2.119 zu konfigurieren, nach erneutem bootConfigMode und speichern mit unveränderten Werten nimmt er wieder die DHCP Adresse 192.168.2.71 ????

Irgend wo hier muss der Hund begraben sein (die Ausgabe in der Console ist verbunden mit 192.168.2.119, jedoch in Wahrheit mit der 192.168.2.71)

jp112sdl, 2018-03-22:
Das Problem liegt irgendwo in den Tiefen der [esp8266/Arduino](https://github.com/esp8266/Arduino) Lib.
Ich kanns halt weder reproduzieren oder testen. Deshalb mach ich da auch kein Issue auf. 
Zum Thema fixed ip oder static ip gibts auch ein bisschen was... 
bspw: https://github.com/esp8266/Arduino/issues/1959
Aber ich hab schon alle möglichen Varianten von
`WiFi.config(ip, dns, gateway, subnet);` durch.
Vorm `.begin()`, danach, nach erfolgtem Connect... Nix hilft.

elbman, 2018-03-22:
??? ich weiß auch nicht weiter und finde keinen Zusammenhang. 
Reproduzieren kann ich es an meinem 2. Adapter (älter) auch nicht. 
Vielleicht liegt es am Adapter. Werde mich vorerst damit zurechtfinden, 
vielleicht stößt jemand demnächst ja noch auf des Rätsels Lösung

closed 2018-03-22 by elbman

jp112sdl, 2018-03-22:
Als Workaround die MAC-Adresse im Router reservieren, damit er immer dieselbe bekommt.