jp112sdl/SonoffHMLOX #44: communication error in openHAB2

thefox73, 2018-04-13:
First of all thank you very much for your spending your time in developing this firmware.
I flashed a Sonoff basic successfully and connected it to my CCU2 via CuXD.
The device works flawlessly in HomeMatic.
As I'm using openHAB2 as well, I tried to add the Sonoff in OH, too.
The discovery detects the Sonoff CUXD device and I'm able to add it as a thing. But then the things is showing a connection error

[image]

For the moment I use the http-binding to send commands from OH2 to the Sonoff basic. and I pull the Sonoff state via JSON into OH2, in case the Sonoff was switched via HomeMatic. So I have the actual state in OH2.
This works fine. But is there a way to get the HomeMatic thing working in OH2 directly?

jp112sdl, 2018-04-13:
Hi @thefox73 
I'm sorry but I do not know anything about OpenHab... 
The only 3rd-pty controller appliance I can help with is ioBroker.
And there it is easy - simply add another hm-rpc instance and select "CuxD" as connection type.

thefox73, 2018-04-14:
Okidoki, not a problem.
As I said, the openhab integration is working for me via http binding and JSON. So everything is fine :-)
Thank you

closed 2018-04-14 by thefox73