jp112sdl/SonoffHMLOX #46: Problems with Sonoff-Basic

JoeWiseman, 2018-04-21:
Hi,
just got my S20-plugs today.... And flashing also as building them into CCU worked fine! Great Job!!

Now I' thought to transform the tasmota-flashed basic-switches into SonoffHMLOX ....
What I did:
1. Basic was running tasmota. Via the web interface I uploaded the SonoffHMLOX.ino.generic.bin and did an firmware-update .... which failed! (>> Isn't that possible in generell?)
The Basic is showing up as an AP, but cannot be connected.... Either the computer won't connect to the network or - when connected - the configuration-page cannot be reached....

2. The same Basic is plugged to the programmer as described and the SonoffHMLOX.ino.generic.bin has been flashed via Win10-computer. the same result.

3. took another Basic-switch (also already flashed with tasmota) and flashed it with the SonoffHMLOX.ino.generic.bin ... same result

The network is instable, not "connectable", or the configuration-webpage cannot be found...

Any ideas?

Regards an thanks in advance for thinking about a solution.

Joe
P.S. Will it be possible to flash the SonoffHMLOX.ino.generic.bin from the tasmota-web-interface for my in-use S20???

jp112sdl, 2018-04-21:
Hi, 
can you please try it with release [1.0.20](https://github.com/jp112sdl/SonoffHMLOX/releases/download/1.0.20/SonoffHMLOX.ino.generic.bin)

I never used the tasmota firmware. So I do not know, what it may modify in depth in the eeprom of the sonoff device.
But I've heard from other users, that you can **not** change from Tasmota to SonoffHMLOX via web.
You have to use serial upload.

jp112sdl, 2018-04-21:
Ja :) 

Kannst du mal diese Version testen?
Dort kommt auch noch mehr Debug Output im AP-Modus

[SonoffHMLOX.ino.generic.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/1934977/SonoffHMLOX.ino.generic.bin.zip)

JoeWiseman, 2018-04-21:
Also, letzte Wasserstandsmeldung:
Mit einem Basic hat die 1.0.14 geklappt ... Danach konfiguriert und dann auf 1.0.22 update ... scheint zu laufen...
Der andere wollte so nicht. Das gleiche Beschwerdebild wie zuvor. 
Ich probiere jetzt mal die Debug-Version aus.... und melde mich dann zurück ...

JoeWiseman, 2018-04-21:
Also, ich habe mich mit einem Tablet verbunden und das lief soweit. Konfig habe ich jetzt noch nicht gespeichert..... 
Jetzt werde ich das mal vom PC aus versuchen.....

Hier der Output von der Tablet-Verbindung:
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 13
cnt 
Config-Modus nicht aktiviert.
loadSystemConfig mounting FS...
loadSystemConfig mounted file system
/config.json not found.
ssid = Sesamstrasse@2GHz, psk = *********************************
*WM: Adding parameter
*WM: sonoffmodel
*WM: Adding parameter
*WM: ccu
*WM: Adding parameter
*WM: loxudpport
*WM: Adding parameter
*WM: sonoff
*WM: Adding parameter
*WM: hmpowervariable_pow
*WM: Adding parameter
*WM: hmecvariable_pow
*WM: Adding parameter
*WM: custom_powermeasureinterval_pow
*WM: Adding parameter
*WM: restorestate
*WM: Adding parameter
*WM: leddisabled_switch
*WM: Adding parameter
*WM: gpio14mode_switch
*WM: Adding parameter
*WM: custom_gpio14assender_switch
*WM: Adding parameter
*WM: backendtype
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: custom_ip
*WM: Adding parameter
*WM: custom_netmask
*WM: Adding parameter
*WM: custom_gw
state: 5 -> 0 (0)
rm 0
del if0
mode : softAP(5e:cf:7f:97:2wifi evt: 1
STA disconnect: 8
4:7c)
*WM: SET AP
AP-Modus ist aktiv!
*WM: Starting Web Portal
*WM: dns server started with ip: 
*WM: 192.168.4.1
*WM: HTTP server started
wifi evt: 8
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
add 1
aid 1
station: 78:00:9e:bb:9e:36 join, AID = 1
wifi evt: 5
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 9
wifi evt: 9
wifi evt: 7
*WM: Request redirected to captive portal
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
*WM: Handle root
*WM: Request redirected to captive portal
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
*WM: Handle root
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
mode : sta(5c:cf:7f:97:24:7c) + softAP(5e:cf:7f:97:24:7c)
add if0
wifi evt: 8
scandone
*WM: Scan done
*WM: Sesamstrasse@2GHz
*WM: -63
*WM: EasyBox-A40F08
*WM: -70
*WM: Sent config page
*WM: Handle root
*WM: Handle root
*WM: Handle root
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7

jp112sdl, 2018-04-21:
Das sieht alles sehr gut aus.
Also vom Sonoff aus ist eigentlich alles in Butter.

Mich wunder, dass das 3x kommt:
```
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
```

Aber auf jeden Fall sollte die Config-Seite angezeigt werden.
Wenn nicht automatisch (captive portal), dann mindestens beim manuellen Aufruf im Browser mit http://192.168.4.1

JoeWiseman, 2018-04-21:
sooo, das sah prima aus. Vielleicht müssen die Geräte einfach mehr miteinander reden ;-)
Getestet habe ich den Basic jetzt nur übers Webinterface. Scheint aber zu funktionieren ...
Hattest Du da jetzt außder der Debug-Ausgabe noch etwas geändert?

-----SNIPP-----
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
add 1
aid 1
station: 48:45:20:e2:09:b4 join, AID = 1
wifi evt: 5
wifi evt: 9
wifi evt: 9
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
scandone
*WM: Scan done
*WM: Sesamstrasse@2GHz
*WM: -61
*WM: EasyBox-A40F08
*WM: -67
*WM: Sent config page
*WM: Request redirected to captive portal
*WM: Handle root
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
*WM: Request redirected to captive portal
*WM: Handle root
*WM: WiFi save
*WM: Parameter
*WM: sonoffmodel
*WM: 0
*WM: Parameter
*WM: ccu
*WM: 192.168.1.220
*WM: Parameter
*WM: loxudpport
*WM: 
*WM: Parameter
*WM: sonoff
*WM: Sonoff-Basic-03
*WM: Parameter
*WM: hmpowervariable_pow
*WM: 
*WM: Parameter
*WM: hmecvariable_pow
*WM: 
*WM: Parameter
*WM: custom_powermeasureinterval_pow
*WM: 10
*WM: Parameter
*WM: restorestate
*WM: 1
*WM: Parameter
*WM: leddisabled_switch
*WM: 
*WM: Parameter
*WM: gpio14mode_switch
*WM: 0
*WM: Parameter
*WM: custom_gpio14assender_switch
*WM: 
*WM: Parameter
*WM: backendtype
*WM: 0
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: custom_ip
*WM: 
*WM: Parameter
*WM: custom_netmask
*WM: 
*WM: Parameter
*WM: custom_gw
*WM: 
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
scandone
state: 0 -> 2 (b0)
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with Sesamstrasse@2GHz, channel 5
dhcp client start...
wifi evt: 0
ip:192.168.1.240,mask:255.255.255.0,gw:192.168.1.1
wifi evt: 3
*WM: Connection result: 
*WM: 3
station: 48:45:20:e2:09:b4 leave, AID = 1
rm 1
bcn 0
del if1
pm open,type:2 0
mode : sta(5c:cf:7f:97:24:7c)
Should save config
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Already connected. Bailing out.
*WM: IP Address:
*WM: 192.168.1.240
Wifi Connected
CUSTOM STATIC IP: 0.0.0.0 Netmask: 0.0.0.0 GW: 0.0.0.0
saving config
wifi evt: 6
wifi evt: 8
{"ip":"0.0.0.0","netmask":"0.0.0.0","gw":"0.0.0.0","ccuip":"192.168.1.220","sonoff":"Sonoff-Basic-03","restoreOldState":true,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}
state: 5 -> 0 (0)
rm 0
pm close 7
del if0
usl
wifi evt: 1
STA disconnect: 8

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

jp112sdl, 2018-04-21:
Am Code nicht.

Jedoch wird die esp8266/arduino Library für das ganze Wifi-Zeugs verwendet.

Die ist mega buggy und es gibt täglich neue commits, wovon ich immer hoffe, eine "guten" zu erwischen. Das scheint so einer zu sein :)

JoeWiseman, 2018-04-21:
OK, dann werde ich mal diese Version ganz doll lieb haben und bei mir einen Ehrenplatz auf der Festplatte reservieren, damit ich in Zukunft diese erst einmal nutzen kann um dann ggf. ein Firmware-Update zu starten.
Viele Grüße und vielen Dank für den tollen Support!!!

JoeWiseman, 2018-04-21:
Hmm seltsam .... der erste Basic, der ohne die DEBUG-Verisioin zickte auf einmal wieder .... vermutlich wegen der Lib..... Dann habe ich die Debug-Version erneut geflashed und dann brauchte ich aber auch gar keine Daten zur Konfiguration mehr einzugeben....
Vielleicht wird da auch irgendetwas nicht richtig gelöscht, sodass noch Reste im System ihr Unwesen treiben??

JoeWiseman, 2018-04-21:
Hier die Ausgabe dazu:

scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with Sesamstrasse@2GHz, channel 5
dhcp client start...
wifi evt: 0
Config-Modus nicht aktiviert.
loadSystemConfig mounting FS...
loadSystemConfig mounted file system
loadSystemConfig reading config file
loadSystemConfig opened config file
Content of JSON Config-File: /config.json
{"ip":"0.0.0.0","netmask":"0.0.0.0","gw":"0.0.0.0","ccuip":"192.168.1.220","sonoff":"Sonoff-Basic-02","restoreOldState":true,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}

JSON OK
ssid = Sesamstrasse@2GHz, psk = ********************************
Connecting WLAN the classic way...
bcn 0
del if1
pm open,type:2 0
mode : sta(5c:cf:7f:97:30:01)
sleep disable
sl
.wifi evt: 8
scandone
usl
.ip:192.168.1.239,mask:255.255.255.0,gw:192.168.1.1
wifi evt: 3
Wifi Connected

WLAN erfolgreich verbunden!
SSID: Sesamstrasse@2GHz
IP Address: 192.168.1.239
Gateway Address: 192.168.1.1
signal strength (RSSI):-71 dBm

Sonoff Modell = Switch / S20
Starte OTA-Handler...
Starte UDP-Handler an Port 6676...
getStateFromCUxD url: http://192.168.1.220:8181/cuxd.exe?ret=dom.GetObject(%22Sonoff-Basic-02%22).Address()
[hostByName] Host: 192.168.1.220 is a IP!
result: CUX2801001:10
HomeMaticConfig.ChannelName = CUxD.CUX2801001:10
getLastState mounted file system
laststat.txt existiert
getLastState FileContent = 0
Switch Relay to 0 with transmitState = 1
setLastState mounted file system
setLastState (0) saved.
setStateCUxD url: http://192.168.1.220:8181/cuxd.exe?ret=dom.GetObject(%22CUxD.CUX2801001:10.SET_STATE%22).State(0)
[hostByName] Host: 192.168.1.220 is a IP!
HTTP CUxD.CUX2801001:10.SET_STATE success
result: false
Sonoff-Basic-02 - Boot abgeschlossen, SSID = Sesamstrasse@2GHz, IP = 192.168.1.239, RSSI = -72, MAC = 5C:CF:7F:97:30:01
webSwitchRelayOn(), keine Parameter, TimerSeconds = 0
Switch Relay to 1 with transmitState = 0
setLastState mounted file system
setLastState (1) saved.
Sending Web-Reply: {"state": 1, "timer": 0, "resttimer": 0, "fw": "1.0.22"}
Switch Relay to 0 with transmitState = 0
setLastState mounted file system
setLastState (0) saved.
Sending Web-Reply: {"state": 0, "timer": 0, "resttimer": 0, "fw": "1.0.22"}
webSwitchRelayOn(), keine Parameter, TimerSeconds = 0
Switch Relay to 1 with transmitState = 0
setLastState mounted file system
setLastState (1) saved.
Sending Web-Reply: {"state": 1, "timer": 0, "resttimer": 0, "fw": "1.0.22"}
Switch Relay to 0 with transmitState = 1
setLastState mounted file system
setLastState (0) saved.
setStateCUxD url: http://192.168.1.220:8181/cuxd.exe?ret=dom.GetObject(%22CUxD.CUX2801001:10.SET_STATE%22).State(0)
[hostByName] Host: 192.168.1.220 is a IP!
HTTP CUxD.CUX2801001:10.SET_STATE success
result: false
Switch Relay to 1 with transmitState = 1
setLastState mounted file system
setLastState (1) saved.
setStateCUxD url: http://192.168.1.220:8181/cuxd.exe?ret=dom.GetObject(%22CUxD.CUX2801001:10.SET_STATE%22).State(1)
[hostByName] Host: 192.168.1.220 is a IP!
HTTP CUxD.CUX2801001:10.SET_STATE success

jp112sdl, 2018-04-22:
Wenn es jetzt mit der letzten Version hier geklappt hat, werd ich da mal ne 1.0.23 draus machen. 

Log sieht schick aus.

>Dann habe ich die Debug-Version erneut geflashed und dann brauchte ich aber auch gar keine Daten zur Konfiguration mehr einzugeben....
Vielleicht wird da auch irgendetwas nicht richtig gelöscht, sodass noch Reste im System ihr Unwesen treiben??

Flashen hat nichts mit dem EEPROM-Speicher (wo die WLAN Zugangsdaten liegen) zu tun. Der bleibt erhalten. Auch das Dateisystem (SPIFFS), wo die config liegt.

JoeWiseman, 2018-04-22:
Danke für Deine Antwort,

Gibt es denn einen einfachen Weg den ganzen Speicher zu löschen, damit man so etwas wie einen "Clean-Flash" durchführen kann. 
Ich habe bei der Tasmota-Software ein Python-Tool gefunden, aber das ist für mich zu kryptisch mit der Installation.

Ich könnte mit vorstellen, dass durch das Nicht-Löschen des Speichers Fragmente des Speichers zur Konfiguration genutzt werden, sodass es evtl. zu Komplikationen mit der Software kommt?! Ist aber nur so eine Idee.

Ich komme auch nur deswegen darauf, da es bei "den Tasmotas" noch keine Probleme mit irgendwelchen unterschiedlichen Bibliotheken gab.

VIele Grüße
Joe

jp112sdl, 2018-04-22:
Ich habe mich damit nie wirklich befasst.
Aber auf die Schnelle habe ich bei Google das hier gefunden:
https://community.openenergymonitor.org/t/esp8266-howto-reset-eeprom-and-correct-settings-for-arduino-ide/4111/4

Tasmota wird m.W. auch nicht von einem Einzelkämpfer entwickelt... da ist viel mehr Bewegung drin.
Ich hab die SonoffHMLOX nur mal für meine Belange entwickelt und den Quellcode öffentlich gemacht. Hätte nie gedacht, dass es mal so einen Anklang findet :) 
Aber man muss da echt am Ball bleiben... Sich die Commits der Libs anschauen, drauf reagieren, testen... das alles kann ich schon zeitlich leider nicht leisten.

<hr/>

>Ich könnte mit vorstellen, dass durch das Nicht-Löschen des Speichers Fragmente des Speichers zur Konfiguration genutzt werden, sodass es evtl. zu Komplikationen mit der Software kommt?!

Eigentlich nicht... Ich wüsste nicht, was da reinspielen sollte.
Wifi AP+PSK liegen im EEPROM, da ist nix dran zu rütteln.
Der Rest der Config liegt im SPIFFS in einer Datei, die immer komplett geschrieben wird (als nicht nur inhaltlich an geänderten Stellen verändert).

JoeWiseman, 2018-04-22:
Nochmals vielen Dank für Deine ausführliche Antwort.
Leider bin ich nur einer von diesen Halbwissenden Bastlern, die einen Quellcode nur aus einigen geklauten Schnipseln zusammensetzen, sodass ich bei der Entwicklung nich wirklich eine Hilfe wäre....
Wobei sich mir auch die Frage aufdrängt was überhaupt noch zu entwickeln ist. 
Für mich - der schon seit Jahren nach einer kostengünstigen Erweiterung von HM sucht - ist das schon der Himmel auf Erden .....
Ich wollte die Schalter immer nur über die CCU und per Web-Befehle stuerun können und dann noch eine Rückmeldung an die CCU bekommen.
Und das hast Du alles umgesetzt!!!
Echt grandios!! Vielen DANK dafür!!

closed 2018-04-22 by JoeWiseman

jp112sdl, 2018-04-22:
Noch interessanter wird es, wenn man den Sonoff 4Ch auf Homematic Funk umbaut.
ESP runter - Arduino Pro Mini + CC1101 Funkmodul rein.
Kein CUxD, kein WLAN... direkt in HM anlernbar, sogar Direktverknüpfungen gehen - wie beim HM-LC-SW4-SM :) 

[image]

[image]

[image]

JoeWiseman, 2018-04-23:
Cool, das würde ja bedeuten, dass man einen 4-fach-Schaltaktor für knappe 25€ bekäme (wenn man mal von dem Stundenlohn beim Basteln absieht). Wirklich coole Sache, aber ich glaube da bin ich raus .... Das ist mir eine Spur zu "hoch".
... ich dachte immer, dass die HM-Geräte so einen geheimen Funkverkehr haben? Ist denn da mittlerweile etwas veröffentlicht worden??

jp112sdl, 2018-04-23:
> ... ich dachte immer, dass die HM-Geräte so einen geheimen Funkverkehr haben? Ist denn da mittlerweile etwas veröffentlicht worden??

Es gibt ne Lib im Netz und der CCC hat bereits 2012 das BidCos Protokoll "entschlüsselt"

Gelegenheitsbastler, 2018-04-23:
Ich habe ein kleines Platinchen entworfen, dass statt des ESP in den Sonoff gelötet wird. Wenn ich keinen Fehler rein gebaut habe, sollte das ziemlich einfach zu nutzen sein. ESP raus, Platine rein, CC1101 drauf, ATMega328 per Rx/Tx (Arduino Bootloader ist schon drauf) flashen und fertig. 

Bei Interesse kann ich das gerne vorstellen, wenn die Platinen eingetroffen sind (und funktionieren ;-) )

jp112sdl, 2018-04-23:
Der Sonoff 4CH Umbau müsste auch erstmal in ein eigenes Repo

JoeWiseman, 2018-04-23:
Hallo Ihr beiden,
das Thema finde ich zunehmend spannend, da dieses ewige Flashen und Konfigurieren per WLAN irgendwie nervig ist. Vielleicht kann Du @Gelegenheitsbastler ja Deine Konstruktion einmal genauer vorstellen?

Gelegenheitsbastler, 2018-04-23:
Naja, so ganz viel gibt es da noch nicht vorzustellen. Die Platine sieht aus wie ein großes "T". Sie wird statt des ESP in den Sonoff gelötet. Auf der Platine befindet sich ein Atmega328P mit minimalster Peripherie (mangels Platz), so wie eine Buchsenleiste (ist breiter als der ESP, daher die "T" Form), um die CC1101 Platine hinein zu stecken. Vor Kopf sind vier Lötpads Tx/Rx, GND und Reset. Das war es im Grunde auch schon. 

Der Atmega muss einmalig mit einem Sketch versehen werden. Die Einstellungen werden nach dem Anlernen über die HM Steuerung getätigt.

Wie schon geschrieben, werde ich das Platinchen vorstellen, wenn es eingetroffen ist und funktioniert. Ich fand die Idee so genial, dass ich mir einen 4Ch Sonoff bestellt habe, obwohl ich dafür aktuell eigentlich gar keinen Bedarf haben ;-)

Ich habe auch noch vier andere Platinen erstellt. Die habe ich die Tage im Forum vorgestellt. Dort findest Du alle Infos dazu.

jp112sdl, 2018-04-23:
@JoeWiseman 
Das nächste Problem (für einen Laien auf dem Gebiet) ist, 
a) den 328P auf die Platine zu löten und ihn auch 
b) mit einem Bootloader zu versehen bzw. 
c) anschließend den Sketch zu flashen.

Gelegenheitsbastler, 2018-04-24:
Wenn es nicht in eine Massenproduktion ausartet, dann könnte ich a und b und bei Bedarf auch c für Joe erledigen. Der Zeitaufwand ist überschaubar.

JoeWiseman, 2018-04-24:
Hallo,
ich denke, wenn ich eine Anleitung habe sollte das mit dem Löten (wenn sich der Schwierigkeitsgrad im Rahmen hält) schon möglich sein... Ich habe ja auch alle meine HM-Jalousieaktoren selber zusammengebaut.

@jp112sdl Dass das mit dem Flashen ein fester Bestandteil ist ist klar ... Bei dem von Dir bereitgestellten Image und dem Flashen geht das ja auch ratz fatz. Bei der Tasmota-Software habe ich immer noch ein paar Einstellungen vornehmen, das Image compilieren und dann Flashen müssen. Und wenn man dann "länger" mal kein Gerät mehr zum Einsatz gebracht hat, dann muss man sich immer wieder einlesen und überlegen. Alles in Allem war das recht nervig.

@Gelegenheitsbastler 
#1 In welchem Forum hast Du die Platine vorgestellt? Kannst Du mir bitte den Link posten?
#2 Wäre denn dieser Umbau auch für die S20er und die Basic-Baureihe geeignet. Für einen 4fach Schalter habe ich einfach zu wenige schaltbare Geräte ;-) Ich habe mir mal einen Dual gekauft, der liegt aber - wegen zu schlechter WLAN-Abdeckung in meiner Garage - leider ungenutzt in der Ecke.

Gelegenheitsbastler, 2018-04-24:
Wenn Du Dir zutraust, den Atmega328P TQFP32 löten zu können, dann sollten das RM 2.0mm auch kein Problem sein. 
#1 
https://homematic-forum.de/forum/viewtopic.php?f=18&t=43162
https://homematic-forum.de/forum/viewtopic.php?f=18&t=43243
https://homematic-forum.de/forum/viewtopic.php?f=18&t=43247
https://homematic-forum.de/forum/viewtopic.php?f=18&t=43274

#2 
Jaein. Beim 4Ch ist der ESP als komplettes Modul aufgelötet, wodurch man etwas mehr Platz hat. Beim S20 ist der "nackte" ESP aufgelötet. Aber der Umbau ist relativ simpel. Beim Basic müsste ich erst nachschauen. Aber aus der Erinnerung war da auch ein "nackter" ESP verbaut.

Ein Beispiel für den S20 findest Du hier:
https://homematic-forum.de/forum/viewtopic.php?f=18&t=43162

JoeWiseman, 2018-04-24:
Danke für die Links,
die Löterei ist doch eine Ecke komplizierter als beim Zusammenbau meiner Jalousieaktoren ....
Für mich ist das immer auch eine Kosten/Nutzen-Abwägung und ich glaube, dass ich da doch erheblichen Aufwand treiben müsste um so eine Geschichte zusammenzubauen :-(.
Daher werde ich wohl zunächst bei der Variante mit dem WLAN und dem CUxDaemon-bleiben. Das bekomme ich gut umgesetzt. Und um ehrlich zu sein habe ich auch nicht soooo viele Sachen die ich noch automatisieren wollte, sodass sich die Kosten in Grenzen halten dürften. 
Vielleicht gibt es ja in Zukunft einen Hersteller, der HM-kompatible Produkte anbietet ....
Die Sonoff-Komponenten werden ja auch schon mit Tasmota-Firmware von deutschen Händlern angeboten.

jp112sdl, 2018-04-24:
>Daher werde ich wohl zunächst bei der Variante mit dem WLAN und dem CUxDaemon-bleiben.

Für ein bisschen Deko-Licht würde ich den Umbauaufwand auch nicht betreiben.
Da wo es nicht auf Zuverlässigkeit ankommt, lasse ich Sonoffs weiterhin mit WLAN im Einsatz.

>Vielleicht gibt es ja in Zukunft einen Hersteller, der HM-kompatible Produkte anbietet ....

Höchstens für HmIP, weil eQ-3 dafür eine SDK an andere Hersteller verkauft. 
Fürs klassische HM nicht.