jp112sdl/SonoffHMLOX #47: Sonoff POW Schaltzustand an Homematic

chip131001, 2018-04-28:
Hallo, ich habe den Sonoff POW geflasht und mit Homematic eingerichtet. 
Es funktioniert alles wie es soll. Wenn ich aber den Sonoff komplett vom Strom trenne und wieder einschalte dann wird der Schaltzustand nicht an Homematic übertragen beim Drücken des Tasters oder über das Webinterface. Wenn ich in der Konfiguration die IP der CCU2 einmal ändere, und speichere funktioniert wieder die Übertragung des Schaltzustandes zu CCU2. Wie kann ich das abstellen?

jp112sdl, 2018-04-29:
Hallo.

Für mich sieht es so aus, als würde nach dem Hochfahren des Sonoff die CCU2 nicht erreicht werden.

Ist es möglich, dass du ein serielles Log anfertigst und hier postest?
FTDI anschließen, 115200 Baud einstellen und zB mit Putty verbinden.
Anleitungen findest du im Netz zahlreich.
z.B. hier: https://pbxbook.com/voip/sputty.html

chip131001, 2018-04-29:
Hallo,
ich habe eben den Sonoff neu geflasht und keine statische ip Adresse vergeben, sondern von der Fritzbox eine IP zuweisen lassen. Jetzt funktioniert alles wie es soll. Liegt es vielleicht an der statischen IP?

jp112sdl, 2018-04-29:
Das ist durchaus möglich. Da gibts momentan Schwierigkeiten... aber nur manchmal. Was es mir fast unmöglich macht, der Ursache auf den Grund zu gehen.
Am besten, du legst eine Reservierung an, so dass dein Sonoff immer dieselbe IP bekommt.

closed 2018-04-30 by chip131001