jp112sdl/SonoffHMLOX #49: now switch after flash Obi Wifi with version x.24

BigJohnson68, 2018-05-03:
i haved flash two adapters succesfull and can switch on the website on/off, but the power is always on.
i don´t can power off with the power button on the adapter direct.
what is wrong, what is the error?

BigJohnson68, 2018-05-05:
Hallo Jerome,
oh je, ich hatte beim ersten Test den Schuko-Stecker in die enthüllte Wifi-Steckdose gesteckt, dabei hatte der Stecker direkten Strom und nicht die Schaltung vom Board. Das erklärt auch den Dauerstrom!

läuft jetzt absolut perfekt - vielen Dank für deine Mühe

füge doch deinen Screenshot deiner Beschreibung dem OBI-Stecker bei, hatte dies bei der ersten Konfig am Iphone übersehen

Gibt es irgend wann mal ein Firmware für die 433er Steckdosen über die Sonoff Bridge.
Dies wäre für mich voll cool, da ich aktuell diese über Fhem an Loxone weiterreiche

würde mich auch gerne als Tester zur Verfügung stellen...

jp112sdl, 2018-05-05:
> füge doch deinen Screenshot deiner Beschreibung dem OBI-Stecker bei, hatte dies bei der ersten Konfig am Iphone übersehen

Ich bin immer zu faul bei Änderungen, die ganzen Screenshots auszutauschen, deshalb hab ich auf einen solchen verzichtet. Geschrieben hab ich es ja. :) 

>Gibt es irgend wann mal ein Firmware für die 433er Steckdosen über die Sonoff Bridge.
Dies wäre für mich voll cool, da ich aktuell diese über Fhem an Loxone weiterreiche

Nein, weil ich grundsätzlich nix von dem 433MHz Kram halte, da es nur unidirektional arbeitet.