jp112sdl/SonoffHMLOX #51: Homematic - path curl

PaulMerk, 2018-05-05:
in my CCU the curl is not located at
/usr/local/addons/cuxd/curl

instead it is located in my CCU at
/usr/local/addons/cuxd/**extra**/curl

I propose to adjust the Wiki for both possibilities

additional hint: in case of problems, check firewall configuration

jp112sdl, 2018-05-05:
Which CUxD version are you using? Maybe you should use a newer one.

With the latest release, curl is located in `/usr/local/addons/cuxd`.
On RaspberryMatic and on CCU2.

Sure, it is also located in `/usr/local/addons/cuxd/extra/`, but that's the "old" path.

PaulMerk, 2018-05-06:
I use CUxD 2.0.0 or 2.0.1 in my CCUs, so it is not dependent on the CCU version.
It seems to me, that it depends on the version used at the time of first installation of CUxD where curl is located.

closed 2018-05-06 by PaulMerk