jp112sdl/SonoffHMLOX #54: Feste IP Adresse ändern

feuerwasser85, 2018-05-16:
Da ich nirgends etwas dazu gefunden habe:
Wie kann die IP Adresse eines bereits eingerichteten Switch geändert werden? (zB. Sonoff S20)
Ich habe es nachträglich mit `http://<IP>/wifisave?s=SSID&p=XXXX&sonoffmodel=0&ccu=192.168.0.9&loxudpport=&sonoff=&hmpowervariable_pow=&hmecvariable_pow=&custom_powermeasureinterval_pow=10&restorestate=1&gpio14mode_switch=0&backendtype=0&custom_ip=192.168.0.10&custom_netmask=255.255.255.0&custom_gw=192.168.0.1` versucht, leider ohne Erfolg. Auch in der Fritz Box lässt sich die IP nicht manuell verändern.

feuerwasser85, 2018-05-17:
Hat funktioniert, danke! Tut mir leid, dass ich mich hier wie ein Anfänger angestellt habe, den Punkt hatte ich glatt überlesen. Im Config-Mode funktioniert mein Link oben perfekt für die gesamte Konfiguration.

jp112sdl, 2018-05-17:
Kein Problem. Schön, dass es jetzt läuft 👍

closed 2018-05-17 by jp112sdl