jp112sdl/SonoffHMLOX #56: Schaltzustand des CuxD Gerätes wird nicht aktualisiert bei manueller Schaltung

ralfthiel, 2018-05-24:
Mit 1.0.27 lässt sich über das CuxD Gerät zwar das Sonoff Gerät schalten, wird jedoch manuell, also am Gerät selbst geschalten, aktualisiert sich der Zustand des CuxD Gerätes nicht.
Mit 1.0.23 war es noch OK, ab wann der Fehler auftritt kann ich derzeit nicht sagen.
Getestet mit S20.

jp112sdl, 2018-05-24:
Was kommt im Browser bei `http://<ip>/reloadCUxD` ?

ralfthiel, 2018-05-24:
CUxD Address = CUxD.CUX2801002:3

jp112sdl, 2018-05-24:
Drück jetzt mal auf den Taster. Wird nun der Schaltzustand wieder übertragen?

ralfthiel, 2018-05-24:
Ja geht wieder!

jp112sdl, 2018-05-24:
Ok, dann lässt sich das Problem etwas eingrenzen.

Wenn du den S20 aus der Steckdose rausziehst und wieder reinsteckst (Gerät 1x stromlos machen) - ist dann das Verhalten wieder wie eingangs beschrieben? Tasterdruck wird nicht übertragen?

ralfthiel, 2018-05-24:
Nein, es funktioniert auch nach stromlos machen wieder zuverlässig.

jp112sdl, 2018-05-24:
Sehr seltsam.

Beim Starten wird nach dem Herstellen der WLAN Verbindung 5x versucht, von der CCU die CUxD-Adresse zu ermitteln, die dem Sonoff-Gerätenamen zugehörig ist.
Schlägt es fehl, kann auch anschließend keine Übertragung des Schaltzustands erfolgen.

Warum es bei dir fehlgeschlagen war, lässt sich nun wohl nicht mehr herausfinden.

Man kann jedoch mit `/reloadCUxD` das Ermitteln der CUxD-Adresse manuell jederzeit anstoßen.

closed 2018-05-24 by jp112sdl

ralfthiel, 2018-05-24:
Danke alles klar.
Ich konnte es selbst bei dem S20, den ich wieder von der FW 1.0.23 auf die 1.0.27 hochgezogen habe nicht mehr nachvollziehen.
Nach 1-maligem `http://<ip>/reloadCUxD` auf allen S20 ist das Problem erledigt.
Danke für all Deine Mühe, gute Arbeit!

jp112sdl, 2018-05-24:
Vielen Dank!