jp112sdl/SonoffHMLOX #57: S20 Hat probleme nach der Ersteinrichtung ins Wifi zu kommen.

lukeatspace, 2018-06-03:
Hallo,

ich hab ein tolles Problem und noch keine Lösung via google oder etc gefunden. Wenn ich die SonoffHMLOX flashe, kann ich mich auf das Wlan connecten. Alle Einstellungen machen. Wenn ich auf Save drücke, dann hab ich kurz ein Ping aussetzer, bleibe im Wlan vom Sonoff und er Connected sich nicht ins HeimWifi. 

Gibt es da eine Lösung?

Vielen Dank für die Tolle Arbeit! by the way :)

jp112sdl, 2018-06-03:
Das passiert eigentlich nur, wenn keine Verbindung zu deinem WLAN hergestellt werden kann.
Dann fällt er wieder zurück in den AP-Modus.
WPA Key richtig eingegeben?

Ansonsten mal die seriellen Debugmeldungen (115200 Baud) posten. Vielleicht ist dann mehr zu erkennen.

jp112sdl, 2018-06-21:
Gibt es einen neuen Stand?

lukeatspace, 2018-06-23:
Halo Jérôme,

erstmal muss ich  mich entschuldigen für die späte Antwort und auch vielen Dank für die schnelle Antwort.
Hab ein 3 Monate altes Kind was momentan sehr viel Zeit in Anspruch nimmt. ;)

Also ich hab mal meine Versuchsreihe etwas erweitert. 

Aktuell nutze ich eine Apple Extreme als Wifi station die an einer Fritzbox 7430 hängt die den DHCP DNS etc macht. 

Bekomme es nicht hin das sich die S20 in dieses Netzwerk dauerhaft verbindet. 
Bin leider neu in diesem Gebiet und wollte mal fragen wie ich das ganze Debuge ?

Viele Grüße

Dominik

lukeatspace, 2018-06-23:
Was total komisch ist, wenn ich eine Fritz 7270 direkt daneben stelle, ohne Passwort, geht der Connect auch nicht. Gehe ich aber hin und flashe es mit der normalen Tasmota Firmware, ist der S20 sogar direkt im richtigen Heimnetz. 

Also das debugging wäre jetzt wichtig ;) denn ich hab keine Idee mehr. 

Viele Grüße

Dominik

jp112sdl, 2018-06-23:
> Also das debugging wäre jetzt wichtig ;) 

- Verbindung des Sonoff mit dem PC via FTDI herstellen (wie beim Flashen)
- Putty herunterladen:
https://the.earth.li/~sgtatham/putty/latest/w32/putty.exe
- serielle Verbindung zwischen Putty und dem FTDI COM Port herstellen, wie zB hier erklärt:
https://pbxbook.com/voip/sputty.html
  - nicht vergessen, Baudrate: 115200

Es sind nun im Putty-Fenster die seriellen Debugmeldungen zu sehen.
Diese am besten mal vom Starten des Sonoff, über den Konfig-Mode bis zum "Speichern"-Button-Klick komplett nach pastebin.com kopieren und den Link hier als Kommentar einfügen.
Lieber ein paar Zeilen zu viel als zu wenig.
Bitte nicht den kompletten Loginhalt hier direkt reinkopieren... das wird nur ein meterlanges Gescrolle ;)

lukeatspace, 2018-06-23:
Nun das Debugging:

https://pastebin.com/NxNce8d4

Viele Grüße und Vielen Dank

Dominik

jp112sdl, 2018-06-23:
Hui... das hab ich noch gesehen: `failed to open config file for writing`

Wenn das mal nicht von Tasmota kommt... die verbiegen da glaub ich jede Menge an der Speicheraufteilung.
Kannst du bitte mal im WifiManager auf "Factory Reset" klicken?
Dann wird das SPIFFS neu formatiert.
Evtl. klappts dann...

LG, Jérôme

lukeatspace, 2018-06-23:
Also das ist gerade ein ganz neuer s20 der noch nie Tasmota gesehen hat. Bekomme aber gerade wieder den identischen Fehler. 

https://pastebin.com/kMF5aSTC

jp112sdl, 2018-06-23:
Das ist ja seltsam. 
Gibt es etwa eine neue Hardware Revision vom S20?
Was kommt denn beim Factory Reset an Debugmeldungen? Theoretisch müsste der SPIFFS Format auch fehlschlagen. 

Ansonsten bin ich auch ziemlich ratlos.

lukeatspace, 2018-06-23:
Also auf dem S20 steht folgendes drauf S20_EU_V1.3 2017-2-16 1802

Folgende Info's sieht man im Konfig Menü

Chip ID
2877109
Flash Chip ID
1327198
IDE Flash Size
1048576 bytes
Real Flash Size
1048576 bytes
Soft AP IP
192.168.4.1
Soft AP MAC
x
Station MAC
x

ich weis nicht ob man daraus etwas erkennen kann.

lukeatspace, 2018-06-23:
https://pastebin.com/tQCpTXEj Auszug aus dem Factory Reset Debug.

Ich danke dir aber schonmal vielmals. So erkenne ich das es nicht unbedingt an mir liegt ;)

jp112sdl, 2018-06-23:
Hmm... 2017 ist jetzt schon etwa älter und bisher hat noch niemand dieses Problem gehabt. Daher überlege ich immer noch, ob beim Flashen bei dir irgendwas schief gelaufen sein könnte. 

Hast du die Möglichkeit, einen kompletten Chip Erase durchzuführen?
Am besten mal googlen... dazu brauchst du ein anderes Tool. Bin hier nur am Handy und im Urlaub. Ich kann es dir hier schwer in zwei Sätzen beschreiben. 

Bin gerade etwas planlos.

Eine Idee hätte ich noch, aber das wird erst frühestens Ende nächster Woche was.

lukeatspace, 2018-06-23:
Oh öhm das ist super nett, aber dann mach dir doch keinen Stress wenn du im Urlaub bist. Sie laufen ja mit der Tasmota Firmware. Würde halt nur viel lieber deine nehmen ;-) Somit hat das echt Zeit.

lukeatspace, 2018-06-23:
Die Chance besteht das ich etwas falsch gemacht habe beim Flashen. Die einzige Erfahrung von Arduino hab ich mit dem Board vom 3d Drucker. Wobei es nach der Anleitung von dir ja eigentlich sehr einfach aussieht. :) Viele Grüße

jp112sdl, 2018-06-23:
Spontan fällt mir noch ein... probier mal eine ältere Release. 1.0.20 zB

lukeatspace, 2018-06-23:
Hab es probiert mit der 1.0.20 wie von dir vorgeschlagen. 

saving config
failed to open config file for writing
 
Hat leider nicht funktioniert

jp112sdl, 2018-06-24:
Krasse Sache... Hab ich noch nie gehabt/gesehen sowas...

adadnc, 2018-07-02:
Kannst Du mal versuchen das veröffentlichte Binary zu verwenden und dies direkt mit esptool.py to flashen?

`$ esptool.py --port /dev/ttyUSB0 write_flash -fs 1MB -fm dout 0x0 SonoffHMLOXino.generic.bin`