jp112sdl/SonoffHMLOX #58: TransmitState per Webseite oder per UDP

Kater432, 2018-06-07:
Hallo,

tolles Projekt, habe es schon einige Zeit drauf und mir nun noch einige OBI Steckdosen geholt.
Meine Frage ist bezogen auf den transmitstate.

Über die switch.ino wird beim drücken des Tastern der Status mittels TRANSMITSTATE übertragen.
Kommt aber der Befehl über die Webseite direkt "webSwitchRelayOn" oder per "udpMessage" wird kein Status mehr rausgeschickt. Das wäre aber sicherlich ne gute Sache, wenn ich den Status zurückbekomme, wenn ich im Loxone schalte per UDP oder HTTP oder einfach direkt per Webseite. 
Hat es einen Grund warum das deaktiviert ist ? 

**web.ino**
void webSwitchRelayOn
bool _transmitstate = NO_TRANSMITSTATE;

**SonoffHMLOX.ino**
if (udpMessage == "1" || udpMessage == "on")
      switchRelay(RELAYSTATE_ON, NO_TRANSMITSTATE);

VG
Christian

jp112sdl, 2018-06-07:
Es ist nicht deaktiviert...

**Web.ino**
```
const char HTTP_ONOFF_BUTTONS[] PROGMEM = "<span class='l'><div><button name='btnAction' onclick='SetState(\"/1?ts=1&t=\"+document.getElementById(\"timer\").value); return false;'>AN</button></div><div><table><tr><td>Timer:</td><td align='right'><input class='i' type='text' id='timer' name='timer' placeholder='Sekunden' pattern='[0-9]{1,5}' value='' maxlength='5'></td></tr></table></div><div><button name='btnAction' onclick='SetState(\"/0?ts=1\"); return false;'>AUS</button></div></span>";
...
void webSwitchRelayOn
bool _transmitstate = NO_TRANSMITSTATE;
...
if (WebServer.argName(i) == "ts") {
        _transmitstate = WebServer.arg(i).toInt();
...
 switchRelay(RELAYSTATE_ON, _transmitstate);
```

Das Klicken auf "AN" setzt `ts=1`.

**Relay.ino**
```
void switchRelay(bool toState, bool transmitState) {
...
    if (GlobalConfig.BackendType == BackendType_Loxone) sendLoxoneUDP(String(GlobalConfig.DeviceName) + "=" + String(RelayState));
...
```

Ist Loxone als Backend konfiguriert, wird eine UDP Nachricht gesendet.

Hinweis zu:
**SonoffHMLOX.ino**
```
if (udpMessage == "1" || udpMessage == "on")
switchRelay(RELAYSTATE_ON, NO_TRANSMITSTATE);
```
Das ist nur für das Schalten per UDP (also Empfang einer UDP Nachricht)
=> Wenn du vom Miniserver aus das Relais schaltest, kommt keine Quittung zurück, weil der Sender ja weiß, dass er den Sonoff eingeschaltet hat.

Kater432, 2018-06-07:
Hallo, 
danke für die Rückinfo, habe es nochmal getestet, der UDP Monitor im Loxone wollte wohl nicht so wie ich wollte. 
Per Webseite auf ON / OFF funktioniert, wie du geschrieben hast (TS=1)
Per direkter Aufruf IP/on oder IP/off aber wieder nicht. Dafür ist es ja deaktiviert in den Zeilen, hatte das mit dem TS dazwischen übersehen ;-)

Bzgl. UDP oder direkt per Webseite IP/on vom Loxone, denke ich, es wäre aber trotzdem schön eine Rückmeldung per UDP zu erhalten. So kann man in Loxone gegenprüfen ob wirklich geschaltet wurde, ob die Steckdose erreichbar ist. UDP ist ja nur einfach ins Netz gesendet ohne Rückinfo. 

Würde mir das auch selber aktivieren, habe aber keine Ahnung vom compilieren in die bin Datei ;-)

jp112sdl, 2018-06-07:
Mal andersrum gefragt: Kann man aus Loxone nicht statt UDP auch einen HTTP Befehl absetzen?
Der gibt ja bei Erfolg dann 200 zurück... 
Bzw. würde mittels http://<ip>/1?ts=1 auch die UDP Quittung an Loxone gesendet.

Andererseits: Jeder UDP Befehl wird auch quittiert...
https://github.com/jp112sdl/SonoffHMLOX/blob/7b11e9814e9b21e81404a519266b15e766b54c51/SonoffHMLOX/UDP.ino#L12

Ich wollte am Code jetzt nicht unbedingt was ändern... das bringt die anderen User sonst durcheinander ;)

Kater432, 2018-06-07:
Jo, danke, so gehts auch- 
Per HTTP Befehl setze ich die Steckdose von Loxone, 
habe dort das ?ts=1 angehangen und nun bekomme ich den UDP Status wieder zurück.

closed 2018-06-07 by Kater432