jp112sdl/SonoffHMLOX #59: Firmware 1.0.27 falsche Werte für Leistung in VA und Strom

Tobbi35, 2018-06-08:
Hallo ich habe die neuste Firmware aufgespielt,
es stimmen der Strom und die Leistung in VA nicht, Strom ist um die Hälfte zu klein und
damit auch der VA Wert. Werte von Spannung und Leistung in W stimmen.

Bitte mal nach schauen

jp112sdl, 2018-06-08:
Mit einer älteren Firmware Version sind die Werte in Ordnung?

Tobbi35, 2018-06-08:
Hallo Jèrome,
Ja da stimmten die Werte noch. Habe nochmal mit einer ohmschen Last kalibriert, es bleibt bei den Messfehler. 
Habe leider zwei POW mit diesem update versehen, sonst hätte ich nochmal gecheckt.

Danke für die schnelle Antwort

Gruß Jürgen

jp112sdl, 2018-06-08:
Sehr seltsam. 
Ich kann es nicht nachvollziehen.
Habe meinen POW soeben mal angeschlossen und eine 20W Halogenlampe angeschlossen.
Es werden mit 20,62W angezeigt und 19,8VA.

Führe ich eine Kalibrierung durch, erhalte ich anschließend auch die 20 W, jedoch über 100VA.
Woher das kommt ist mir unklar.

Jedoch stimmt der Wert in Watt immer.

Eventuell entferne ich bei Gelegenheit noch die VA-Anzeige. Bei herkömmlichen Leistungsmessern erhält man ja auch meist nur die Angabe in Watt.

Ich kann sonst nur empfehlen, eine ältere Firmware einzusetzen. 
Am Code für den POW hat sich seit langer Zeit nichts geändert.

loxonoff, 2018-07-15:
Hallo Jèrome
Zunächst möchte ich mich für Deine Mühen und die Bereitschaft zu teilen bedanken.

Auch ich habe das Problem mit den falschen Messwerten, von meiner Seite absolut reproduzierbar.
Leider habe ich bisher nur einen POW, bin noch in der Testphase um zu eruieren, was am sinvollsten mit meiner Loxone-Steuerung funktioniert.

Sobald ich kalibriere, steigt der Messwert der Stromstärke immer auf ~ 1 Ampere, was dann eine Leistungsaufnahme in VA von ~ der Betriebsspannung ergibt.
Rein rechnerisch völlig OK, allerdings wird die Wattzahl absolut richtig dargestellt, geprüft mit einer 40 Watt Glühbirne und einer 116 Watt Halogenglühbirne.

In beiden Fällen aber der fehlerhafte Messwert der Stromstärke, jeweils ~ 1 Ampere...
Lösche ich die Kalibrierungsdaten, passen die Werte wieder ungefähr.

Die Formatierung auf der Übersichtsseite ist auch verschoben, die Werte überdecken die Einheiten.
Siehe beigefügte Screenshots.

Was mir noch aufgefallen ist, gehört aber nicht hierher, der Haken, um die LED(s) zu deaktivieren, scheint in der neuen Version ebenfalls ohne Funktion zu sein.
Oder ist das so gewollt, dass im Ruhezustand alle LEDs aus sind und wenn das Relais aktiv ist dann beide leuchten?
Das nur am Rande, da ja die ungewollten Funktionen vielleicht irgendwie miteinander verknüpft sein könnten...

Bei Rückfragen, gerne und jederzeit...

Mit den besten Grüßen, Tobias

[image]
[image]

jp112sdl, 2018-07-15:
Hallo Tobias.

> Zunächst möchte ich mich für Deine Mühen und die Bereitschaft zu teilen bedanken.

Vielen Dank für den Dank! :)

>Auch ich habe das Problem mit den falschen Messwerten, von meiner Seite absolut reproduzierbar.
Leider habe ich bisher nur einen POW, bin noch in der Testphase um zu eruieren, was am sinvollsten mit meiner Loxone-Steuerung funktioniert.
Sobald ich kalibriere, steigt der Messwert der Stromstärke immer auf ~ 1 Ampere, was dann eine Leistungsaufnahme in VA von ~ der Betriebsspannung ergibt.
Rein rechnerisch völlig OK, allerdings wird die Wattzahl absolut richtig dargestellt, geprüft mit einer 40 Watt Glühbirne und einer 116 Watt Halogenglühbirne.
In beiden Fällen aber der fehlerhafte Messwert der Stromstärke, jeweils ~ 1 Ampere...
Lösche ich die Kalibrierungsdaten, passen die Werte wieder ungefähr.

Das ist echt - doof. Also bei mir haut es hin. Immer. 
Ich bin schon am Überlegen, ob es den POW mit unterschiedlichen HW-Revisionen gibt!?
Es gibt jedenfalls eine Version mit einem komplett anderen Mess-Chip. 
Aber da der m.W. auch ganz anders funktioniert, als der HLW8012, hättest du dann erst gar keine Werte erhalten.
Momentan bleibt mir da nur, zu raten, bei den default-Einstellungen zu bleiben und keine manuelle Kalibrierung vorzunehmen.

> Was mir noch aufgefallen ist, gehört aber nicht hierher, der Haken, um die LED(s) zu deaktivieren, scheint in der neuen Version ebenfalls ohne Funktion zu sein.
Oder ist das so gewollt, dass im Ruhezustand alle LEDs aus sind und wenn das Relais aktiv ist dann beide leuchten?
Das nur am Rande, da ja die ungewollten Funktionen vielleicht irgendwie miteinander verknüpft sein könnten...

Da hab ich grad im Dev-Zweig noch eine Änderung vorgenommen.
Es betrifft beim POW jedoch auch nur die WLAN LED. Die rote Relais-LED ist intern fest verdrahtet.

> Die Formatierung auf der Übersichtsseite ist auch verschoben, die Werte überdecken die Einheiten.
Siehe beigefügte Screenshots.

Was nutzt du für ein Endgerät?

loxonoff, 2018-07-15:
Hallo Jèrome

Vielen Dank für die schnelle Rückmeldung.

Ich benutze einen Windows-PC und als Browser den Firefox in der aktuellen Version.
Habe es noch nicht auf dem Mobiltelefon probiert, da mir das immer zu viel gefummel ist.
Ausserdem möchte ich mit der Web-UI am liebsten wenig zu tun haben, möchte, wenn möglich, alles mit dem Loxone MIniserver erschlagen.
Schaue es mir aber gleich, um Dir entsprechendes Feedback geben zu können, mal an.
Habe hier ein iPhone und ein Android liegen, jeweils mit dem Standardbrowser...

Auf dem iPhone schreibt er den Spannungswert in Einheit (V), ist aber aktuell der einzige der angezeigten Werte, welcher vor dem Dezimaltrennzeichen (Punkt) dreistellig ist.
Probiere es kurz mal mit der 116 W Birne...
Okay, nun schreibt er auch die Messwerte von W und VA in den Einheiten-Bereich.
Seltsamerweise hat er mir nach der ersten Messung auch noch die Werte der 40W-Glühlampe angezeigt, die scheinen noch irgendwo gespeichert gewesen zu sein.
Erst mit der zweiten Messung (Messintervall auf Standard gelassen => 10 Sekunden) wurden diese auf die plausiblen Werte aktualisiert.

Nun unter Android (Chrome)...
Skaliert wunderbar, ich weiss warum ich die Apple-Geräte nicht mag und fühle mich wieder bestätigt ;-)

Hoffe ich konnte etwas helfen.

Beste Grüße, Tobias

jp112sdl, 2018-07-15:
> Skaliert wunderbar, ich weiss warum ich die Apple-Geräte nicht mag und fühle mich wieder bestätigt ;-)

Habe gerade den Fehler im CSS gefunden. Wird mit der nächsten Version gefixt.
Dass es bei dir unter Android (Chrome) geht, ist dann der Fehler 😎 

Btw: Ich setze ausschließlich Apple-Technik ein. 😉

jp112sdl, 2018-07-15:
Hmm... also du benötigst **in** der Arduino noch folgende Bibliotheken:
- [HLW8012](https://github.com/xoseperez/hlw8012)
- die neueste [ArduinoJSON](https://arduinojson.org/v5/doc/installation/)
- die [ESP8266/Arduino Version ausm Git](https://github.com/esp8266/Arduino#using-git-version) (nicht die Release-Version!)

Alle [Dateien](https://github.com/jp112sdl/SonoffHMLOX/tree/develop_pow/SonoffHMLOX) (der Link zeigt zum Develop-Branch, wo die letzten Änderungen drin sind) müssen gemeinsam in einem Ordner liegen, der SonoffHMLOX heißt. 
Das wars eigentlich.

jp112sdl, 2018-07-15:
Wichtig sind dann noch die richtigen Boardeinstellungen beim Kompilieren:
[image]

loxonoff, 2018-07-15:
Hallo Jèrôme

Nun habe ich auch das Dach für das "o" in Deinem Namen gefunden ^^

Vielen dank, ich will es mal probieren, ansonsten warte ich bis Du eine .bin zur Verfügung stellst.

Habe vor ein paar Jahren mal mit den "nackten" AVRs gearbeitet, ohne Arduino-Unterbau.
Allerdings bin ich da ziemlich raus, wobei die Arduino-Umgebung es einem scheinbar recht einfach macht.
Tolle Sache, überlege schon meinem Sohn mal etwas Equipment zu besorgen, vielleicht findet er Interesse daran...

Besten Dank und Grüße aus dem wunderbar sonnigen NRW, Tobias

jp112sdl, 2018-07-15:
Die Arduino IDE ist sehr ungelenk.
Kein wirkliches Syntax-Highlighting (bis auf ein paar Keywords), keine Sprungmarken... eigentlich ist es nur ein simpler Texteditor mit Kompilerfunktion.
Atmel Studio gibts leider nur für Windows. Bleibt noch Atom mit Platform.IO, aber da bin ich zu faul, mich einzuarbeiten.
Letztendlich habe ich mich an die Arduino IDE gewöhnt. Man muss halt viel mitdenken, was man bei modernen Entwicklungsumgebungen ja kaum noch muss, wenn ich an Visual Studio denke.

Hier ist das Kompilat der letzten Develop-Version:
[SonoffHMLOX.ino.generic.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/2195859/SonoffHMLOX.ino.generic.bin.zip)

Grüße aus irgendwo zwischen Hannover und Berlin :)

loxonoff, 2018-07-15:
Ja, Atom mit Platform.IO, damit habe ich mich die letzten Tage rumgeschlagen...
Hatte ja auch etwas in Richtung Tasmota geschaut... mir graut es aber einfach davor eine weitere Schicht (z.B. ioBroker oder FHEM) zwischen Aktoren und die Loxone zu quetschen.
Schlussendlich muss man dann an etlichen Stellen suchen wenn irgendwann mal irgendetwas klemmt, mehr als ungünstig!
Vor allem muss man dann noch das gebastelte Konstrukt im Kopf haben um die Zusammenhänge zu verstehen. Ich bezweifele allerdings ganz stark das ich das in einem Jahr noch weiss...
Daher ist Deine Entwicklung in meinen Augen Gold wert, eine direkte Anbindung ist einfach optimal.

Hast Du zukünftig ggf. noch vor weitere Sonoff-Devices in Deine Software zu implementieren?
Habe hier daheim noch einen 4CH liegen, falls Du Hardware benötigst.
Die Tage bekommen ich noch einen 4CH Pro, allerdings nicht wegen der 433MHz-Anbindung, ich brauche potenzialfreie Kontakte, die bringt der Pro gleich mit.
Bei dem 4CH ist die Spannungsversorgung direkt mit auf die Relaiskontakte gelegt, ist also nicht potentialfrei.
Ich würde den 4CH aber gerne zur Ansteuerung von Ventilen meiner Gartenbewässerung nehmen, die vertragen nur 24 VAC...
Hatte aber schon geschaut, notfalls könnte man die 230 VAC von den Relais separieren, man müsste dann aber auf der Platine grossflächig Leiterbahnen zerstören, irgendwie unschön.
Die Versorgungsspannung des ESP könnte man über eine vorbereitete 2.5mm Klinkenbuchse, welche aber nicht aufgelötet ist, herstellen. Man könnte dafür ein 5V Handyladekabel verwenden.
Man benötigt die 5V, da die Relaisspulen damit angesteuert werden, der Rest wird mit 3,1 - 3,3 V versorgt.

Man könnte natürlich auch noch Koppelrelais verwenden, dann würde aber der LowBudget-Gedanke der Sonoff's nicht mehr greifen und wie schon geschrieben, ich mag es so einfach wie möglich, ohne zusätzliche und unnötige Schichten.

Würde Dir den 4CH sofort zur Verfügung stellen... vorausgesetzt, Du hast Lust und Zeit!

Beste Grüße, Tobias

jp112sdl, 2018-07-15:
Nein, weitere Geräte sind nicht geplant.
Der Loxone Support für die Basic ist eigentlich nur entstanden, weil @jenshavelberg Loxone nutz, wir sehr gut befreundet sind und er einen preiswerten Aktor brauchte.

Ich selbst nutze HomeMatic und arbeite ausschließlich mit 868MHz Funk (BidCos Protokoll).

Für den 4CH (Pro) habe ich einen Umbau auf HomeMatic-Funk erarbeitet, weil ich nicht weiter auf WiFi setzen möchte, da es immer eine zusätzliche Infrastruktur (WLAN / AP, ggf. Routing) verlangt und der ESP8266-Core für Arduino schlecht programmiert ist (am meisten nervt [#2330](https://github.com/esp8266/Arduino/issues/2330)).

[image]

Das Gerät kann ich nativ im HomeMatic System anlernen und - der aus meiner Sicht größte Vorteil vom HomeMatic Funk - mit anderen Funksendern (Wandtaster, Fernbedienungen) direkt verknüpfen, so dass ich den Aktor auch schalten kann, wenn mir mal die Zentrale explodiert.

Aber jetzt wirds hier OT...

loxonoff, 2018-07-15:
Statische IP scheint die Lösung zu sein, oder wohl eher nur ein Workaround...

Okay, dann bin ich vielleicht auch auf dem falschen Weg, muss mir das Ganze doch noch einmal genauer überlegen.
Es gibt von Loxone ja auch Extensions mit potentialfreien Kontakten, sind zwar immer nur 14 Stück für einen Haufen von Geld, aber ich habe die Loxone-Geschichte nunmal angefangen...

Habe nebenbei einen Dauerping auf den ESP (POW) gestarte, bisher ohne Timeout.
Allerdings verwende ich keinen Router, welcher mir nebenbei auch das WLAN zur Verfügung stellt.
Verwende eine richtige "Firewall", ein entsprechendes Modem davor und mit einem eigenen separaten WLAN-Netz für Handy-Radio-etc und ein eigenes Netzwerksegment für M2M-IoT ohne weiteren Zugriff von oder nach aussen.
Die APs sind von Unifi, was besseres hatte ich privat noch nie.
Hatte die letztens auch in der Firma ausgerollt, dort arbeiten 35 Stück ohne Probleme... kann ich nur empfehlen.

Vielen Dank bis hierher, ich werde noch einmal in mich gehen und alles sacken lassen, vielleicht bleibe ich doch nativ bei Loxone und beende die Spielerei hiermit...

Beste Grüße und einen schönen Abend
Tobias

loxonoff, 2018-07-17:
Hallo Jèrôm

Ich hatte heute Abend etwas Zeit um mir die Platine mal unters Mikroskop zu legen.
Der Chip für Spannungs- und Strommessung scheint ein anderer als der Dir bekannte zu sein.
Er heisst CSE7759, siehe Foto.

BEste Grüße und einen schönen Abend / gute Nacht,
Tobias

[image]

jp112sdl, 2018-07-18:
Dann ist das Chaos ja komplett 👍 
https://github.com/xoseperez/espurna/issues/696

Mittlerweile gibt es den POW offensichtlich in 4 verschiedenen Ausführungen.

loxonoff, 2018-07-18:
Guten Morgen Jerome
Ich kann Dir den POW gerne zukommen lassen, entweder in nagelneu oder mit bereits aufgelöteter Stiftleiste.

Beste Grüße, Tobias