jp112sdl/SonoffHMLOX #62: S20 vergisst Konfiguration nach Stromunterbrechung

PaulMerk, 2018-06-21:
Nach Stromunterbrechung verlieren meine Schalter S20 (4 Stück) die Konfiguration (Wlan SSID, Passwort).
Verbindung via http ist nicht möglich.
Die Fritzbox zeigt Geräte als verbunden an, jedoch kein ping.

Nur Neu-Konfiguration via 192.168.4.1 möglich.
Eintragungen zu Modell, CCU-IP, Bootverhalten, Backend bleiben erhalten.

Ist die Ursache die Einstellung IP via DHCP ?
Probehalber habe ich mal auf feste IP-Adresse umgestellt.

Was kann ich sonst noch tun für einen möglichst verlässlichen Betrieb ?

jp112sdl, 2018-06-21:
> Nach Stromunterbrechung verlieren meine Schalter S20 (4 Stück) die Konfiguration (Wlan SSID, Passwort).
Verbindung via http ist nicht möglich.
Die Fritzbox zeigt Geräte als verbunden an, jedoch kein ping.

Das spricht nicht dafür, dass sie die Konfiguration verlieren.
Hört sich eher nach 
https://github.com/esp8266/Arduino/issues/2330 an.

Oder auch https://github.com/jp112sdl/SonoffHMLOX/issues/40 was aber behoben ist.

> Eintragungen zu Modell, CCU-IP, Bootverhalten, Backend bleiben erhalten. 

Auch das ist ein Indiz, dass die Konfig nach wie vor vorhanden ist.

Ob DHCP die Ursache ist, weiß ich nicht. 
Bei mir läuft alles zuverlässig, habe jedoch auch einen Dauer-Ping zu laufen.

Ansonsten, probier mal diese FW, ob es damit besser klappt:
[SonoffHMLOX_2105b8b.ino.generic.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/2123780/SonoffHMLOX_2105b8b.ino.generic.bin.zip)

jp112sdl, 2018-06-21:
> ein Ping via CUxD (Doku 5.8.4) hilft wohl nur bei der Überwachung.

Nein, der Dauer-Ping hält in diesem Fall die ARP-Requests bzw -Replies am Leben.
Und ja - man setzt den CUxD Ping eigentlich für die Überwachung ein.

> Firmware-Versionen via OTA auf je einem flashen ?

Ja, kannst du OTA machen.

> Wenn sinnvoll wäre etwas zu loggen, ist damit das Fenster vom ESPLoader gemeint oder der Serielle Monitor von der Arduino-IDE (dann müsste ich die Gehäuse wieder aufschrauben) ?

In beiden Fällen müsstest du das Gehäuse aufschrauben.

Ich würde momentan erstmal nichts loggen. Wenn es ein generelles Problem wäre, hätten sich schon mehrere Leute zu Wort gemeldet.
Schauen wir erstmal, ob es mit der obigen Firmware besser wird.

PaulMerk, 2018-06-21:
@ping
welche Einstellungen in CUxD empfiehlst du ?
Ich habe SWITCH|INTERVAL_ALIVE  und SWITCH|INTERVAL_FAIL jeweils auf 60 stehen. 
SWITCH|INTERVAL_FAIL 5 und SWITCH|THRESHOLD 1
Ist das ok ?

@Firmware
Ich habe jetzt alle Geräte mit dieser Firmware upgedated. Wenn es wieder Probleme geben sollte, melde ich mich.

jp112sdl, 2018-06-21:
Ja, 60 Sekunden bei ALIVE sind ok, FAIL habe ich auf 15 Sekunden eingestellt. Den Rest lass auf default.

closed 2018-06-21 by PaulMerk