jp112sdl/SonoffHMLOX #63: Sonoff Switch Leuchtstoffröhren schalten

chip131001, 2018-06-25:
Hallo,
ich habe 2 Leuchtstoffröhren an Sonoff Switch angeschlossen. Mir ist aufgefallen das der Sonoff sporadisch neu startet. Wenn ich den Sonoff an eine normale Lampe anschließe passiert das nicht. Kann es sein das der Sonnoff für das schalten von Leuchtstoffröhren nicht geeignet ist?

jp112sdl, 2018-06-25:
Hallo,

ich kann dir das nicht beantworten. Habe nur ein bisschen Code für den Sonoff programmiert. Wie es EMV-technisch aussieht, weiß ich nicht. 
Kann mich aber entsinnen, schon mal von jemandem mit Problemen bei Leuchtstoffröhren gehört zu haben.

jp112sdl, 2018-06-25:
Es gibt elektronische Soft-Starter. Vielleicht schaffen die Abhilfe.

closed 2018-06-27 by jp112sdl

chip131001, 2018-07-11:
Hallo,

ich habe mir ein Koppelrelais bestellt und an den Sonoff angeschlossen. Seit dem ist es mit dem neu starten besser geworden. Seit ca. 1,5 Tagen läuft er ohne neu zu starten. Aber heute habe ich das Licht eingeschaltet und kurze Zeit später wieder ausgeschaltet, hat auch funktioniert. Ein paar Sekunden später startet der Sonoff neu und das Licht ist wieder an (ich habe den mit Absicht so konfiguriert damit ich das erkennen kann das er neu gestartet ist).
In der Weboberfläche der Fritzbox kann ich auch sehen das der Sonoff sich im WLAN abgemeldet und wieder angemeldet hat.
Gibt es bestimmte Gründe wann der Sonoff sich selbst neu startet, oder ist es normal das der Sonoff in bestimmten Zeitabständen neu startet? 
Ich weis nicht wie ich dem Fehler auf die Spur kommen kann.

jp112sdl, 2018-07-11:
Moin,

normal ist es nicht. 
Ich hab leider auch keine Idee, wie man dem Problem näher kommen könnte.
Zumal auch zyklisch keine Abfragen im Hintergrund laufen, die dann auf einmal das System zum Absturz bringen könnten.
Es gibt intern einen Millisekundenzähler, der aber erst nach ~49 Tagen überläuft, was auch abgefangen wird.

Man könnte höchstens dauerhaft den FTDI angeschlossen lassen und mal das serielle Logging mitschneiden, um zu schauen, was unmittelbar vor dem Reboot passiert.