jp112sdl/SonoffHMLOX #64: Problem bei S20 bei Ein-Ausschalt-Knopf

Zoelibat, 2018-06-30:
Hallo,

ich binde in Loxone die S20-Steckdose ein wie in der Beispielimplementation hier dargestellt: https://github.com/jp112sdl/SonoffHMLOX/wiki/Loxone

Schalte ich jetzt über den Schalter an der Steckdose ein, dann leuchtet die LED und die Steckdose ist ein.
Schalte ich jetzt wieder aus, dann ist die LED kurz aus und sofort wieder an. Er schaltet anscheinend die Steckdose ab und gleich wieder ein.

Nehme ich jetzt die virtuellen Eingänge aus Loxone raus, dann funktioniert auch der Schalter an der Steckdose ganz normal. 

Wie kann ich das Problem beheben/umgehen?

jenshavelberg, 2018-07-01:
Hi! 
Das Problem mit dieser Endlosschleife hatte ich zu Beginn auch. Aktuell fällt es mir schwer, mich an die Lösung zu erinnern. Was machst du denn mit den Eingängen in deiner Steuerung? Setzt du damit gleich wieder irgendeinen 2-Tasten Schalter oder ähnliches? Dieser hat dann wieder einen Ausgangsimpuls zu deinem Aktor und so kam es glaube ich bei mir zu dieser Endlosschleife. 
Vielleicht kannst du mir ein paar mehr Infos / Screenshots über deine Config zukommen lassen. Vielen Dank!
Gruß Jens

closed 2018-07-15 by jenshavelberg

Zoelibat, 2018-07-22:
Hi,

sorry für die späte Antwort. Der Aufbau ist gleich wie im Screenshot der Beispielimplementation.

[image]

"Steckdose_ein" und "Steckdose_aus" sind die virtuellen Eingänge, "SOO Steckdose" der virtuelle Ausgang.
Es funktioniert rein durch das Schalten in der App einwandfrei. Ebenso geht das Einschalten mittels dem Knopf auf der Steckdose. Schalte ich aber die Steckdose über den Knopf an der Steckdose ab, dann schaltet er kurz ab und gleich wieder ein. In der App steht die Steckdose nun auf "Aus", aber an der Steckdose selber leuchtet das Licht und sie ist ein.
Jetzt muss ich nochmal an der Steckdose drücken, um diese wirklich auszuschalten. Im App-Schalter muss ich zweimal drücken, um diese wieder auszuschalten.

Ich möchte gerne die Steckdose schalten können:
- über den Schalter an der Steckdose selber
- über die App (Ein/Aus)
- über eine Logik in meiner "Loxone-Welt", z.B. wenn Nacht, dann ein und wenn Tag, dann aus - oder ähnliches.

Wichtig wären mir die ersten 2 Möglichkeiten. Und diese auch abwechselnd und dort liegt eben jetzt genau das Problem.

Wie kann ich den erneuten Ausgangsimpuls verhindern? Zusätzliche Abfragen oder durch einen anderen Aufbau (anderer Schalter, ...).