jp112sdl/SonoffHMLOX #65: Gesamtverbrauch wird beim POW bei Stromausfall auf 0 gesetzt

ralfkaminski, 2018-07-13:
Hallo
Gibt es eine Möglichkeit, dass der Zähler des Gesamtverbrauchs beim POW nach einem Stromausfall nicht zurück gesetzt wird? 
Mit freundlichen Grüßen 
Ralf

jp112sdl, 2018-07-13:
Hallo Ralf, 

nein, das ist nicht vorgesehen. 
Bei den Homematic Zwischensteckern mit Leistungsmessung ist es auch so.

Es müsste sonst sekündlich bzw. in sehr kurzen Intervallen der Verbrauchswert im EEPROM abgelegt werden, was sich negativ auf die max. Schreibzyklen des Flash Speichers auswirkt.

PaulMerk, 2018-07-13:
Ich habe in anderer Konstellation ein weiteres CUxD Gerät eingerichtet, dessen daten überleben. Ob das hier mit einer Systemvariablen auch machbar wäre, habe ich nicht probiert.

jp112sdl, 2018-07-13:
Ja, das geht natürlich.
So funktionieren auch die Energiezähler in der CCU.
Es gibt einen Zähler Gerät und einen Zähler CCU2.
Der Zähler CCU2 kumuliert nur die Werte vom Gerät.
Das kann man auch mit einem kleinen Programm erledigen.
Wenn man die systeminternen Programme einblendet, sieht man sogar die internen Rechenprozeduren und kann sich dort Anregungen holen.

ralfkaminski, 2018-07-13:
Hallo zusammen.
Vielen Dank für die schnellen Rückmeldungen.
Mir kam dabei noch folgende Idee, um den Flashspeicher zu schonen. 
Kann der POW auch den zuletzt gespeicherten Wert der Systemvariable abfragen. Wenn dieser größer als der des POW ist, könnte er dem Wert der ccu verwenden. 
Wäre das noch ein denkbarer Ansatz für die Zukunft? 
Viele Grüße 
Ralf

jp112sdl, 2018-07-14:
Ein Ansatz ist es wohl, aber ich denke nicht, dass ich vorläufig an der Firmware programmieren werde.

jp112sdl, 2018-07-14:
Mal nebenbei noch gefragt: Welche FW Version setzt du ein? Die 1.0.27?

Zu der Idee mit dem Auslesen der Systemvariable:
Was ist nach einem Stromausfall? 
Sonoff POW und WLAN sind definitiv eher wieder da, als die CCU/RaspberryMatic.
Dann könnte die Variable beim Booten nicht abgefragt werden und die ganze Geschichte beginnt ja doch wieder bei 0 😃 
Vorschläge, wie man es umgehen sollte/könnte?

ralfkaminski, 2018-07-14:
Hi, 
Da hast du natürlich recht :-) 
Gibt es die Möglichkeit, dass die CCU DIE Variable im POW verändern kann? Dann könnte dies nach einem Stromausfall geschehen. 
Ich setzt die FW 1.0.27 ein

jenshavelberg, 2018-07-14:
@jp112sdl Kann die CCU nicht validieren, ob der neue Wert welcher vom POW empfangen wurde, kleiner als der bisher bekannte Wert ist? In diesem Fall muss die CCU so clever sein und den Wert ignorieren. Geht das nicht?

jp112sdl, 2018-07-14:
Doch das geht. 
Das wäre *die* Möglichkeit. 
THX. 

Btw.: hier is grad Strom weg -.-

jp112sdl, 2018-07-14:
Ach so. Nee. Falscher Ansatz. 
Der POW setzt eine Variable in der ccu. Völlig ungeachtet von irgendwas.

jenshavelberg, 2018-07-14:
ok, dann andere Logik: der POW bekommt einen einstellbaren Schwellwert, ab welchem er der CCU Werte senden darf. Also alles unter in Summe verbrauchten 0,5 kW/h oder was auch immer wird noch nicht in die CCU geschrieben. Somit ist die Phase nach einem Reboot für ne weile "geschützt". Bis dahin wird die CCU geschafft haben, die bisherige Energiesumme an den POW gesendet zu haben. Oder noch eins weiter: bevor der POW keinen Verbrauch durch ``addEnergyCounter`` erhalten hat, darf er der CCU keine Werte zurückgeben... 

welcher Vorschlag passt dir?

jp112sdl, 2018-07-15:
Der 2.
Ergänzung zu dem 2. Vorschlag:
- POW bootet, erreicht CCU nicht und merkt sich ein `unreach`-Flag.
- bei `unreach` wird gar nicht erst weiter versucht, Werte zu senden, bis die CCU `enableTransmission` an den POW geschickt hat.

Finde ich logisch sinnvoller, da man evtl. gar nichts aufaddieren und nur die Übertragung wieder aktivieren möchte.
Ginge nach deinem Vorschlag zwar mit `addEnergyCounter?value=0` auch, ist optisch aber etwas unschön.

jp112sdl, 2018-07-15:
So - hier der nächste Versuch.

1. Es gibt jetzt in der Weboberfläche die Möglichkeit, auszuwählen, ob beim Booten der letzte Werte von der CCU geholt werden soll oder nicht. Ich werd mal noch "Energiezähler" in "Energiezählerwert" umbenennen.
[image]

2. Ist die Energiezähler-Variable beim Starten des POW nicht erreichbar, wird die Übertragung deaktiviert.

3. Sie kann wieder aktiviert werden mit
`http://192.168.1.35/enableEnergyCounterTransmission?load=0` oder `load=1`
  - wenn `load=0` wird **nur** die Übertragung (nach vorheriger erneuter Prüfung auf Erreichbarkeit) aktiviert (mit dem Energiezählerwert der bei 0 begann)
  - wenn `load=1` wird **die Übertragung** (nach vorheriger erneuter Prüfung auf Erreichbarkeit) **aktiviert** und der Wert der Energiezähler-Variable auf den bisher angefallenen Stand aufaddiert *(wie bei der Option "Lade Energiezähler von CCU beim Starten")*

[SonoffHMLOX.ino.generic.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/2195489/SonoffHMLOX.ino.generic.bin.zip)

ralfkaminski, 2018-07-15:
Hallo,
habe gerade die letzte Firmware getestet.
Dabei ist mir aufgefallen, dass die Variable für den Energiezähler in der CCU nicht beschrieben wird.
Daraufhin habe ich das "Original Release 1.0.27" wieder geflasht und danach wurde die Variable wieder beschrieben.
Könntest Du das noch einmal prüfen.
Denn der Ansatz in der überarbeiteten Firmware ist super.
Wenn die CCU per Skript dann noch überprüft, ob der gemeldete Wert vom POW kleiner ist als der in der Systemvariable gespeicherte, kann die CCU dann auch den gemeldeten Zählerstand auf den letzten Wert aufaddieren und an den POW zurücksenden. Dann sind wirklich alle Szenarien für eine kontinuierliche Verbrauchszählung abgefangen.
Schönen Abend noch.
Gruß Ralf

jp112sdl, 2018-07-15:
> Dabei ist mir aufgefallen, dass die Variable für den Energiezähler in der CCU nicht beschrieben wird.

Evtl. hat der Variablen-Check nicht funktioniert.
Probier es mal mit `http://192.168.1.35/enableEnergyCounterTransmission?load=0` oder `load=1` wie vorher beschrieben. Und schau mal, welche Ausgabe kommt.

ralfkaminski, 2018-07-17:
Hallo,
Habe die Firmware noch einmal neu geflasht. Und jetzt funktioniert alles einwandfrei. Die Variablen werden von der CCU sowohl gelesen, als auch beschrieben. 
Vielen Dank für die schnelle Umsetzung. Bei dem nächsten offiziellen Release sollte das Feature auf jeden Fall enthalten bleiben. 
Schönen Abend noch. 
Gruß Ralf

jp112sdl, 2018-07-17:
Vielen Dank für das positive Feedback!

Ja, dann pack ich das so ins nächste Release 😄

PaulMerk, 2018-10-07:
bitte Release Release 1.0.28 (o.ä.) inkl. dieser Funktionalität zur Verfügung stellen.
Danke
Paul

jp112sdl, 2018-10-08:
@PaulMerk Ich habe die Binary noch mal mit der neuesten esp8266/Arduino Unterstützung kompiliert.

Bitte mal 2-3 Tage testen, dann mach ich ein Release.
Kann es leider nicht selbst machen, da ich nur noch 2 Sonoff Basic im Einsatz hab.

PaulMerk, 2018-10-08:
Danke für die Datei. Testen im Betrieb kann ich leider erst in den Weihnachtsferien, wird also noch was dauern.
Paul
--
Mit Mut und Entschlossenheit gegen Intoleranz Montag,