jp112sdl/SonoffHMLOX #67: Feature Request: Unterstützung für Sonoff Pow Rev2

feuerwasser85, 2018-07-17:
Hallo. Bitte könntest Du den Pow Rev2 von Sonoff auch mit in Deine Firmware mit aufnehmen? Wäre super, da der Rev1 ein Auslaufmodell ist und der Rev2 wesentlich mehr Funktionen hat.
Sollte das nicht möglich sein, dann never mind.
Danke!

jp112sdl, 2018-07-17:
Hi!
Welche Funktionen "mehr" hat denn der POW V2?
Wenn ich irgendwann mal einen haben sollte (was nicht so schnell passieren wird, weil ich keinen benötige), dann könnte ich da mal was bauen.
Ansonsten siehts vorerst leider schlecht aus...

jp112sdl, 2018-07-18:
Btw.: Der Wildwuchs bei den POWs wird immer schlimmer...
Es werden immer mehr verschiedene Mess-Chips verbaut.
https://github.com/xoseperez/espurna/issues/696

closed 2018-08-08 by jp112sdl