jp112sdl/SonoffHMLOX #69: POW schaltet keinen Strom, wenn kein WLAN verfügbar ist

ralfkaminski, 2018-08-01:
Hallo, 
heute wollte ich alle meine Geräte, die Dauerhaft laufen (fritzbox, raspi, nas ...) hinter einem POW anschließen, um den Stromverbrauch zur protokollieren. Den POW habe ich so konfiguriert, dass er nach einem Stromausfall den letzten Schaltzustand wieder herstellen soll.
Da er aber keine WLAN-Verbindung zur Fritzbox aufbauen konnte (Steckdosenleiste war noch aus) hat er auch keinen Strom geschaltet.
Erst als ich die Fritzbox an eine andere Steckdose angeschlossen habe und das WLAN Netz verfügbar war, schaltet der POW auch den Strom ein.
Könnte man den Programmablauf so gestalten, dass er den letzten Schaltzustand schon herstellt, bevor das eingestellte WLAN verfügbar ist?

Gruß
Ralf Kaminski

jp112sdl, 2018-08-01:
Das Problem ist, dass kein "offline"-Betrieb vorgesehen ist.
Ich könnte es so umbauen - jedoch führt der Sonoff nach erfolglosem WiFi-Connect einen Reboot durch. 
Das ist dann eine Endlosschleife, bis wieder WLAN verfügbar ist.

jp112sdl, 2018-08-01:
@BigJohnson68 Tja, so ist das, wenn man ein originäres WLAN-Device kauft.
Ein Auto hat auch keine Pedale als Reserve, falls der Motor ausfällt ;)

ralfkaminski, 2018-08-02:
Mir kam zu dem Thema noch die folgende Idee:
Kann man den automatischen Reboot über einen Schalter in der Konfiguration eventuell optional einstellen. Dann könnte er den letzten Schaltzustand wiederherstellen, auch ohne WLAN-Verbindung.
Sobald das WLAN dann verfügbar ist, könnte man den POW per Tastendruck mit dem WLAN manuell verbinden.
Oder aber wenn es möglich ist, die WLAN-Verbindung über ein Intervall automatisch versuchen herzustellen.