jp112sdl/SonoffHMLOX #70: Verbrauchszähler über Loxone vom Pow abrufen.

BigJohnson68, 2018-08-01:
ich bin zu blöd um diese Zeile in Loxone zu schieben.

Aufruf im Browser:
http://192.168.0.64/getPower
Resultat
[image]

<voltage>223.00</voltage><current>1.28</current><powerw>340.25</powerw><powerva>285.75</powerva><energycounter>2.84</energycounter>

einen Loxone Socket Air habe ich erfolgreich mit den Energiedaten gefüttert. 

könnt ihr mir da irgendwie weiterhelfen.

ebenso die Frage, warum die Zahlen bei mir so komisch übereinanderligen, ist am iphone und am PC gleich

[image]

vielen Dank vorab

jp112sdl, 2018-08-01:
> ebenso die Frage, warum die Zahlen bei mir so komisch übereinanderligen, ist am iphone und am PC gleich

Das wird demnächst mal gefixt. Ist ein CSS Fehler.

Zu der anderen Frage kann dir @jenshavelberg evtl. die Tage was sagen, da ich kein Loxone nutze.

jenshavelberg, 2018-08-02:
Bei Befehlserkennung mit ``<voltage>\v`` etc. bekommst du kein Ergebnis? Ich hab keinen POW zum Testen hier, wäre aber davon ausgegangen, dass es so funktioniert. Siehe auch: https://www.loxwiki.eu/display/LOX/Virtueller+HTTP-Eingang

jp112sdl, 2018-12-11:
Kommt hier noch jemand? Nee? Dann können wir ja zu machen. ^^

closed 2018-12-11 by jp112sdl