jp112sdl/SonoffHMLOX #73: Shelly 1 Wifi Relais

meskax, 2018-09-13:
Hi,

weiß nicht genau, an welcher Stelle ich das fragen kann und darf.
Wäre es möglich das HMLox auch auf den Shelly 1 Switch zu adaptieren, sodass
dieser auch unter Homematic läuft? Tasmota lässt sich flashen und nutzen

Danke und beste Grüße

jp112sdl, 2018-09-13:
Hi.

Ich werde an der SonoffHMLOX, außer Bugfixing, keine Änderungen mehr vornehmen.
Das betrifft auch das Einpflegen neuer Gerätetypen. 

Du solltest meine Firmware dennoch recht einfach auf den Shelly 1 bekommen, wenn du in der 
- [SonoffHMLOX.ino#L33](https://github.com/jp112sdl/SonoffHMLOX/blob/c4b7a90a03ef4f66d1480117343b852f6347f0a2/SonoffHMLOX/SonoffHMLOX.ino#L33) aus der `0` eine `5` und 
- [SonoffHMLOX.ino#L37](https://github.com/jp112sdl/SonoffHMLOX/blob/c4b7a90a03ef4f66d1480117343b852f6347f0a2/SonoffHMLOX/SonoffHMLOX.ino#L37) aus der `12` eine `4`.

machst.

BigJohnson68, 2018-09-14:
Servus,
den Quellcode habe ich jetzt gefunden, aber wie mache ich mit den Einstellungen für Shally 1 daraus eine Firmware? Hoffentlich nicht mit Atom.IO ;-(

https://github.com/jp112sdl/SonoffHMLOX/blob/master/SonoffHMLOX/SonoffHMLOX.ino

merci und sorry für die Frage

meskax, 2018-09-14:
> Hi.
>
> Ich werde an der SonoffHMLOX, außer Bugfixing, keine Änderungen mehr vornehmen.
> Das betrifft auch das Einpflegen neuer Gerätetypen.
>
> Du solltest meine Firmware dennoch recht einfach auf den [...]

Hi! Danke für den Hinweis, werde ich ausprobieren ;) Beste Grüße!

jp112sdl, 2018-09-15:
@BigJohnson68 
Du brauchst die Arduino IDE mit installierter ESP8266 Board-Unterstützung sowie die ArduinoJSON Library.

jp112sdl, 2018-10-17:
Dank eines großzügigen Spenders, der mir Shelly-Hardware hat zukommen lassen, gibt es nun eine Firmware für dieses Modell:
https://github.com/jp112sdl/ShellyHMLOX/releases

Die Einrichtung entspricht derselben wie bei SonoffHMLOX.

closed 2018-10-17 by jp112sdl

BigJohnson68, 2018-10-18:
supi für die Klasse Abeit
habe bei der Erstkonfig vergessen den UDP-Port zu hinterlegen.
Sobald ich das Shelly neu flasche bekommt er automatisch die von mir vergebene IP und zieht sich wohl die komplette vorherige Config.
Wie kann ich das Gerät komplett reseten, so dass ich das Gerät wieder auf 192.168.4.1 erreichen kann.
noch besser wäre natürlich, die Eingabe des Ports wie beim Sonoff auf die Konfigseite wieder sichtbar zu machen

weiter so

jp112sdl, 2018-10-18:
> Sobald ich das Shelly neu flasche bekommt er automatisch die von mir vergebene IP und zieht sich wohl die komplette vorherige Config.

Neu flashen löscht nie die Konfiguration. Diese liegt in einem anderen Speicherbereich.

> Wie kann ich das Gerät komplett reseten, so dass ich das Gerät wieder auf 192.168.4.1 erreichen kann.

Ein kompletter Reset ist nicht vorgesehen. Man kann den Konfigurationsmodus jederzeit wieder aufrufen, indem man im Browser z.B. die URL `http://shelly-ip/bootConfigMode` aufruft (siehe [Wiki](https://github.com/jp112sdl/SonoffHMLOX/wiki/Direkte_HTTP_Steuerung)) oder der Taster an Klemme SW gegen GND beim Anlegen der Stromzufuhr gedrückt wird.

> noch besser wäre natürlich, die Eingabe des Ports wie beim Sonoff auf die Konfigseite wieder sichtbar zu machen

Das sollte auch so sein. Muss ich mal checken.

Bitte dann künftig Probleme zum Shelly wegen der Übersichtlichkeit direkt dort verfassen: 
https://github.com/jp112sdl/ShellyHMLOX/issues

Danke! 😄

BigJohnson68, 2018-10-18:
Hallo Jerome,

 

vielen Dank noch für die Tipps

 

hatte vergessen auf Loxone umzustellen

config reset hat ebenfalls bestens funktioniert

 

merci nochmals und sorry wegen falscher Rubrik

 

Gruß

 

Johannes

 

https://github.com/jp112sdl/ShellyHMLOX/issues