jp112sdl/SonoffHMLOX #75: Fehler beim Flashen

wgerold, 2018-10-20:
Hallo,
bin ein kompletter Neuling.
Habe mir 5 Switch gekauft und möchte die flashen.
bekomme jedes Mal den folgenden Fehler:

jp112sdl, 2018-10-20:
Hi.
Das Flashen hat funktioniert.
Du musst das Gerät nun noch konfigurieren.
https://github.com/jp112sdl/SonoffHMLOX/wiki/Sonoff_Konfiguration

wgerold, 2018-10-20:
Danke für die Info. Leider merkt er sich die eingegebene fixe IP nicht...
Liebe Grüße

jp112sdl, 2018-10-20:
Kannst du mal das komplette Log posten?
Alles was kommt, ab dem Klicken "Speichern"

wgerold, 2018-10-21:
Hallo! Sorry, ich weiß leider nicht wie das geht. Was muss ich da machen?
LG

jp112sdl, 2018-10-21:
Na wo hast du denn die Meldungen gesehen, die du in deinem ursprünglichen Beitrag gepostet hast?

wgerold, 2018-10-21:
Hier die Ausgabe vom ESP-Loader:

wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
WM: failed to connect and hit timeout
bcn 0
del if1
usl

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

Danke für deine Geduld!

jp112sdl, 2018-10-21:
Hmm, ich kann da keinen Fehler erkennen. 
Das ist nur der Hinweis, dass im AP Modus keine Anmeldung erfolgte und somit nach dem Timeout ein Reboot erfolgt

wgerold, 2018-10-21:
Nächster Versuch: 
wifi evt: 7
wifi evt: 7
add 1
aid 1
station: a0:88:69:e9:95:62 join, AID = 1
wifi evt: 5
wifi evt: 7
wifi evt: 9
wifi evt: 9
wifi evt: 7
wifi evt: 9
wifi evt: 7
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
wifi evt: 7
wifi evt: 7
*WM: Request redirected to captive portal
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
[... 544 lines cut ...]
3ffffed0:  0000000f 00000008 00003031 00000000  
3ffffee0:  00000000 00000000 00000000 4010020c  
3ffffef0:  3ffffee4 3ffffee0 00000001 3ffffd34  
3fffff00:  3ffffd1c 3ffffd04 3ffffcec 4021d0ec  
3fffff10:  3ffe9913 3fff0e64 3fff199c 3fff1a80  
3fffff20:  3fff146c 3fff1390 3fff199c 4020d909  
3fffff30:  00000000 00000000 00000000 00000000  
3fffff40:  00000000 00000000 feefeffe feefeffe  
3fffff50:  feefeffe feefeffe feefeffe feefeffe  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  feefeffe feefeffe feefeffe feefeffe  
3fffff90:  00000000 feefeffe feefeffe feefeffe  
3fffffa0:  feefeffe feefeffe feefeffe 3fff1a5c  
<<<stack<<<

last failed alloc call: 40106E8C(220)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

jp112sdl, 2018-10-24:
Irgendwas ist da komisch...
```
*WM: DUP AP: XXXXXXXXXXXXX
*WM: XXXXXXXXXX
*WM: -56
*WM: XXXXXXXXXX
*WM: -60
```

wgerold, 2018-10-24:
Hallo!

Ich habe da, soweit ich mich erinnern kann, meine persönlichen Daten vom WLAN unkenntlich gemacht.... Brauchst du die?
Dann kann ich dir gerne alles schicken...

Danke nochmals für dein Engagement!

Lg, Wolfgang

wgerold, 2018-10-26:
Hallo!

Heute nochmal versucht. Wieder das selbe - er merkt sich die IP nicht und bekommt über DHCP irgendeine IP zugewiesen. Untenstehend das Log (meine Daten habe ich anonymisiert - MEINE-SSID, MEINE-GEWÜNSCHTE-IP, etc.)

wifi evt: 7
add 1
aid 1
station: a0:88:69:e9:95:62 join, AID = 1
wifi evt: 5
wifi evt: 9
wifi evt: 9
wifi evt: 9
wifi evt: 7
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
*WM: Request redirected to captive portal
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Handle root
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
wifi evt: 7
wifi evt: 7
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
mode : sta(bc:dd:c2:0e:46:f0) + softAP(be:dd:c2:0e:46:f0)
add if0
wifi evt: 8
wifi evt: 7
wifi evt: 7
scandone
*WM: Scan done
*WM: DUP AP: MEINE-SSID
*WM: DUP AP: MEINE-SSID
*WM: MEINE-SSID
*WM: -36
*WM: MEINE-ANDERE-SSID
*WM: -72
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
err already associed!
station: a0:88:69:e9:95:62 leave, AID = 1
rm 1
wifi evt: 6
wifi evt: 7
add 1
aid 1
station: a0:88:69:e9:95:62 join, AID = 1
wifi evt: 5
wifi evt: 9
*WM: Sent config page
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
wifi evt: 7
wifi evt: 7
wifi evt: 7
*WM: Request redirected to captive portal
*WM: Handle root
wifi evt: 7
wifi evt: 7
*WM: Request redirected to captive portal
*WM: Handle root
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
*WM: Request redirected to captive portal
*WM: Handle root
wifi evt: 7
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
wifi evt: 7
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
*WM: Request redirected to captive portal
*WM: Handle root
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
*WM: Request redirected to captive portal
*WM: Handle root
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
*WM: Request redirected to captive portal
*WM: Handle root
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
*WM: Request redirected to captive portal
*WM: Handle root
*WM: WiFi save
*WM: Parameter
*WM: sonoffmodel
*WM: 0
*WM: Parameter
*WM: ccu
*WM: IP-MEINER-CCU
*WM: Parameter
*WM: loxudpport
*WM: 
*WM: Parameter
*WM: sonoff
*WM: SWITCH1
*WM: Parameter
*WM: hmpowervariable_pow
*WM: 
*WM: Parameter
*WM: hmecvariable_pow
*WM: 
*WM: Parameter
*WM: custom_powermeasureinterval_pow
*WM: 10
*WM: Parameter
*WM: restorestate
*WM: 0
*WM: Parameter
*WM: leddisabled_switch
*WM: 
*WM: Parameter
*WM: gpio14mode_switch
*WM: 0
*WM: Parameter
*WM: custom_gpio14assender_switch
*WM: 
*WM: Parameter
*WM: backendtype
*WM: 0
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: custom_ip
*WM: MEINE-GEWÜNSCHTE-IP
*WM: Parameter
*WM: custom_netmask
*WM: 255.255.255.0
*WM: Parameter
*WM: custom_gw
*WM: MEIN-GATEWAY
*WM: static ip
*WM: MEINE-GEWÜNSCHTE-IP
*WM: static gateway
*WM: MEIN-GATEWAY
*WM: static netmask
*WM: 255.255.255.0
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: MEINE-GEWÜNSCHTE-IP
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 2
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
scandone
switch to channel 6
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 7
cnt 

connected with MEINE-SSID, channel 6
ip:MEINE-GEWÜNSCHTE-IP,mask:255.255.255.0,gw:MEIN-GATEWAY
ip:MEINE-GEWÜNSCHTE-IP,mask:255.255.255.0,gw:MEIN-GATEWAY
wifi evt: 0
wifi evt: 3
*WM: Connection result: 
*WM: 3
station: a0:88:69:e9:95:62 leave, AID = 1
rm 1
bcn 0
del if1
pm open,type:2 0
mode : sta(bc:dd:c2:0e:46:f0)
Should save config
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Already connected. Bailing out.
*WM: IP Address:
*WM: MEINE-GEWÜNSCHTE-IP
Wifi Connected
CUSTOM STATIC IP: 0.0.0.0 Netmask: 0.0.0.0 GW: 0.0.0.0
Custom IP Address is set!
saving config
wifi evt: 6
wifi evt: 8
{"ip":"MEINE-GEWÜNSCHTE-IP","netmask":"255.255.255.0","gw":"MEIN-GATEWAY","ccuip":"IP-MEINER-CCU","sonoff":"SWITCH1","restoreOldState":0,"ledDisabled":false,"backendtype":0,"loxudpport":"","powervariable":"","ecvariable":"","measureinterval":10,"sonoffmodel":0,"gpio14mode":0,"gpio14assender":false}
state: 5 -> 0 (0)
rm 0
pm close 7
del if0
usl
wifi evt: 1
STA disconnect: 8

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

jp112sdl, 2018-10-26:
Ich kann keine Fehler/Probleme aus dem Log feststellen.

Kannst du bitte mal diese Version flashen und es damit probieren:
[SonoffHMLOX.ino.esp8285.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/2519925/SonoffHMLOX.ino.esp8285.bin.zip)

wgerold, 2018-10-26:
Damit scheint es leider überhaupt nicht zu funktionieren.

Hier die Infos von Punkt "Chip-Info":
Chip ID
935664
Flash Chip ID
1335429
IDE Flash Size
1048576 bytes
Real Flash Size
1048576 bytes
Soft AP IP
192.168.4.1
Soft AP MAC
BE:DD:C2:0E:46:F0
Station MAC
BC:DD:C2:0E:46:F0

Ich muss jetzt dann leider weg. Kann erst morgen weiter probieren...
Danke für deine Geduld!

wgerold, 2018-10-27:
Hallo!
Was mich immer irritiert sind folgende Zeilen:

CUSTOM STATIC IP: 0.0.0.0 Netmask: 0.0.0.0 GW: 0.0.0.0
Custom IP Address is set!
saving config

lg

jp112sdl, 2018-10-27:
Das sollte nur beim ersten Speichern passieren.
Nach dem Reboot sollte dann dort die richtige IP stehen.
Ansonsten ist beim Speichern was schief gelaufen, oder es konnte beim Speicher keine Verbindung mit dem WLAN hergestellt werden (diese wird vor dem Schreiben der Config-Datei auf Richtigkeit/Funktion geprüft).

jp112sdl, 2018-10-27:
Schön, dass du eine Lösung gefunden hast.

Weitere Firmwareversionen sind nicht geplant, da mein Schwerpunkt auf anderen Projekten liegt.
Und wenn ich gerade mal so den Code mit den neuesten Libraries kompiliere, siehts auch nicht gut aus... 🙄  Da müsste ich echt noch mal viel Zeit und Arbeit investieren.

closed 2018-10-27 by jp112sdl

wgerold, 2018-10-27:
Sorry, leider hab ich gerade entdeckt, dass er sich überhaupt keine Einstellungen merkt und wenn er länger vom Netz ist die komplette Konfiguration verliert. Somit leider doch keine brauchbare Lösung...

jp112sdl, 2018-10-27:
Ich bin da wirklich überfragt... Bei 99% der anderen Nutzer funktioniert es. Daher vermute ich auch keinen Bug im Code. Aus den Logs geht auch nix hervor, dass irgendwas schief läuft.

> wenn er länger vom Netz ist die komplette Konfiguration verliert.

(Die Dauer ist unerheblich. Die Config liegt in einer Datei im Flash.)

Alternativ kann ich dir da leider nur raten, Tasmota oder eine andere Firmware zu verwenden.

jp112sdl, 2018-10-27:
Ob es was bringt, weiß ich nicht.
Aber einen Versuch ist es noch wert... Nimm mal die:
[SonoffHMLOX.ino.esp8285.bin.zip](https://github.com/jp112sdl/SonoffHMLOX/files/2521745/SonoffHMLOX.ino.esp8285.bin.zip)