jp112sdl/SonoffHMLOX #76: Obi Wifi model

FiegeU, 2018-11-19:
Hello.

What exact model of the OBI plug is supported ?

This:
https://www.obi.de/hausfunksteuerung/wifi-stecker-schuko/p/2291706

Or that (or both):?
https://www.obi.de/hausfunksteuerung/wlan-steckdose-weiss/p/4717583

Thanks
Ulrich

FiegeU, 2018-11-20:
Okay, danke, wer lesen kann ist wohl klar im Vorteil ;).
Scheinbar scheint man dieses Modell ja nicht zu bekommen, oder hätte jemand eine Lieferadresse zur Hand ? Der Onlineshop hat sie derzeit nicht.

Mal eine andere Frage, ist dieser zeitweise kein WiFi Connect bzw keine Erreichbarkeit eigentlich jetzt vollständig gefixt ? Vor ein paar Monaten stand ja im Raum, dass es an der unterliegenden Firmware liegt.
Danke.

jp112sdl, 2018-11-20:
> oder hätte jemand eine Lieferadresse zur Hand ?

Ich denke nicht, dass deinen Aufruf hier großartig jemand lesen wird ;)

Frag mal im [HomeMatic Forum](https://homematic-forum.de/forum/viewforum.php?f=76).

> Mal eine andere Frage, ist dieser zeitweise kein WiFi Connect bzw keine Erreichbarkeit eigentlich jetzt vollständig gefixt ?

Ich glaube nicht. Ich entwickle aber auch nicht mehr weiter an der WiFi Lösung sondern nur am Umbau auf HomeMatic Funk für diverse andere Zwischenstecker.

FiegeU, 2018-11-21:
Moin.
> Ich glaube nicht. Ich entwickle aber auch nicht mehr weiter an der WiFi Lösung sondern nur am Umbau auf HomeMatic Funk für diverse andere Zwischenstecker.

Welche Hardware benutzt Du denn dafür ?

jp112sdl, 2018-11-21:
> Welche Hardware benutzt Du denn dafür ?

https://github.com/jp112sdl/AskSinPPCollection/blob/master/Markdown/Umbau.md

jp112sdl, 2018-11-21:
Einerseits das.
Zweitens kannte ich die AskSinPP Lib zum Zeitpunkt der Entwicklung der WiFi-Firmware noch nicht.
Drittens ist es wesentlich entspannter, was die Integration in Homematic (Funk) angeht.
Kein Umweg mehr übers WLAN, Möglichkeit von Direktverknüpfungen, etc.pp.