jp112sdl/SonoffHMLOX #77: ESP826612E / 12F support

beastieboy, 2018-12-09:
I tried to flash this firmware (with some GPIO mapping modifications) to ESP8266-12F but SPIFFS doesn't work. Any changes for support of above modules? Mine had 4MB flash size.

jp112sdl, 2018-12-09:
Hmm... there is nothing special with SPIFFS and ESP8266 12-E.
Please try the button "Factory Reset" when you are in AP-mode.
This will format the SPIFFS.

beastieboy, 2018-12-09:
I tried it for many times but without success. Maybe I can put here some debug info from serial console in few days. Are you sure that SPIFFS should not be initiated here with separate file written to the flash memory?

jp112sdl, 2018-12-09:
I'm not sure. And I'm not working on ESP code anymore. So the only way you could accomplish your task is using google or ask in other communities... sorry. :/

closed 2018-12-11 by jp112sdl