jp112sdl/SonoffHMLOX #79: WLAN Verbindung

telebodo, 2018-12-11:
Hallo guten Tag,
danke für deine Super Software,
ich habe bemerkt, dass sich die Sonoff's alle 30 Sek. Neu im WLAN anmelden.
Kann ich diese Zeit ändern und wie?

Danke Mirko

jp112sdl, 2018-12-11:
Hi.
Ist mir nicht bekannt. Ich habe da nichts vorgegeben.
Das erfolgt evtl. durch eingebaute Energiesparmaßnahmen.
Ist die Funktion dadurch eingeschränkt?

telebodo, 2018-12-11:
Hallo,

danke für die schnelle Antwort. Wenn sich die Sonoff's neu verbinden sind sie für die Alexa nicht erreichbar.
Von der Seite her wäre eine längere Zeit besser. Ich habe keine Energiesparmaßnahmen in der Fritzbox.

telebodo, 2018-12-11:
Und in den WIFI Einstellungen gibt es keine Einstellung die das vorgibt?

Mit einer anderen Tasmota Version geschieht dies nicht. die ist aber nicht so Optimal für die Homematic wie deine :-)

jp112sdl, 2018-12-11:
> Und in den WIFI Einstellungen gibt es keine Einstellung die das vorgibt?

Ist mir zumindest nicht bekannt.

> Mit einer anderen Tasmota Version geschieht dies nicht.

Das kann gut sein. :) Da steckt auch viel mehr (und vor allem aktive) Entwicklung in dem Projekt...

telebodo, 2019-01-05:
Hallo guten Abend, ich würde gern noch mal auf mein Problem zurück kommen.
Ich habe mal in der SonoffHMLOX.ino nachgeschaut und einen Eintrag in Zeile 343 

 //Reconnect WiFi wenn nicht verbunden (alle 30 Sekunden)
  if (WiFi.status() != WL_CONNECTED) {
    WiFiConnected = false;
    if (millis() - LastWiFiReconnectMillis > 30000) {
      LastWiFiReconnectMillis = millis();
      DEBUG("WiFi Connection lost! Reconnecting...");
      WiFi.reconnect();
    }
  } else {
    if (!WiFiConnected) {
      DEBUG("WiFi reconnected!");
      WiFiConnected = true;
}
Könnte dort mein Problem liegen. Kannst du mir sagen ob ich diesen Wert auf 3000000 mill.sek. setzten kann. 
Wenn nicht könntest du mir eine SonoffHMLOX.bin mit dem Wert zukommen lassen?

Danke Mirko