jp112sdl/SonoffHMLOX #82: Sonoff Basic Rückmeldung an CCU

Spessart9, 2018-12-23:
Hallo,

leider funktioniert bei mir die Rückmeldung beim Drücken der lokalen Taste am Sonoff Basic nicht. Ich verwende eine CCU3 mit aktueller Rasperrymatic-Software. Was könnte die Ursache sein?

Gruß
Rainer

jp112sdl, 2018-12-25:
Hi,

> Was könnte die Ursache sein?

- z.B. Name im Sonoff stimmt nicht mit dem Namen des CUxD **Kanals** überein
- falsche CCU IP eingetragen

Führe mal ein `http://<sonoff-ip>/reloadCUxD` aus und schau mal was kommt.

Spessart9, 2018-12-25:
Hallo,
der Gerätename lautet „Sonoff-Basic-001“. Die IP-Adresse passt.
Der CUXD Name lautet „Sonoff1“, entsprechend ebenso der Name in der CCU3. Der Kanalname lautet  wie der Gerätename „Sonoff-Basic-001“. Gibt es hier Beschränkungen bzgl. der Länge bzw. der benutzten Zeichen?
Beim Aufruf von „IP/reloadCUxD“ erscheint immer „CUxD Address = CUxD.“. Offensichtlich passt die Zuordnung nicht.
Muss die Namensvergabe von Beginn an so sein, oder können die Namen auch nachträglich geändert werden?

Gruß
Rainer

jp112sdl, 2018-12-25:
Hmm,

> Der Kanalname lautet wie der Gerätename „Sonoff-Basic-001“. Gibt es hier Beschränkungen bzgl. der Länge bzw. der benutzten Zeichen?

Ja, die liegt aber bei 255. Da bist du locker noch drunter.

Teste bitte mal folgenden Aufruf im Browser (<ccu> noch ergänzen):
`http://<ccu>:8181/cuxd.exe?ret=dom.GetObject(%22Sonoff-Basic-001%22).Address()`

Da sollte sowas bei rauskommen:
```
<xml>
<exec>/cuxd.exe</exec>
<sessionId/>
<httpUserAgent>
User-Agent: ....
</httpUserAgent>
<ret>CUX2801003:1</ret>
</xml>
```

In `<ret>...</ret>` muss die CUxD Adresse erscheinen.

> Muss die Namensvergabe von Beginn an so sein, oder können die Namen auch nachträglich geändert werden?

Die Namen können auch nachträglich geändert werden.
Die Reihenfolge ist hier entscheidend. Es empfiehlt sich, zuerst den CUxD Kanalnamen zu ändern und dann den Sonoff Namen.
Wenn du in der Weboberfläche des Sonoff auf den Button "Konfiguration" klickst, kannst du dort den Namen ändern. Beim Speichern wird auch gleich geprüft, ob eine entsprechende CUxD Geräte Adresse abgerufen werden kann.

Spessart9, 2018-12-25:
Hallo,

ich bekomme folgende Rückmeldung:

<?xml version="1.0" encoding="ISO-8859-1"?>
<xml>
<exec>/cuxd.exe</exec>
<sessionId/>
<httpUserAgent>User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/64.0.3282.140 Safari/537.36 Edge/17.17134</httpUserAgent>
<ret>CUX2801001:1</ret>
</xml>

Kann es etwas mit den Firewall-Einstellungen der CCU zu tun haben?

Gruß
Rainer

Spessart9, 2018-12-25:
`<?xml version="1.0" encoding="ISO-8859-1"?>
<xml>
<exec>/cuxd.exe</exec>
<sessionId/>
<httpUserAgent>User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/64.0.3282.140 Safari/537.36 Edge/17.17134</httpUserAgent>
<ret>CUX2801001:1</ret>
</xml>`

jp112sdl, 2018-12-25:
> Kann es etwas mit den Firewall-Einstellungen der CCU zu tun haben?

Eigentlich nicht. Der Sonoff macht auch nix anderes, als genau diese URL aufzurufen und die CUxD-Adresse aus dem Inhalt des ret-Wertes auszuwerten.

Oder sind deine Sonoffs in einem anderen Netz, das evtl. von der Firewall ausgenommen ist?

Spessart9, 2018-12-26:
Danke schon mal für Deine Hilfe. Alle Geräte (Sonoff, CCU und PC) sind im gleichen Netz. 
Kann es etwas mit den Zeichen im Gerätenamen ("-") zu tun haben? Sonst noch eine Idee was ich ausprobieren kann?

jp112sdl, 2018-12-26:
Lass bitte mal ein serielles Log mitlaufen.
FTDI anschließen, ein Terminalprogramm (zB Putty) seriell auf 115200 Baud stellen und schauen, was kommt.
Am besten mal vom Boot an alles anzeigen lassen.
Oder zumindest das, was kommt, wenn du `http://<sonoff-ip>/reloadCUxD` ausführst.

Fiiti, 2019-01-02:
Hätte das gleiche Problem.
Nachdem ich den Namen ohne Sonderzeichen und erheblich auf 12 Zeichen gekürzt hatte lief es sofort.
Wobei ich nicht testete, was genau das Problem war.

Spessart9, 2019-01-05:
So, heute wieder etwas Zeit gehabt zum rumspielen.
Nachdem ich CUXD mehrfach gelöscht, den Raspi neu gestartet und CUXD neu installiert und konfiguriert habe, läuft es. Danke für die Unterstützung!

closed 2019-01-05 by Spessart9